letscontrolit/ESPEasy #3665: [P022 - Extra IO - PCA9685] PCAPWM,0,4095 is not working

heinemannj, 2021-06-02:
Dimming of an LED via standard PWM command is working fine.
Dimming via **PCAPWM,0,4095 is not working**.

**I have already downgraded ESPEasy firmware to a previous release from May 2021.** - No change

`MC: ESP32-WROM-32`

`Plugin: P022 - Extra IO - PCA9685`

Device: https://www.az-delivery.de/products/pca9685-servotreiber?_pos=2&_sid=0c3a094d7&_ss=r
Datasheet: https://cdn.shopify.com/s/files/1/1509/1638/files/PCA9685_Servotreiber_Datenblatt.pdf?17187252465155634920

[image]

[image]

[image]

[image]

[image]

[image]

[image]

1684480: WD   : Uptime 28 ConnectFailures 0 FreeMem 208372 WiFiStatus WL_CONNECTED ESPeasy internal wifi status: Conn. IP Init
1698480:  Webserver args: 0: 'cmd' length: 13
1698481: HTTP: PCAPWM,0,4000
1698486: PCA 0x40: PWM 0
1714481: WD   : Uptime 29 ConnectFailures 0 FreeMem 207872 WiFiStatus WL_CONNECTED ESPeasy internal wifi status: Conn. IP Init

TD-er, 2021-06-02:
Can you test with a build somewhere at the end of last year?

Also you did show a result of an I2C scan. Does this scan work too after testing it?

heinemannj, 2021-06-02:
In addition I've tried with **ESP8266** - **firmware mid of last year**

[image]

Nearly the same result => s**tate has been changed from 0 to -1**:

[image]

After performing above command I2C scan as following:

[image]

TD-er, 2021-06-02:
The reason for the I2C scan is related to this issue: https://github.com/letscontrolit/ESPEasy/issues/3664
But apparently your situation does not have anything to do with it.

What I2C clock frequency do you use?
Can you set both high and low clock freq. to 100 kHz?

heinemannj, 2021-06-02:
I've switched I2C clock frequency from 400kHz to 100kHz on both MC ESP-32 and on ESP-8266.

No change

In addition I've connected to ESP-8266 I2C-Bus an BH1750 sensor which is working fine ...

TD-er, 2021-06-02:
I do have one of those in my sensor collection, but not sure if I can test it before this weekend.

heinemannj, 2021-06-02:
In parallel I will try to test with ESP-32 arduino sketch following vendoe instructions.
But my final goal is to have it up and running with ESPEasy ;-)

TD-er, 2021-06-02:
> In parallel I will try to test with ESP-32 arduino sketch following vendor instructions.
> But my final goal is to have it up and running with ESPEasy ;-)

Sure and it should...
Have not seen any reports of it not working in a while, so either it is not a popular plugin, or it should just work.
I have never used it though.

Dickesplumpel, 2021-06-21:
Hi, have the same problem, tried ESP82xx_mega-20200608, ESP82xx_mega-20210223 and ESP82xx_mega-20210503 on two different PCA9685 boards, different PWM values, "state" is always 0 and the voltage on the selcted "pin" is always 0V on both i2c bus clock frequencies.

heinemannj, 2021-06-24:
The following change for power supply (separation of power for ESP32 and PCA9685s) has reduced my impact:

[image]

**Issue A:**

However: **Switching on both at the same time (ESP32 and PCA9685s) is disturbing ESP32**

If ESP32 is powered on only
- alone (Power supply for PCA9685s is off)
- Connect via ESPEasy WebGUI is successful
- PCA9685s are powered on afterwards
- Go to ESPEasy Hardware page => submit
- Check on ESPEasy "Tools - I2C Scan" page if boths PCA9685s are connected

PCA9685s I2C connects are working for a limited period of time.

**Issue B:**

PCA9685 device settings and how they work are fully intransparent for me.

[image]

At least

```
INVRT: 0
OCH: 0
OUTDRV: 1
OUTNE: 10

=> 00110
```
Binary to Decimal converter: https://www.rapidtables.com/convert/number/binary-to-decimal.html

`Mode2 => 0x6 (6)`
is working for me.

[image]

**Issue C:**

The command implementation of

- PCAPWM
- deviceID.PWM

is working but buggy in general

**Any help is appreciated - My skill for electrical setup is very limited.**

TD-er, 2021-06-24:
Can you test (on ESP8266) with a build of roughly 2 years ago as that's from before we started to make GPIO handling an ESPEasy core task.

Dickesplumpel, 2021-06-24:
Hi, mega-20190305_normal_ESP8266_4M, mega-20190216_normal_ESP8266_4096, mega-20190121_normal_ESP8266_4096, mega-20181204_normal_ESP8266_4096 show wrong state. In Version mega-20181128_normal_ESP8266_4096 shows right status for PWM but my servo shields don't work, maybe they are broken or dead.
Mega_20181204 shows on command pcapwm,0,1000
{
"log": ": PWM 0",
"plugin": 22,
"pin": 0,
"mode": "PWM",
"state": 0
}
Mega_20181204 shows:
{
"log": "0: PWM 0",
"plugin": 22,
"pin": 0,
"mode": "PWM",
"state": 1000
}

heinemannj, 2021-06-25:
If have tested with **Release mega-20181127** and **two working PCA9685**s:

**IC2 stability is OK** - NO issue identified so far

**1. PCA9685** - for 1. LED: switiching is working:
[image]

**2. PCA9685** - for 2. LED: switiching is also working:
[image]

However **in "Pin State Buffer" only one entry**:
[image]

Dickesplumpel, 2021-06-25:
New setup now:
Wemos-D1-mini USB powered, ESP_Easy_mega_20210503_normal_ESP8266_4M1M, external V+ (10V in my case), Vcc 3,3V from Wemos and one motorshield Mode2 0x6 1500Hz Range 4095 shows wrong state but works correct, can adjust outputvoltage from 0-10V for controlling actuators for lifting valves - great, happy, jump - wrong state doesn't matter in my case because I get feedback over changing water temperature! (tested 1 hour now)
Same Hardware but mega_20181128 shows right state and works correct (tested only 10 minutes).

heinemannj, 2021-07-08:
There was a big change on mega-20181130 on which P022 plugin code was modified:

https://github.com/letscontrolit/ESPEasy/commits/mega-20181130
https://github.com/letscontrolit/ESPEasy/commit/b49921eb9acbc9c5e6c90737f9a4395a994436ac
https://github.com/letscontrolit/ESPEasy/pull/2057

tonhuisman, 2022-12-29:
@Dickesplumpel @heinemannj Çan you retest with current [ESPEasy release](https://github.com/letscontrolit/ESPEasy/releases), as this issue is most likely fixed.

Dickesplumpel, 2022-12-30:
Hello, everything works fine with current ESPeasy release, thank you very much! :-)

closed 2022-12-30 by TD-er