letscontrolit/ESPEasy #3666: Issue with packed_decodeUplink.js payload formatter

filipq, 2021-06-02:
The packed_decodeUplink.js doesn't seem to be working correctly. After copy & paste to TTN Console V3 -> Applications -> Payload formatters -> Uplink -> Javascript it won't save changes. Instead, it will generate the below error:

"Invalid link: embedded message failed validation" 

Details:

{
  "code": 3,
  "message": "error:pkg/errors:validation (invalid `link`: embedded message failed validation)",
  "details": [
    {
      "@type": "type.googleapis.com/ttn.lorawan.v3.ErrorDetails",
      "namespace": "pkg/errors",
      "name": "validation",
      "message_format": "invalid `{field}`: {reason}",
      "attributes": {
        "field": "link",
        "name": "SetApplicationLinkRequestValidationError",
        "reason": "embedded message failed validation"
      },
      "correlation_id": "673ce19fef1d425491255a09ae7b2872",
      "cause": {
        "namespace": "pkg/errors",
        "name": "validation",
        "message_format": "invalid `{field}`: {reason}",
        "attributes": {
          "field": "default_formatters",
          "name": "ApplicationLinkValidationError",
          "reason": "embedded message failed validation"
        },
        "cause": {
          "namespace": "pkg/errors",
          "name": "validation",
          "message_format": "invalid `{field}`: {reason}",
          "attributes": {
            "field": "up_formatter_parameter",
            "name": "MessagePayloadFormattersValidationError",
            "reason": "value length must be at most 16384 runes"
          },
          "code": 3
        },
        "code": 3
      },
      "code": 3
    }
  ],
  "request_details": {
    "url": "/as/applications/test-app12/link",
    "method": "put",
    "stack_component": "as"
  }
}

TD-er, 2021-06-03:
It is too large.
For TTN v3, I have created a vendor template.
This includes the full v3 decoder as vendor templates do not (yet?) have this issue.

[image]

[image]

After finishing the vendor template steps, the application specific payload script is installed:
[image]

See also this discussion on [TheThingsNetwork forum](https://www.thethingsnetwork.org/forum/t/how-to-reduce-size-of-payload-formatter-script-now-larger-ones-are-blocked/47879)

filipq, 2021-06-04:
Hello 

Thank you very much for your reply. All working fine now. 

Regards
Filip

closed 2021-06-04 by filipq

TD-er, 2021-06-04:
You were a bit too fast... as I have not had time to document this new vendor feature.
It just got merged by TTN less than 2 weeks ago.

Glad it is now working :)