letscontrolit/ESPEasy #3668: ESPEasy Modbus TCP (Siemens Logo 8)

THAWINKELS, 2021-06-03:
I want to communicate with my Siemens logo 0ab8, how can I implement the modbus protocol?

[image]

Anyone who can help me?

uzi18, 2021-06-03:
On ESPEasy we have modbus client only available, so you can fetch data to espeasy  tasks. 
But it could be nice feature to have also server mode.

THAWINKELS, 2021-06-03:
> On ESPEasy we have modbus client only available, so you can fetch data to espeasy tasks.
> But it could be nice feature to have also server mode.

How should it look like in the Rules interface on the ESP easy?

uzi18, 2021-06-03:
No it is imposible now, but maybe we can add some commands for rules to read/write data to remote device.
You should ask for new rules feature or add new modbus controller, but there are many types of modes to support, @TD-er what do you think?

Can try to check what is needed to communicate with LOGO by modbus tcp/udp.

TD-er, 2021-06-03:
> Can try to check what is needed to communicate with LOGO by modbus tcp/udp

Siemens 7 step.

I have once implemented a layer (on my previous job) to interact with Siemens PLCs.
Problem with the 7 step layer is that you need to change the PLC code to allow access to the registers.

THAWINKELS, 2021-06-03:
[image]
[image]

I can add the ESP in my network in Logo soft somfort from Siemens, And can choose the correct Bit,Word...

uzi18, 2021-06-03:
For logo it is Logo Soft Comfort and it is not a problem but security.

uzi18, 2021-06-04:
[image]
From logosoft comfort 8.2 help system