letscontrolit/ESPEasy #3672: [ESP32] Webserver may wait forever without WDT reset

TD-er, 2021-06-05:
When there is a programming error, for example the code looping forever generating a list of GPIO selections rendering the web page.
(I deny that ever happened while testing....)
The ESP does not reboot/crash, but the webpage does also not timeout and the unit remains unaccessible forever until it reboots.

Serving a web page should have some kind of watchdog timer which does allow to kick out of a function serving the web page.
Not sure yet how we could implement this using the RTOS features, but would be nice to trigger some exception to jump out of the stuck code.

qiudecai, 2021-10-18:
The power supply should remain stabl
[image]
e

TD-er, 2021-10-18:
> I think the original issue here is that WDT doesn't seem to work (at all) on ESP32, at least I've not seen it being triggered here.

True, the original issue is more about it not being triggered while serving web pages.
It can be triggered in other occasions.

qiudecai, 2021-10-18:
Is WDT without work, ESP32 does not implement restart, ESP32 without work, a serial port without output. I will add 220 uf at 3.3 V, thank you

TD-er, 2021-10-18:
> Is WDT without work, ESP32 does not implement restart, ESP32 without work, a serial port without output. [...]

Can you explain a bit more what you're trying to tell here?
I don't think I understand what you intend to say here.

TD-er, 2021-10-18:
Yep, those 2 transistors Q1 and Q2 are used to toggle reset and hold GPIO-0 down to enter programming mode.

How is your board powered?
Some boards with ESP32 do have some issues regarding how the USB to serial chip is powered.
There must be a low drop diode between the VCC pin of the USB connector and the USB to UART chip.
This to allow the chip to be powered also when not using the USB port to power all.

If that's not present, then the USB to serial chip may not be in a known state when you power the board via the 5V pin of the board (or only power it via 3V3 pin)
This may then cause the GPIO-0 pin to be high during boot, as the state of the USB to UART chip is unknown.

qiudecai, 2021-10-19:
The VCC of USB and the VCC of CH340 are both connected to VCC5, and then connected to the diode D6 of MP2303. D6 is the diode you mentioned.

qiudecai, 2021-10-19:
Now I suspect it is the CH340 chip problem