letscontrolit/ESPEasy #3675: Cant connect to AP  ESP32<>ESP8266 (espeasy p2p)

crazybanane, 2021-06-05:
Hello,

I got 1 ESP32 and 1 ESP8266.
I use the ESP32 as Root-Station for a P2P network he will get Informations from the ESP8266. I can configure all functions and in the Wifi Network all functions Work great. 

But when i want to say ESP32 makes AP and i want connect the ESP8266 to it then the connection will not enter or the Firmware crashes completly so that i have to reflash the whole System. 

What i have to do that i can use this system as a seperate network.

Actual Version of espeasy that ist flashed on both devices are Mega 20133/ ESP_Easy_mega_20210503_normal_ESP8266

TD-er, 2021-06-06:
Which end does crash? The ESP32 or the ESP8266?
Does the ESP32 also have another network connection, like Ethernet?

Can you also try if it does work with a build from the end of 2020? I did change some parts on how WiFi connections are being handled and I am not entirely sure if this also may prohibit this specific use case.

Also a number of users has reported issues with WiFi on the 20210503 build.

crazybanane, 2021-06-07:
Hello,

so i tried the last 3 Realeases. Sometimes the whole ESP32 crashes with this known Wifi issues on build 20210503 but is the only build where i can set the Dont´t force to Setup function but it didnt work i got everytime the Setup page that i have to connect to another WiFi. Sometimes i got error:
dhcps: send_offer>>udp_sendto result 0
29584 : Error  : UNKNOWN WIFI/ETH EVENT: 18

Or an unlimited list With Disconnection errors.

No both ESPs didnt get a local Wifi connection because i need the ESP32 as Host where the ESP8266 can connect. I want that it will run as a own network or STandalone with the ESP P2P features.

I tried it as DHCP or with Fixed IP the ESP8266 will connect to the AP of ESP32 but the ESP32 will not accept P2P networking and wont show the ESP8266 as Network Node.

My Prob is i want to build a Garden Watcher but my garden is not in the near of any WiFi 1,5km away...
So i tried at home an little automation System for Scan Soil Moisture in some Areas where plant will grow, so that the ESP8266 will senn this Informations to the ESP32 an he will start a Waterpump an some Valves to watering the Plants they need water. 
Espeasy is the only System i found that is so easy (the name describes it really) to configure and set.
I tried it to make it as an Home Automation System with Raspberry pi and OpenHAB but that is soooooooo complicated so i made it only with the functions from espeasy (ESP P2P Networking and Rules) and it works like a charm but only at home with Local Wifi, Is there any way to get this System Standalone? It would be great if there will be an Option or Version for a Standalone/Host.

After a whole trys i got a stable connection but nodes will not be shown on Start site so need Help.

TD-er, 2021-06-07:
OK, we're dealing with a number of issues here.
I think all may need code changes from my side, but that's not a problem as I'm working on that part of the code mainly to fix those WiFi issues mentioned before ASAP.

The first issue is the reason why you can't use p2p right now.
ESPeasy is probably not listening to UDP traffic on the AP interface.
This needs some code changes in ESPEasy and maybe also a setting to switch this as I'm not 100% sure we can run it on both interfaces at the same time.
But I don't think it is a big change, so that will be done soon.

The next issue is one probably specific to ESP32.
I will have to look into this unknown ETH/WiFi event, to see what it is.
Could be something simple, but might also not be... we'll have to see.

Next one is when ESPEasy considers its network to be available.
Currently the network is considered initialized if:
- Connection is established (ether to an access point or via Ethernet)
- IP is assigned (either static or via DHCP)

However it does not consider it 'established' if it is running as an access point itself.
The web server is started, but not the services like controllers.  (to send data collected by a task)
This might be quite some work to get it right.

As some kind of work-around one could setup a 3rd ESP node, which only acts as an access point.
However I am not sure if the AP mode on an ESP also allows to route IP packets between connected stations.
For example Node 1 = AP, node 2&3 are stations connected to 1's AP.
I don't know if node 2 can even ping node 3.
If you're running an old separate access point, then this AP can route packets between connected nodes via WiFi so node 2 can ping node 3 etc.

I think that's probably the way to go as I have no idea if I can solve all other issues that are now preventing you to create a local p2p network.

crazybanane, 2021-06-27:
Found solution for connection with RPIEasy so i dont need any Internet connection to do what i wanted.

if this is helpfull for Wiki or Forum.

https://github.com/enesbcs/rpieasy/issues/237#issuecomment-869154909

tonhuisman, 2023-09-02:
Not sure if this is still relevant, can probably be closed.

closed 2023-09-02 by TD-er