letscontrolit/ESPEasy #3679: P057 P057_HT16K33_LED.  Missing "Dot" Function for Adafruit 0.56"

Andy200877, 2021-06-07:
### System configuration

Hardware: ESP32

ESP Easy version: mega-20210503

I've a Adafruit 0.56" 7-segment display with HT16K33 Driver.
On this Display is working my the Plugin.
But the Plugin don't set the dot's

I want display a temperatur : 22.1° But the plugin show's 221° at the Display.
To set the Dot's at the bottom i can use : "M 4=128" but all other segments are going off. (1,2,4,8,16,32,64 are the segments and 128 the dot at bottom right of the Number.
The Display has 4 dot's at the bottom and 2 dot's for clock.

Here is a link to the display :
https://www.adafruit.com/product/1002

TD-er, 2021-06-07:
I set the label to "needs info" since I think it might not be a bug in the plugin, but maybe it could be an issue where the dot is not wired on your board.
But I've not used that board myself, so maybe better to wait for someone who has used a number of those boards to confirm (or reject) my hypothesis.

Andy200877, 2021-06-07:
The Board ist wired correct. With the "manual" M Command i can set the Dots. Only devices/values like Temperatur (22.1) is not showing correct. the dot is missing.

tonhuisman, 2021-06-09:
@Andy200877 What exact command are you using to put that temperature value on the display?

tonhuisman, 2021-06-09:
I just made a PR to fix this omission for the `mprint` command. (That seems the only way to simply set text on the display). Currently unable to test myself, as I don't have this hardware available.

@Andy200877 what .bin file did you install on your ESP32 unit? Then I'll provide that for download so you can test the fix.

Andy200877, 2021-06-09:
I've installed "ESPEasy_test_B_ESP32-wrover-kit_4M316k_ETH" on my ESP32. Big THX.

tonhuisman, 2021-06-10:
I've extracted this from the Github build, as I'm not at my home computer.
For update via OTA you can use the normal .bin file, for update via the Flash tool you will need the -factory.bin file (but I guess you know that already).
Files are in the `build_output\bin` folder of this file.

[ESPEasy_test_B_ESP32-wrover-kit_4M316k_ETH.zip](https://github.com/letscontrolit/ESPEasy/files/6628868/ESPEasy_test_B_ESP32-wrover-kit_4M316k_ETH.zip)

Andy200877, 2021-06-10:
Great ! It's working....
But i don't tested by tempertures < 0°C or over 100°C

tonhuisman, 2021-06-10:
> But i don't tested by tempertures < 0°C or over 100°C

Negative temperatures down to -99.9 should work, as the minus sign is supported, and 999.9 should be the max. that could fit on a 4 digit display.

tonhuisman, 2021-06-15:
> Great ! It's working....
> But i don't tested by tempertures < 0°C or over 100°C

Sure it's working as desired? Reviewing the code and my changes, I think the period is shown, but also a digit is skipped, leaving it empty. Don;t think that's correct. I'll fix and post an updated .bin here.

tonhuisman, 2021-06-15:
Made a local build with updated code (based on latest `mega` branch), so the dot should not result in an empty digit.

[ESP_Easy_mega_20210615_test_B_ESP32-wrover-kit_4M316k_ETH.zip](https://github.com/letscontrolit/ESPEasy/files/6658162/ESP_Easy_mega_20210615_test_B_ESP32-wrover-kit_4M316k_ETH.zip)

Pleased report your findings with this build, if possible include a photo of the working display showing a temperature with a decimal value

PR #3683 (2021-07-17): [P057] Add support for periods in display
Any period provided in `mprint` command will show on display.

Resolves #3679 

Todo:
- [x] Await response from issue-reporter
- [ ] Test
closed 2021-07-17 by TD-er