letscontrolit/ESPEasy #3686: SHT30/31/35 temperature offset feature request

tnowak, 2021-06-11:
BME280 module has a great feature setting temperature offset for the sensor.
That would be great if SHT30/31/35 [TESTING currently] module had the same.

Sensirion sensors should have been more accurate, but unfortunately I have a large difference on SHT31 in comparision to other sensors such as Aqara and Philips (~2,5 deg - even bigger than on BME280 :/ )

tonhuisman, 2021-06-12:
I've added this as a feature in a PR.

@tnowak Can you build a .bin yourself, including that PR? Or tell us what .bin you've used, and I can provide it for download here once the build completes.
Please report your test results here, as I don't own any of these SHT3x sensors.

tnowak, 2021-06-13:
I can't build it myself out of PR, but I could test out a binary.
I'm using TEST_A build for ESP8266: ESP_Easy_mega_20210503_test_A_ESP8266_4M1M

tonhuisman, 2021-06-13:
This is the zip of that build, as retrieved from the Github Actions buildservice:
(the .bin file is hidden in the `build_output\bin` folder of this zip file)

[ESPEasy_test_A_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6644438/ESPEasy_test_A_ESP8266_4M1M.zip)

tnowak, 2021-06-13:
Thanks. I've tried it and setting -20 offset (-2 degrees) put my temp sensor from 27.2 degrees to -172.2 degrees :)
So something is wrong in the calculations

tonhuisman, 2021-06-14:
I will fix later today).

tonhuisman, 2021-06-14:
Here's a fixed build I made locally. PR is also updated.

[ESP_Easy_mega_20210614_test_A_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/6650211/ESP_Easy_mega_20210614_test_A_ESP8266_4M1M.zip)

tnowak, 2021-06-14:
Works perfectly now! Thank you!

PR #3687 (2021-06-14): [P068] Add temperature offset setting, with humidity compensation
Add setting for temperature compensation, with recalculation of humidity, ~~borrowed~~ the same as used by the BME280 sensor.

Resolves #3686 

Todo:
- [x] Testing (by requester)
closed 2021-06-14 by TD-er