letscontrolit/ESPEasy #3693: Invalid concatenation API with Web_StreamingBuffer

mcspr, 2021-06-25:
Looking at the operator+= and 2 helper methods:
https://github.com/letscontrolit/ESPEasy/blob/e794fa63e69e87f0204c7dec58e0f0e19ee5b481/src/src/DataStructs/Web_StreamingBuffer.h#L38-L51
I think the intended return value should be `Web_StreamingBuffer&` and not just `Web_StreamingBuffer`, since we stay with the same object.

It seems all return values are copies / moves of the original structure, C++ is pretty happy to comply with the declaration though:
https://en.cppreference.com/w/cpp/language/operator_assignment

TD-er, 2021-06-25:
Ah that's a good point.... Will check immediately.
Very strange I did not notice it before.

TD-er, 2021-06-25:
It really does make the page serving quite a bit more responsive, so that's a big win :)
[image]
Rendering + serving a 14 kByte page in under 140 msec over WiFi on an ESP8266.

N.B. this was after a number of refreshes, to be sure the typical WiFi overhead of roughly half the beacon interval was not playing a significant role anymore.

But still, it used to be roughly 280 - 300 msec minimum for that page.

Sadly it is not a fix when running the latest staging code.

mcspr, 2021-06-25:
> Sadly it is not a fix when running the latest staging code.

Speaking of that, I can't actually test the proposed changes or try to understand the crashing part b/c it crashes on boot for whatever reason:
```
$ pio run -e custom_300_ESP8266_4M1M
$ esptool --baud 460800 --after no_reset write_flash 0 .\build_output\bin\ESP_Easy_mega_20210625_custom_300_ESP8266_4M1M.bin
$ pio device monitor --baud 115200  --echo
 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000d6b00
~ld

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4
tail 4
chksum 0xc9
csum 0xc9
v000d6b00
~ld

... etc ...
```

TD-er, 2021-06-25:
How do you flash it?
I have noticed there actually _is_ an issue with PlatformIO, where it does build the wrong pio env.

Oh and for later... there is also something fishy with the GH Actions build script as soon as I set the "beta" builds to SDK3.
Not sure if related, but since you're way more familiar with the toolchain, I guess you can maybe see if it is related and if so, maybe of help in figuring out what's wrong.

Se [failing GH actions build](https://github.com/letscontrolit/ESPEasy/runs/2906242623?check_suite_focus=true)
This was with the `[core_stage]` set to SDK3

tonhuisman, 2021-06-25:
The speed-improvement with that change is really impressive!
Great find @mcspr 👍

mcspr, 2021-06-25:
Can't reproduce either crash 🤷 For runtime, I think I need more components doing something, @TD-er could you share some config for rules / plugins / etc. so it... does something with strings all the time? Just looking at webpages does not do the trick.

GH Action points to this section of the esp8266 script:
https://github.com/esp8266/Arduino/blob/20de82588e4e9bfc8c9107f9163dda03a1c1e770/tools/platformio-build.py#L324-L329
Only reached with `-DPIO_FRAMEWORK_ARDUINO_MMU_CUSTOM'. This might happen if something modifies CPPDEFINES directly and makes an entry for `-DFLAG` like `("FLAG",)` instead of what must be either `"FLAG"` or `("FLAG", None)`, but I don't see any apparent issues with any of the scripts.

TD-er, 2021-06-25:
These are the settings I used on my test node:
- [config.dat](https://www.dropbox.com/s/rlxrdog0f6wm8of/config.dat?dl=0)
- [rules1.txt](https://www.dropbox.com/s/x9limm141vm9byr/rules1.txt?dl=0)

Not really interesting settings...
And I have some MQTT broker in my network. (hostname domoticz.lan)
Any MQTT broker will do here.

mcspr, 2021-06-25:
For debugging:
```diff
diff --git a/platformio_esp82xx_base.ini b/platformio_esp82xx_base.ini
index 2005285f..4cc2bfc4 100644
--- a/platformio_esp82xx_base.ini
+++ b/platformio_esp82xx_base.ini
@@ -51,7 +51,6 @@ lib_ignore                = ${no_ir.lib_ignore}, ${no_sd.lib_ignore}, ${no_littl
 extends                   = common
 board_build.f_cpu         = 80000000L
 build_flags               = ${debug_flags.build_flags} ${mqtt_flags.build_flags} -DHTTPCLIENT_1_1_COMPATIBLE=0
-build_unflags             = -DDEBUG_ESP_PORT
 lib_deps                  = https://github.com/TD-er/ESPEasySerial.git#v2.0.5, adafruit/Adafruit ILI9341 @ ^1.5.6, Adafruit GFX Library, LOLIN_EPD, adafruit/Adafruit BusIO @ 1.7.5, bblanchon/ArduinoJson @ ^6.17.2, VL53L0X @ 1.3.0, SparkFun VL53L1X 4m Laser Distance Sensor @ 1.2.9, td-er/RABurton ESP8266 Mutex @ ^1.0.2, td-er/SparkFun MAX1704x Fuel Gauge Arduino Library @ ^1.0.1
 lib_ignore                = ${esp82xx_defaults.lib_ignore}, IRremoteESP8266, HeatpumpIR, LittleFS(esp8266), ServoESP32
 board                     = esp12e
diff --git a/platformio_esp82xx_envs.ini b/platformio_esp82xx_envs.ini
index a4535f74..c89ad000 100644
--- a/platformio_esp82xx_envs.ini
+++ b/platformio_esp82xx_envs.ini
@@ -79,6 +79,10 @@ platform_packages         = ${beta_platform.platform_packages}
 build_flags               = ${beta_platform.build_flags}
                             ${esp8266_4M1M.build_flags}
                             -DPLUGIN_BUILD_CUSTOM
+                            -g
+                            -DDEBUG_ESP_OOM
+                            -DUMM_POISON_CHECK
+                            -DDEBUG_ESP_PORT=Serial
 lib_ignore                = ${esp8266_custom_common.lib_ignore}
 extra_scripts             = ${esp8266_custom_common.extra_scripts}

diff --git a/tools/pio/pre_custom_esp82xx.py b/tools/pio/pre_custom_esp82xx.py
index 30bbfa40..213440cb 100644
--- a/tools/pio/pre_custom_esp82xx.py
+++ b/tools/pio/pre_custom_esp82xx.py
@@ -51,6 +51,7 @@ else:

 #    "-DUSES_C016",  # Cache Controller
     "-DUSES_C018",  # TTN/RN2483
+    "-DUSES_C019",  # noticed it flagged as disabled in the config, idk if relevant or not
 #   "-DUSES_C015",  # Blynk

 #    "-DFEATURE_MDNS",
```

Immediately crashes on boot:
```
ctx: cont

sp:       0x3fff2e60
end:      0x3fff3190
offset:   0x00000190

0x4023246e: PluginCall(unsigned char, EventStruct*, String&) at src\src\Globals/Plugins.cpp:326
0x401015a8: get_poisoned at ~\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\cores\esp8266\umm_malloc/umm_poison.c:117

  \-> inlined by: get_poisoned at ~\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\cores\esp8266\umm_malloc/umm_poison.c:105
0x40101ac8: umm_malloc at ~\.platformio\packages\framework-arduinoespressif8266@src-31d658a59f41540201fc3726a1394910\cores\esp8266\umm_malloc/umm_malloc.cpp:821
0x402b7e34: ReleaseMutex at ??:?
0x40219316: std::_Rb_tree<int, std::pair<int const, TimingStats>, std::_Select1st<std::pair<int const, TimingStats> >, std::less<int>, std::allocator<std::pair<int const, TimingStats> > >::_M_insert_node(std::_Rb_tree_node_base*, std::_Rb_tree_node_base*, std::_Rb_tree_node<std::pair<int const, TimingStats> >*) at ~\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.2.0\bits/stl_tree.h:2369
  \-> inlined by: std::_Rb_tree_iterator<std::pair<int const, TimingStats> > std::_Rb_tree<int, std::pair<int const, TimingStats>, std::_Select1st<std::pair<int const, TimingStats> >, std::less<int>, std::allocator<std::pair<int const, TimingStats> > >::_M_emplace_hint_unique<std::piecewise_construct_t const&, std::tuple<int&&>, std::tuple<> >(std::_Rb_tree_const_iterator<std::pair<int const, TimingStats> >, std::piecewise_construct_t const&, std::tuple<int&&>&&, std::tuple<>&&) at ~\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.2.0\bits/stl_tree.h:2468
  \-> inlined by: std::map<int, TimingStats, std::less<int>, std::allocator<std::pair<int const, TimingStats> > >::operator[](int&&) at ~\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.2.0\bits/stl_map.h:522
0x40229d82: std::_Rb_tree_iterator<std::pair<unsigned int const, portStatusStruct> >::operator++() at ~\.platformio\packages\toolchain-xtensa\xtensa-lx106-elf\include\c++\10.2.0\bits/stl_tree.h:287 (discriminator 2)
  \-> inlined by: GPIO_Monitor10xSec() at src\src\ESPEasyCore/ESPEasyGPIO.cpp:422 (discriminator 2)
0x4023f169: run10TimesPerSecond() at src\src\Helpers/PeriodicalActions.cpp:86
0x40241301: ESPEasy_Scheduler::setIntervalTimer(ESPEasy_Scheduler::IntervalTimer_e, unsigned long) at src\src\Helpers/Scheduler.cpp:365
0x40241338: ESPEasy_Scheduler::process_interval_timer(ESPEasy_Scheduler::IntervalTimer_e, unsigned long) at src\src\Helpers/Scheduler.cpp:381
0x402413a5: ESPEasy_Scheduler::process_interval_timer(ESPEasy_Scheduler::IntervalTimer_e, unsigned long) at src\src\Helpers/Scheduler.cpp:386
0x40241a60: ESPEasy_Scheduler::handle_schedule() at src\src\Helpers/Scheduler.cpp:250
0x40241a80: ESPEasy_Scheduler::handle_schedule() at src\src\Helpers/Scheduler.cpp:253

0x4022e57d: handle_unprocessedNetworkEvents() at src\src\ESPEasyCore/ESPEasyWifi_ProcessEvent.cpp:201

0x402304b5: ESPEasy_loop() at src\src\ESPEasyCore/ESPEasy_loop.cpp:122
[... cut ...]
My only hunch was mutex, and removing it's methods seems to fix immediate crash on boot another semi-random one when accessing the controller page:
https://github.com/letscontrolit/ESPEasy/blob/bdeb1e5a35f2483a7fd07cc0d7a833c741b2de6f/lib/raburton_esp8266_mutex/src/esp8266_mutex.c#L21-L39

I am not really reading asm as a native language, but it does not seem right to use a15 reg as-is without actually knowing whether it is used by something else *outside* of the scope of the func. Plus, it does not really do anything on esp8266, since the wrapper class never used with try_lock() to actually check the mutex status.

TD-er, 2021-06-25:
OK, that sounds plausible.
It is relatively old code (not in ESPEasy, but from where I found it), so it is very well possible it might interact with the newer esp8266/Arduino core code.

It would also explain the WDT reboots I saw with the core 3.0.0 code as the lock() is a while loop.

In this code base I only use it for the WiFiAP Candidates list processing.
In a work-in-progress I do use it to process incoming ESP-NOW packets for the new ESPEasy-NOW layer I am working on.
But that code is based on core 2.7.4 and for that purpose the lock really does work well.
On the other hand, I guess on an ESP8266 it is 'good enough' to have a volatile bool (or `uint32_t` for more guaranteed atomic compare-and-set actions using a bitshift for example) instead of such elaborate assembly code messing around with interrupts.
Especially since messing with interrupts will also affect the selected heap in the 2-heap setup.

Too bad I'm away tomorrow, almost all day, since my fingers are itching to try these things out.
But it is already past midnight and if I start doing it now, I know I will see the sunrise before going to bed.

TD-er, 2021-06-26:
OK, I just did some quick coding to get rid of the fishy assembly for the mutex implementation on ESP8266.
Now it is almost 2:20 am, so I will now try to get some sleep...
Let's hope I didn't make some obvious mistakes here.

TD-er, 2021-06-28:
Well I still got crashes, so now every Plugin/CPlugin and NPlugin call makes sure to be in DRAM heap and this does seem to solve the crashes.

I also made 2 lists for scan results.
The scanned_new list will be merged into the existing one from the normal loop call.
I may still need to do some optimizations here as I'm not entirely sure if this scanned list can have duplicates now and also it is a bit fishy now how it does check if items have to be moved from the scanned_new to scanned.

TD-er, 2021-08-12:
The issues mentioned here are all resolved (as far as I can see)...

commit ef052d91 (2022-02-08): [Webserver]  Fix unneeded copy of object when serving data(#3693)

Fixes: #3693
closed 2021-08-12 by TD-er