letscontrolit/ESPEasy #3695: Sensirion SGP40

tnowak, 2021-06-29:
There is Sensirion SGP30 gas sensor support in testing (TEST A)
https://github.com/letscontrolit/ESPEasy/issues/3297

but Sensirion has already superseded it with SGP40 sensor and doesn't recommend using SGP30 in new designs:

[image]

(source: https://www.sensirion.com/en/environmental-sensors/gas-sensors/sgp30/ )

Adafruit already sells SGP40 sensor and a library for it is available:
https://github.com/adafruit/Adafruit_SGP40

Would you consider adding support for it in ESPEasy?

TD-er, 2021-06-29:
Do you already have one of these modules?
I can imagine it won't work right out of the box as the Adafruit libraries often check for a revision byte value in the chip and refuse if that's not matching.
But given it is a successor, I guess the same functionality is very similar to access. Only new functionality will require some programming.

tnowak, 2021-06-29:
I will have one most probably tomorrow.

tnowak, 2021-06-29:
Please mind there are functional changes though.

They are well documented here: https://www.sensirion.com/fileadmin/user_upload/customers/sensirion/Dokumente/9_Gas_Sensors/Sensirion_Gas_Sensors_SGP30_SGP40_transition_guide.pdf
and also described here:
https://learn.adafruit.com/adafruit-sgp40

In short, no TVOC and eCO2 measurements are provided anymore.
"Instead, raw signal from the sensor is processed using their software algorithm to give an overall 'air quality' value form 0 to 500."

tonhuisman, 2023-04-30:
@tnowak I've created a pull request to add a `Gases - SGP4x VOC(/NOx)` plugin to ESPEasy.
Would you be available to test this please, and report your findings either here or in the PR? (link is above this post)

You can download binaries from [this Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/4845131222), and you can test using either the `Collection F`, `Climate` or `MAX` (ESP32 with 16 MB Flash only) builds.

DittelHome, 2023-05-01:
Good news !!
I have tried to install the sgp40 sensor, but i cant find the sensor in the collection f,  I tried this file:
ESP_Easy_mega_20230430_collection_F_ESP8266_4M1M.bin
Right ?

EDIT:
I found it in the climate file ..
Test is going on

tonhuisman, 2023-05-01:
> cant find the sensor in the collection f,

Hm, that's my mistake, intended to include it there, but I didn't add the required code. Will fix.

DittelHome, 2023-05-01:
The sensor is working quit good, i am happy, good job !!
My suggestions:
**Use Compensation** instead of **Use Calibration** as description of the checkbox. 

Does the checkbox **Single event with all values:** makes sense, if we have only one value ?

All my test's were very good !! (also including compensation)

tonhuisman, 2023-05-01:
> Does the checkbox **Single event with all values:** makes sense, if we have only one value ?

That option is standard available for all devices, even when there is only a single value, as you _could_ want to use the #All value for all your events. Multiple values will be passed comma separated.
For this plugin it is still useful when connecting the SGP41, as that also has a `NOx` value, that will show up when switching the Sensor Type. (No need to try to use that with a SGP40 connected, it won't work 😉)

I'll take your suggestion in changing the caption for the checkbox, that's more appropriate.

tonhuisman, 2023-05-01:
Can you also test with the `Low-power measurement` option enabled?

DittelHome, 2023-05-01:
`Low-power measurement `does not work !
After i have enabled the checkbox, rebooting, no more values.

chemmex, 2023-05-01:
I seem to have SGP41, but it only worked being configured as SGP40, expectedly showing only VOC values. Switching to SGP41 gives serial number but no readings at all.

Had to build Sensirion's example sketch to check it is really SGP41 - getting raw VOC and NOx there

tonhuisman, 2023-05-01:
> Switching to SGP41 gives serial number in the log but no readings at all.

There seems to be a timing difference between SGP40 and SGP41 commands, so I may be reading the SGP41 data too soon. Working on it.

chemmex, 2023-05-01:
I found this comment in the example code:
`// During NOx conditioning (10s) SRAW NOx will remain 0`
it really reads first 10 NOx values as 0, but VOC starts reading immediately

tonhuisman, 2023-05-01:
Yes, I've used a similar approach, there is a special SGP41 'conditioning' command to send the compensation data, but only read the VOC. The regular measurement command for SGP41 returns both VOC and NOx.

tonhuisman, 2023-05-02:
> `Low-power measurement `does not work ! After i have enabled the checkbox, rebooting, no more values.
> 
> EDIT: Without LPM all is fine again

I've fixed the Low-power measurement mode, and adjusted timing for SGP41, though that last one is still untested here, so if you guys are able to re-test, then please do. TIA.
Build failed because of GH/internet-stuff, might be available later.

chemmex, 2023-05-02:
Got success with SGP41 setting, both VOC and NOx are available now. Though, NOx is steady 1 in continuous mode and 0 in low power mode. Sample sketch showed some drift in raw NOx values.

DittelHome, 2023-05-03:
Maybe it is intended, but plugin is not yet included in collection F.
Low-power measurement mode is working now !!  
Compensation/Calibration is also fixed .., nice.
* Good job *

tonhuisman, 2023-05-03:
> Got success with SGP41 setting, both VOC and NOx are available now. Though, NOx is steady 1 in continuous mode and 0 in low power mode. Sample sketch showed some drift in raw NOx values.
> 
> P.S. with Show raw data only checked, some drift is also observed

In regular mode (non-raw) the Sensirion NOxGasIndexAlgorithm is applied. Haven't investigated how that _exactly_ works, other than how to implement it in my code (and that was quite easy 😃), but that might be the reason it is rather stable.

> not yet included in collection F.

I still forgot to add it there... 😊

chemmex, 2023-05-03:
Sensirion has very trivial explanation of NOx index algorithm from which two parameters can be derived: 24h "learning time" and 1-500 index value range. From which I can presume that 0 in low power mode could be an invalid value.

The chart of first 6 hours seem to be quite coarse for NOx, definitely not far from the baseline. I think I have to wait 24 hours for it to settle, maybe it will scale to the whole range.  

[image]

DittelHome, 2023-05-04:
From my point of view (SGP40) the plugin can be integrated into the next mega release. I've been using it successfully for a fan controller for two days.

tonhuisman, 2023-05-04:
> the plugin can be integrated into the next mega release

There are still some tasks to complete, like including in Collection F, adding documentation, and some more testing.

PR #4634 (2023-05-12): [P147] Add plugin Gases - SGP4x VOC(/NOx)
Resolves #3695 

Features:
- Add plugin Gases - SGP4x VOC(/NOx), supporting both SGP40 and SGP41
- Included in `Climate` and `Collection F` builds
- Add `calc_CRC8()` helper function

Additional:
- P083: Adjust plugin name

TODO:
- [x] Implement Sensirion VOC and NOx calculation algorithms
- [x] Add documentation
- [x] Testing
  - [x] SGP40 ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/3695#issuecomment-1534631755))
  - [x] SGP41 ([confirmed](https://github.com/letscontrolit/ESPEasy/issues/3695#issuecomment-1533526415))
closed 2023-05-12 by TD-er