letscontrolit/ESPEasy #3700: ESP32 ETH - NTP not working when use Static IP

Chong83, 2021-07-08:
ESP32 Wroom-32D , Install ESPEasy_normal_ESP32_4M316k_ETH using on Lan8720, Network Woking Fine, Can ping can Connect over LAN
but NTP not working When i setup Static IP on Ethernet, only Leave empty for DHCP Working, Wifi Not Connected
i am already try few NTP server, Leave empty for Default, time.google.com, my.pool.ntp.org, all same
but when i use ethernet dhcp, ntp working fine 

Used version : mega-20210503
ESPEasy_normal_ESP32_4M316k_ETH 

not test on other version

tonhuisman, 2021-07-08:
Can you update to the 2021-06-15 release using the Web-flasher: https://td-er.nl/ESPEasy/ (select the same build-type as before) That already has a lot of networking related improvements (though I'm not sure if it solves your issue)

uzi18, 2021-07-08:
403 forbiden

tonhuisman, 2021-07-08:
> 
> 
> 403 forbidden

Depends on some excluded IP-ranges. Contact @TD-er to have your IP white-listed or you could use a VPN service to ' change' your location.

Chong83, 2021-07-11:
> Can you update to the 2021-06-15 release using the Web-flasher: https://td-er.nl/ESPEasy/ (select the same build-type as before) That already has a lot of networking related improvements (though I'm not sure if it solves your issue)

Binary Filename:⋄ | ESP_Easy_mega_20210615_normal_ESP32_4M316k_ETH

NTP Problem Fix in This Version, Thanks.

Chong83, 2021-07-11:
Test on but same
ESP_Easy_mega_20210615_normal_ESP32_4M316k_ETH
ESP_Easy_mega-20210503_normal_ESP32_4M316k_ETH

Same this is my test result

DHCP - IP Address, Subnet, DNS, Network Type All Correct & NTP Working, Network Working
[image]

Just Fill in IP Address, NTP not Working when Restart & Reconnect Power, Network Working
[image]

Fill in Subnet & DNS
Network Type Incorrect Show WiFi, Still Show DHCP, IP & DNS & Subnet 0.0.0.0
[image]

TD-er, 2021-07-11:
How do you configure NTP?
If the DNS is not set and you have not entered the NTP server as an IP address, then the NTP service will never work.

To access an IP outside your own subnet, you also need a gateway.
With only an IP set and no subnetmask, it should not be considered as valid IP settings, so that's a bug for sure.

What I find strange is that your system info is showing IP/subnet as 0.0.0.0 even when you have set IP & subnetmask.
Maybe the WiFi settings are checked here? -> Bug.

TD-er, 2021-07-12:
There is also another issue with NTP and ESP32 + Eth.
If a node is connected to Eth, it seems NTP is not run anymore and also switching timezone/DST does not update the system time accordingly.
Also one node on  my desk was running for a few days and was over a second behind, thus not synced using NTP for a while.

tonhuisman, 2021-09-26:
This issue is most likely resolved after merge of pr #3775

TD-er, 2021-09-26:
Yep, good catch. Will close it.
Let me know if there are some issues left related to this.

closed 2021-09-26 by TD-er