letscontrolit/ESPEasy #3701: [ESP32] Improve HW Multiplexing feature for energy plugin 102 - PZEM-004Tv30-Multiple [TESTING]

jmexposi, 2021-07-10:
### Checklist
- [x] The title of this issue is "to the point" and descriptive.
- [x] This issue describes what is happening.
- [x] This issue describes what components are affected (e.g. name of plugin/controller)
- [x] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x] Tested previous/other build (mention which one already tested)
- [x] Tested on other node to make sure hardware isn't defective.
- [x] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

I need (if it's possible) to use hardware multiplexing feature with Tca9548a to split 11 PZEM energy sensors & 2 BME280 temp sensors. Seems that multiplexing feature isn't implemented yet for PZEM.

### Steps to reproduce

1. Setup in harware section: activate the multiplexing functionality : Tca9548a 
2. Install plugin BME280 and setup in channel 1 of Tca9548a : works fine
3. Install plugin PZEM-004T : and "plug" it into channel 0. I2c config schema changes but the options are different and it's not possible to select multiplexer channel 0. I can't configure device and doesn't works.

### System configuration

Hardware:
ESp32
Tca9548a multiplexer
PZEM-004T energy meter

ESP Easy settings/screenshots: 

[image]
[image]
[image]

TD-er, 2021-07-10:
I don't understand the problem here.
The PZEM does need an UART port.
What you have selected is the option (as I think we also discussed [on the forum](https://www.letscontrolit.com/forum/viewtopic.php?p=53798#p53798)) of an I2C to UART chip.
The I2C multiplexer is meant for I2C devices which cannot be set to another I2C address or to create extra isolated I2C busses for example to mix 5V and 3v3 I2C devices or combine devices that don't support higher I2C clock frequencies.

Right now, we do support an abstraction layer for UARTs so all plugins that need an UART can switch between SW serial/HW serial and I2C to UART chips.
The I2C to UART chip does support upto 16 I2C addresses, so in theory that may add 32 UART ports.
I have a board I made myself which does have 3 of these chips on board adding 6 UART ports and that's working fine.
It does of course share the I2C bus, so for rather 'chatty' devices you may run into practical bandwidth issues.
On 400 kHz I2C clock I guess the upper limit is roughly 100 - 150 kbps continuous. (bursts can be higher due to 64 byte FIFO buffers in the chips per UART)

To summarize, I don't think this is really an issue.

jmexposi, 2021-07-10:
Ok understood. Following your suggestion, I have separated the I2C channel to pins 32 and 33 to use it with the BME280 and for now I will try to use the 3 HwSerial channels to control the PZEMs (mapping Hw1 to avoid conflicts) and in the future try to include UART bridge as you recommended.

In any case, if possible, this could be considered as an improvement to be able to control multiple devices separating them by different channels using a single channel of the controller.

TD-er, 2021-07-10:
> In any case, if possible, this could be considered as an improvement to be able to control multiple devices separating them by different channels using a single channel of the controller.

That's already possible, as you can select the channel (A/B), and the I2C address when chosing an I2C uart port.
And since you add it per task, it should already be separated.

But just for the PZEM, I find it rather strange it apparently cannot use several on the same UART port, as that's typically one of the benefits of using RTU Modbus.

TD-er, 2021-10-14:
See PR: #3807

I think this should perhaps also be fixed by that PR.
What happened was that the ESP32 serial port may become unavailable after `end()` was called followed by a `begin()`.
This is exactly what may happen on `PLUGIN_INIT`, thus when saving settings, but also when using multiple instances of a plugin using the same UART port, like when using Modbus RTU.

Can you try this fix?

jmexposi, 2021-10-14:
Hi,
@TD-er  This weekend  I will prepare a test environment (SC16IS752) and I report results as
much as possible. the other multiplexer that I mentioned before is: atlas scientific 8:1 Serial Port Expander (see attach)

@vortigont  Ok!, it's very interesting: I will try and give you feedback.

and I attach photo of my PZEM last project update
Many Tnks!!!
[image]
[image]

BR ,
JMER

jmexposi, 2021-10-18:
Hi @TD-er,
I build now and test but I couldn't get the CJMCU-752 to work with the PZEM (next, I will test the 752 in a separate ino sketch file):
[image]

If I try inn HW mode without the 752 works:
[image]

with 752 I2C Scan detects device:
[image]

I fetch & build latest MEGA branch:
[image]
[image]

My config.h file:
[Custom.zip](https://github.com/letscontrolit/ESPEasy/files/7368606/Custom.zip)

TD-er, 2021-10-19:
What do you mean by "couldn't get the CJMCU-752 to work with the PZEM" ?
Doesn't the I2C selection appear in the task config? Or it doesn't work with the I2C bridge?

jmexposi, 2021-10-19:
Hi, Bridge doesn't work: please, see attached images for current config.

Tnks!

TD-er, 2021-10-19:
ESP32 is not critical to which pins you're using for I2C.
Just make sure it isn't an input-only pin and not used for something else.
See also [documentation](https://espeasy.readthedocs.io/en/latest/Reference/GPIO.html#best-pins-to-use-on-esp32) for reference.
As long as you can see the I2C device, you should be fine I guess.
Maybe also add some pull-up resistors on both SDA and SCL, as it is often unknown what may be placed on those boards.
N.B. the same may apply to the serial line signals, so better also check those.