letscontrolit/ESPEasy #3703: Timestamp is not saved or read from RTC after wake up from deep sleep

chemmex, 2021-07-12:
Conditions: an EspEasy unit with ESP_Easy_mega_20210615_normal_ESP8266_1M.bin. One dummy plugin enabled. NTP off, DateTime set manually. 

After reboot by command or reset by button:

`
75 : Info   : INIT : Free RAM:34960
76 : Info   : INIT : Soft Reboot #1 - Restart Reason: Software/System restart
76 : Info   : FS   : Mounting...
82 : Info   : FS   : Mount successful, used 46686 bytes of 113201
132 : Info   : FS   : Success garbage collection
169 : Info   : FS   : Success garbage collection
177 : Info   : CRC  : SecuritySettings CRC   ...OK
180 : Info   : WiFi : Start network scan all channels
204 : Info   : INIT : Free RAM:31736
204 : Info   : INIT : I2C
205 : Info   : INIT : SPI not enabled
289 : Info   : INFO : Plugins: 47 [Normal] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
290 : Info   : EVENT: System#Wake
296 : Info   : Webserver: start
298 : Info   : Time set to 1626094809.000
299 : Info   : Current Time Zone: STD time start: 2021-10-31 03:00:00 offset: 0 min
302 : Info   : Local time: 2021-07-12 13:00:09
303 : Info   : EVENT: System#Boot
391 : Info   : EVENT: TaskInit#dummy=12,0
490 : Info   : EVENT: Time#Initialized
726 : Info   : Dummy: value 1: 0
781 : Info   : EVENT: dummy#rebootcounter=0
`

Same preconditions, deepsleep 60 seconds by command or from rules:

`
76 : Info   : INIT : Free RAM:34960
78 : Info   : INIT : Deep Sleep #1 - Restart Reason: Deep-Sleep Wake
78 : Info   : FS   : Mounting...
84 : Info   : FS   : Mount successful, used 46686 bytes of 113201
93 : Info   : CRC  : SecuritySettings CRC   ...OK
96 : Info   : WiFi : Start network scan all channels
119 : Info   : INIT : Free RAM:31848
120 : Info   : INIT : I2C
121 : Info   : INIT : SPI not enabled
206 : Info   : INFO : Plugins: 47 [Normal] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
207 : Info   : EVENT: System#Wake
214 : Info   : Webserver: start
215 : Info   : EVENT: System#Boot
294 : Info   : EVENT: TaskInit#dummy=12,0
636 : Info   : Dummy: value 1: 2
701 : Info   : EVENT: dummy#rebootcounter=2
`

Conclusion: after reboot, the unit reads timestamp from RTC and re-initializes the system clock. After deep sleep, timestamp is not read and hence no clock initialized.

TD-er, 2021-07-12:
I just browsed through the source code and found this:

```c++
void ESPEasy_time::restoreLastKnownUnixTime(unsigned long lastSysTime, uint8_t deepSleepState)
{
  static bool firstCall = true;
  if (firstCall && lastSysTime != 0 && deepSleepState != 1) {
    firstCall = false;
    timeSource = Restore_RTC_time_source;
    externalTimeSource = static_cast<double>(lastSysTime);
    // Do not add the current uptime as offset. This will be done when calling now()
  }
}
```

And where it is called:
```c++
      if (RTC.deepSleepState != 1)
      {
        node_time.restoreLastKnownUnixTime(RTC.lastSysTime, RTC.deepSleepState);
      }
```

What looks strange is this part:
`deepSleepState != 1`

But that state is set to `1` when entering deep sleep.
So it does look intentional, but I am not 100% sure why...

Only thing I can think of is that the time does not make sense if the deepsleep is longer than a few seconds.

What we can do is either store the intended deep sleep time and add it when booting.
But you can also wake a node via other means than the deep sleep timer.

chemmex, 2021-07-18:
Took some time to test, so sorry for late response. 

`deepSleepState != 1`

If this check is skipped, the timestamp is restored from RTC, so this part of the problem seems to be solved. 

My aim, though, was to make Cache Controller work in deepsleep mode, and I suspected it is not writing anything without proper timestamp. Now I finally got it working, so can adjust the time offline knowing the actual sleep time of the node.

Trying to set up Cache Controller generated a whole bunch of questions, but it is subject to separate investigation.

TD-er, 2021-07-18:
OK, I will think about a more elegant way to keep the time.
Perhaps I really should implement the external battery powered RTC clock chip, as is also requested on the forum a number of times last weeks.

closed 2021-07-22 by chemmex