letscontrolit/ESPEasy #3704: Feature Request: Display Release/Version number in Access Point screen

Wookbert, 2021-07-14:
It would be nice if ESPEasy could display the release number in the top right corner of the Access Point screen. This would be helpful if you take a previously flashed (but not yet set up) ESP device from the shelf.

TD-er, 2021-07-14:
That's actually a very useful feature.

Wookbert, 2021-07-22:
Actually... this wouldn't only be useful info in the AP screen, but also in the Config screen:

[image]

tonhuisman, 2021-07-25:
Without changing any css, this would be quite easy to achieve:

[image]

That's today's build 😄 parsed from the binary filename as shown on the Info page. Also tried the full filename, but that seems a bit too much/intruding (the green background comes for 'free' with the `div_r` class)

Wookbert, 2021-07-30:
### Another idea: Check for Update on click of release info

How about clicking that green release info label checks the installed version against the latest github release? It could open a modal dialogue:

- a) A newer release 20210730 is available. (Along with buttons **_Not now_** and **_Show on GitHub_**).
- b) You are up-to-date. (Along with button **_OK_**).

In a future release dialogue a) could take one directly to the update procedure.

tonhuisman, 2021-07-30:
I've removed that green background already (updated screenshots in the PR), as TD-er didn't like it much (same here, btw).

For checking the current version we should have an actual version in place somewhere, and currently we don't have that (the Build number like 20114 is the settings struct version only, not the build or software version). And I dislike the idea of using the 7 character git hash for that, as that's a) not sequential, and b) not pronounceable.
We could do something like 2.0.<builddate> but that is something to start a discussion on, I'd say.

Wookbert, 2021-07-30:
I actually liked that green button.

If not actually checking the version, perhaps we could at least add a link, so when clicking, a new tab with the github release page opens (giving one the chance to manually check).

TD-er, 2021-07-30:
A link to the new releases sounds quite useful.

And it's not about not willing to highlight the build version. More that the green background wasn't feeling well in the header.
It felt a bit out-of-place.
But maybe it is something that can get used to.

I'm not in favor of automatically checking for a newer version.
It is really user friendly, etc, but one of the biggest features of an Open Source project like ESPEasy is that we're trying to be as free as possible, not limiting users, etc.
Another "selling" point of ESPEasy is that we're not Google/Facebook/etc. So the user is not the product, functionality and ease of use is.
This also means we do collect as little info as possible from users.
There is a tracking system on the read-the-docs page, but that's very limited. I only know the number of views and from which country.

This also limits development as we simply don't have a clue which plugins/controllers are the most used/wanted.
But I do find the transparency and the value of "opt-in" instead of "opt-out" more important.

tonhuisman, 2021-07-30:
I've added a link to the Releases page on Github on the Build name/date.

The header (link not shown)

[image]

And the footer (link included):

[image]

Wookbert, 2021-07-30:
> I'm not in favor of automatically checking for a newer version.

By no means I'm advocating for **_automatic_** version checking (at least not now), nor automatic (or even forced) updates. Checking (in terms of comparing installed vs. available) provides just some comfort.

> ... as free as possible, not limiting users ...

Whether we would be checking for updates or simply link to the github release page as @tonhuisman has implemented now: It only happens if the user deliberately clicks the release info. I don't see how that is limiting the user.

TD-er, 2021-07-30:
[...]
> Whether we would be checking for updates or simply link to the github release page as @tonhuisman has implemented now: It only happens if the user deliberately clicks the release info. I don't see how that is limiting the user.

That's exactly how I intended to be.
It is not tracking, just providing a link which anyone would be using anyway to check for a new version.
So we're on the same track here.

Wookbert, 2021-07-30:
Can you explain to me in two sentences that fear of tracking? Not the fear, but exactly what the tracking means? Sending exactly which info to where? Trying to understand, because I might have an alternative solution.

TD-er, 2021-07-30:
Lots and lots of products have call-back enabled by default.
This can be as harmless as just providing a notification if a new version is available.
But it also provides the manufacturer some kind of statistics of how often specific features are being used, on how many devices it is being used, etc.
Still relatively harmless, but this should already be an opt-in in my opinion and not enabled by default.

Now where things can get ugly.
You don't know what is sent along with the query.
It could be something like "fetch latest version" and then locally compare it with your running version to conclude if there is a newer version available.
Or it could encapsulate all used plugins/tasks just for statistics of the one on the receiving end of that query, but not really beneficial for the end user.

Let's asume ESPEasy is hugely popular, with millions of installed units running. (one can dream, right ? ;) )
Then for an attacker it becomes interesting to intercept those queries as they may provide usable information to find an attack vector to misuse ESPEasy to get into the network or let all nodes worldwide perform a DDoS attack, etc.
All you need to do is trying to redirect the user to another URL, or try to falsify the reply from the web server.
For example, let the user believe there is a new version out there, but it has malware on board.
If this hard-coded link is pointing to a domain name which some day may become expired, and someone registers it, nasty things can happen.
Wouldn't be the first time, as only one or two weeks ago an once large CDN (Content Delivery Network) had its domain name expired and a porn site registered it. So lots and lots of news sites with old articles were suddenly serving different kind of videos along with their news items.

So that's why I'm all in favor of opt-in instead of opt-out and also not automatically let a unit "phone home".
Having a link a user needs to click is as much opt-in as you can get.
And generating some short-hand info which a user can copy to a clipboard and paste onto a forum for support is also "opt-in" (as we have on the sysinfo page)

Right now I am implementing the "fetch instructions from a pre-configured source" as a new feature, but that's only going to be usable in custom builds and not in the main build a user may download.
Just to make it "opt-in" and support deployment of large number of nodes possible, but only for specific use cases as it may require some extra knowledge.
I'm all open about it, no secrets etc. but that also means I don't want to know about "your" secrets (e.g. passwords, internal deployment structure, etc.). Because if I know it, others with less good intentions may also get to know them ;)

Wookbert, 2021-07-30:
> one or two weeks ago an once large CDN (Content Delivery Network) had its domain name expired and a porn site registered it.

Yep, saw that case in the news. 

Can't we just pull the number of the latest release from the GitHub server (or the letscontrolit.com server) and display it if higher than the installed version? (Not sure if this is any different than what you describe).

TD-er, 2021-07-30:
That then has to be done in JavaScript in the browser, since the ESP8266 doesn't have enough RAM to fetch a HTTPS call
This then would trigger all kinds of browser script concerned about "safe browsing" to send out warnings to the end user that this site tries to access other sites too.

Wookbert, 2021-07-30:
OK, understood...

**Suggestion:** Add tooltip to @tonhuisman’s release number link saying `Click to show latest version on GitHub`. I'd also use target=_blank, so it is opened in a new page/tab for easy comparison.

TD-er, 2021-07-30:
That with opening in a new tab/window is a good suggestion.
Tooltips can be useful, but I'm not entirely sure about its usability lately as lots of users use the ESPEasy web UI on touch devices like a phone. Those render tooltips a bit less useful these days.

Wookbert, 2021-07-30:
> lots of users use the ESPEasy web UI on touch devices like a phone
How can you tell? Are you, um... _tracking_ their devices? 🤪

It would never come to my mind to setup/adjust an ESPEasy via Smartphone. Takes three times as long. Using a real computer here. And would claim that most other ESPEasy and Smarthome nerds do too.

TD-er, 2021-07-30:
> How can you tell? Are you, um... _tracking_ their devices? 🤪

LOL

More a gut feeling based on how units are being setup by users.
For initial setup you need to connect to it via WiFi and that's either done via mobile or laptop with WiFi.
Also I do see quite a lot of screenshots on reported issues which are clearly taken from a mobile. Some even show the time, provider, etc. ;)

Wookbert, 2021-07-30:
Well, the tooltip via HTML anchor title tag doesn't hurt, so I'd put it in. (That’s it unless every single byte in the code counts).

tonhuisman, 2021-07-31:
> use `target=_blank`, so it is opened in a new page/tab

I had considered that but didn't add it because it's also not on the link for Let's Control It, on the left side of the footer.

> **Suggestion:** Add tooltip/title tag

That has been on my mind too, just didn't seem needed, for the reason TD-er mentioned and the link is shown when hovered on most browsers, but as I'm usually configuring them from a laptop (usually doing initial setup from a tablet as that's easier to switch WiFi connection), I'll implement both suggestions.

Wookbert, 2021-07-31:
> > use `target=_blank`, so it is opened in a new page/tab
> 
> I had considered that but didn't add it because it's also not on the link for Let's Control It, on the left side of the footer.

Understand the inconsistency fear, but the circumstances are a bit different in this scenario, as one wants to compare the currently installed version number with the latest release. Question is whether most people memorize the existing number, before clicking the link or go to the github page and then memorize the number there before going back!? IMHO opening the Github page in a separate page still adds comfort.

tonhuisman, 2021-07-31:
[image]

NB: Latest 'version' in my world is the state of the current `mega` branch, latest 'release' is available on the Releases page (or on TD-er's web-installer page, but I don't want to link that yet, until it is a structural part of the infrastructure 😃)

Wookbert, 2021-07-31:
I find the inverted tooltip hard to read. Can you change it to black or dark grey text on a yellow background #ffe848.

tonhuisman, 2021-07-31:
That is because my Windows is set to Dark theme and Chrome follows that, it will use the colors you have configured in your OS.

Wookbert, 2021-07-31:
Can’t the OS settings be overwritten through CSS?

tonhuisman, 2021-07-31:
> Can’t the OS settings be overwritten through CSS?

That will cause all kind of issues with people using high-contrast or customized themes, so best leave it to the browser to take care of that.

closed 2021-07-31 by TD-er