letscontrolit/ESPEasy #3715: MQTT subscribe topic works with only 1 word

carefulcomputer, 2021-07-25:
### Checklist
- [X ] The title of this issue is "to the point" and descriptive.
- [ X] This issue describes what is happening.
- [ X] This issue describes what components are affected (e.g. name of plugin/controller)
- [ x] This issue describes how to reproduce it.
- [X ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ X] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ X] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ X] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ X] given a list of active plugins or controllers when applicable.
- [ X] filled out all applicable fields below.

### Steps already tried...
- [ X] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ X] Tested on other node to make sure hardware isn't defective.
- [ x] Verified if the problem is limited to a single plugin/controller

### Summarize of the problem/feature request

To send commands to Espeasy (openhab controller), I set the subscribe topic to "garage/#". It works. But then I set it to something like 'home/garage/#' then it doesn't.  Even "/garage/#" doesn't work. I made sure to match the topic on pubisher side (node-red). Looks like espeasy only accepts single word without leading slash as acceptable subscription topic.

### Expected behavior

Just like publishing topic (which could be multiple words and could have leading slash), subscription topic should allow multiple words and leading slash.

### Actual behavior

Espeasy doesn't listen to MQTT messages if subscription topic has a leading slash or has subtopic (more than one word with '/' as sperator)

### Steps to reproduce

1. First setup espeasy with subscriber topic set to 'garage/#' topic using openhab controller. Send a message from node-red  using same topic and payload as 'Status,GPIO,4'. Confirm that response is received in node-red. This is to confirm test setup (network/connections etc) is setup correctly. Move on to step 2 and 3 to confirm the issue.
2.  Now change the topic in espeasy and node-red to '/garage/#'. Send same payload. There is no response. 
3.  Now change the topic in espeasy and node-red to 'home/garage/#'. Send same payload. There is no response. 

### System configuration

Hardware: Wemos D1 mini

ESP Easy version:  ESP_Easy_mega_20210503_normal_ESP8266_4M1M

ESP Easy settings/screenshots: 
[image]

tonhuisman, 2021-07-25:
What are the exact MQTT topic(s) and payload(s) you are sending?

TD-er, 2021-07-25:
Could very well be that the leading slash is not working.
If so, it is indeed a bug, but on the other hand you should not start MQTT topics with a slash anyway.

Still it should work as there are systems out there still using a leading slash.

carefulcomputer, 2021-07-26:
> What are the exact MQTT topic(s) and payload(s) you are sending?

I have posted exact topics and payload in 'Steps to reproduce'.

> If so, it is indeed a bug, but on the other hand you should not start MQTT topics with a slash anyway.

Sure. However, leading slash is supported in publish topics. So it would be inconsistent implementation. 
So as of now two issues - 
1) Slash are supported in publish but not in subscribe. They should both either support it (i vote for it) or not.
2) Sub topics are not working in subscribe.

kretzp, 2021-07-31:
[image]
This is working on actual source (mega) (with leading /)

carefulcomputer, 2021-07-31:
can you please try 

`/NonSysName/#` 

it didn't work for me. I had to use 

'NonSysName/#'

tonhuisman, 2023-09-02:
This isn't current anymore, so AFAICS this issue can be closed.

closed 2023-09-02 by TD-er

carefulcomputer, 2023-09-09:
so this has been resolved in latest version ?

tonhuisman, 2023-09-09:
Quite a lot has changed w.r.t. Controllers and MQTT-related code since this issue was reported.
If you upgrade to the latest release you can validate that. There haven't been other reports for similar issues, that's why I have suggested to close this.

Btw, MQTT topics _shouldn't_ start with a /, according to MQTT convention. Most likely your MQTT broker doesn't have any topics starting with /, that's why you don't receive any messages when subscribing to such topic.

TD-er, 2023-09-09:
Yep, a `/` indicates a next level in the topic structure.
So starting with a `/` makes no sense within MQTT, as it would imply "empty label" as first level.