letscontrolit/ESPEasy #3716: Software/System restart, Last Action before Reboot: 1 timer, id: 8

Humanoid-XXX, 2021-07-25:
### Checklist
- [ ] The title of this issue is "to the point" and descriptive.
- [ ] This issue describes what is happening.
- [ ] This issue describes what components are affected (e.g. name of plugin/controller)
- [ ] This issue describes how to reproduce it.
- [ ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [ ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ ] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [ ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [ ] given a list of active plugins or controllers when applicable.
- [ ] filled out all applicable fields below.

### Steps already tried...
- [ ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [ ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [ ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

YOUR TEXT GOES HERE

### Expected behavior

YOUR TEXT GOES HERE

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. 
2. 
3. 

### System configuration

Hardware:

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```

```

TD-er, 2021-07-25:
Can you test the build you can flash from here: [https://td-er.nl/ESPEasy/](https://td-er.nl/ESPEasy/)  (bin files can be found [here](https://td-er.nl/ESPEasy/static/) )

Humanoid-XXX, 2021-07-25:
I will try it right away and let you know how it goes!
Thanks for the quick response!

Humanoid-XXX, 2021-07-25:
> Can you test the build you can flash from here: https://td-er.nl/ESPEasy/ (bin files can be found [here](https://td-er.nl/ESPEasy/static/) )

Just tried the latest version.... still did it again, this time the reset reason was Exception... Thanks for your help

Boot:	Exception (2)
Reset Reason:	Exception
Last Action before Reboot:	1 timer, id: 8 

Build:⋄
20114  - Mega
System Libraries:⋄
ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
pygit2_not_installed
Plugin Count:⋄
77 [Normal] [Testing] [TEST_A]
Build Time:⋄
Jun 15 2021 10:31:25
Binary Filename:⋄
ESP_Easy_mega_20210615_test_A_ESP8266_4M1M
Build Platform:⋄
Linux-5.4.0-54-generic-x86_64-with-glibc2.29
Git HEAD:⋄
pygit2_not_installed

TD-er, 2021-07-25:
Is it possible to get some logs from the serial port?

Also how did you flash this build?
Via OTA, or via serial upload?
If it was serial upload, can you post the infor of the flash chip on the sysinfo page and try to flash it via serial (USB to serial probably :) )

TD-er, 2021-07-25:
I do suspect it is running out of memory.
Can you add a task with the system info plugin, trying to get the free memory logged somewhere?

Humanoid-XXX, 2021-07-26:
> Is it possible to get some logs from the serial port?
> 
> Also how did you flash this build?
> Via OTA, or via serial upload?
> If it was serial upload, can you post the infor of the flash chip on the sysinfo page and try to flash it via serial (USB to serial probably :) )

This build was flashed via OTA,
Its a bit difficult to flash this via serial upload due to its location, but i will if i need to, should I?

below is the info page 

I will log the free memory and upate you, thanks again for the help!

Unit Number: | 0
-- | --
Local Time: | 2021-07-26 05:53:57
Uptime: | 0 days 2 hours 40 minutes
Load: | 26.60% (LC=1607)
CPU Eco Mode: | false
Boot: | Manual Reboot (7)
Reset Reason: | External System
Last Action before Reboot: | Background Task
SW WD count: | 0
Memory
Free RAM: | 15568
Heap Max Free Block: | 12640
Heap Fragmentation: | 19%
Free Stack: | 3680
Network
IP Config: | Static

IP / Subnet: | 
Gateway: | 
Client IP: | 
DNS: | / (IP unset)
Allowed IP Range: | 
Connected: | 2h39m
Number Reconnects: | 0
WiFi ❔
WiFi Connection: | 802.11n (RSSI -69 dBm)
SSID: | 
Channel: | 10
Encryption Type: | WPA2/PSK
Last Disconnect Reason: | (1)
Configured SSID1: | 
Configured SSID2: |  
STA MAC: |
AP MAC: | 

WiFi Settings
Force WiFi B/G: | false
Restart WiFi Lost Conn: | false
Force WiFi No Sleep: | false
Periodical send Gratuitous ARP: | false
Connection Failure Threshold: | 0
Max WiFi TX Power: | 17.50
Current WiFi TX Power: | 14.00
WiFi Sensitivity Margin: | 3
Send With Max TX Power: | false
Extra WiFi scan loops: | 0
Periodical Scan WiFi: | false
Firmware
Build:⋄ | 20114 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | pygit2_not_installed
Plugin Count:⋄ | 77 [Normal] [Testing] [TEST_A]
Build Origin: | GitHub Actions
Build Time:⋄ | Jun 15 2021 10:31:25
Binary Filename:⋄ | ESP_Easy_mega_20210615_test_A_ESP8266_4M1M
Build Platform:⋄ | Linux-5.4.0-54-generic-x86_64-with-glibc2.29
Git HEAD:⋄ | pygit2_not_installed
System Status
Syslog Log Level: | None
Serial Log Level: | Info
Web Log Level: | None
Network Services
Network Connected: | ✔
NTP Initialized: | ✔
ESP Board
ESP Chip ID: | 2447751 (0x255987)
ESP Chip Frequency: | 80 MHz
ESP Chip Model: | ESP8266
ESP Chip Cores: | 1
ESP Board Name: | PLATFORMIO_ESP12E
Storage
Flash Chip ID: | Vendor: 0x20 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 14 daily / 20 boot
Sketch Size: | 953 kB (2116 kB free)
Max. OTA Sketch Size: | 1019 kB (1044464 bytes)
OTA possible: | true
OTA 2-step Needed: | false
SPIFFS Size: | 934 kB (844 kB free)

TD-er, 2021-07-26:
You did update  via OTA, but what was the previous version that was installed?
Or to be precise, what was the version that was last flashed via serial? (even if it was Tasmota or other fw)

The reason I ask is that I have recently discovered that some units (no idea what exactly causes it) do behave strange when the bootloader part isn't updated but the rest of the firmware is.
With OTA update not all is updated, but via serial it is (starting at sector 0)

As far as I have narrowed it down now is that it may cause issues if the previous version was using esp8266/Arduino core 2.6.x or older and now we're running core 2.7.4.
Typical symptoms are:
- Difficulty connecting to WiFi
- Random crashes
- Continuous quick reboot loop

The unit where this was happening were running fine on some firmware builds and erratic on other builds with hardly any changes between them, but they all started running fine when updated over serial.

Not saying it is the cause here, but since these crashes are not reported by others lately, it might be.

One of the changes between core 2.6.x and 2.7.0 was how the internals of the flash chips are addressed.
For example for some chips the power level to erase sectors was changed, so really low level stuff.
I have the feeling these random crashes may be the result of simple read errors, where the flash wasn't programmed optimally due to the older code running in the bootloader which isn't updated via OTA. (and thus differs from the running code)

@tonhuisman Maybe also good for you to know this, to try and flash via serial first when running into strange crashes.

Humanoid-XXX, 2021-07-27:
> You did update via OTA, but what was the previous version that was installed?
> Or to be precise, what was the version that was last flashed via serial? (even if it was Tasmota or other fw)
>
> The [...]

Ok so I removed the ESP's from their prototype board (to access the usb port) and Uploaded ESP_Easy_mega_20210615_test_A_ESP8266_4M1M directly using the ESP Flasher tool.
I am still waiting to see if it reboots/freezes.

In the meanwhile i looked at the serial weblog when the esp wasnt connected to anything and it also kept rebooting... not sure if that means anything. but here are some of the logs it captured.

`sl l£₧| ─lα|♥♦♦♦î♀lΣ♦c<çé♥Σ←¢rôcä♦cî≤'nƒlgg£π∞♦c∟x∞Ädsl{d8≤oα►♥♀♀â♦l♀î£♦♦♀c♀gπ|♥∞l╟♦Åcî≥g'τ $îçl ♥ÿ‼‼no♀l`♥♫♥o{çÆôn♦♀b♥l`♥☼{Æô'♦♦c☻l ♥£c♀♀╟∟☻¢$;Æ`♥ⁿég£♥¬U76 : Info   :

INIT : Booting version: pygit2_not_installed (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
77 : Info   : INIT : Free RAM:32912
79 : Info   : INIT : Manual Reboot #19 - Restart Reason: External System
80 : Info   : FS   : Mounting...
105 : Info   : FS   : Mount successful, used 76053 bytes of 957314
122 : Info   : CRC  : SecuritySettings CRC   ...OK
126 : Info   : WiFi : Start network scan all channels
242 : Info   : INIT : Free RAM:29424
243 : Info   : INIT : I2C
244 : Info   : INIT : SPI not enabled
271 : Info   : MCP INIT=-1
275 : Info   : MCP INIT=-1
278 : Info   : MCP INIT=-1
284 : Info   : MHZ19: Init OK
518 : Info   : INFO : Plugins: 77 [Normal] [Testing] [TEST_A] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
519 : Info   : EVENT: System#Wake
557 : Info   : Webserver: start
558 : Info   : Time set to 1627357236.000
560 : Info   : Current Time Zone: STD time start: 2021-10-31 03:00:00 offset: 0 min
563 : Info   : Local time: 2021-07-27 03:40:36
564 : Info   : EVENT: System#Boot
568 : Info   : ACT  : Gpio,16,0
571 : Info   : GPIO : port#16: set to 0
573 : Info   : ACT  : Gpio,16,1
574 : Info   : GPIO : port#16: set to 1
576 : Info   : ACT  : PWM,13,975
578 : Info   : PWM  : GPIO: 13 duty: 975
580 : Info   : ACT  : mcpgpio,1,0
583 : Info   : MCP : port#1: set to -1
587 : Info   : ACT  : mcpgpio,2,0
590 : Info   : MCP : port#2: set to -1
592 : Info   : ACT  : mcpgpio,4,0
594 : Info   : MCP : port#4: set to -1
624 : Info   : MCP   : Port=7 State=-1.00
639 : Info   : MCP   : Port=3 State=-1.00
655 : Info   : MCP   : Port=8 State=-1.00
672 : Info   : SHT3x: Temperature: nan
673 : Info   : SHT3x: Humidity: nan
674 : Error  : Calculate: Unknown token
1013 : Info   : MHZ19: Unknown response: 0 0 0 0 0 0 0 0 0
1040 : Info   : SYS  : 0.00,1.00,100.00,3360.00
1060 : Info   : SYS  : 19752.00
1077 : Info   : MCP  : Port=7 is offline (EVENT= -1)
1081 : Info   : MCP  : Port=3 is offline (EVENT= -1)
1089 : Info   : EVENT: TaskInit#Ventilation_Fan_Status=1,1
1202 : Info   : MCP  : Port=7 is offline (EVENT= -1)
1207 : Info   : MCP  : Port=3 is offline (EVENT= -1)
1212 : Info   : EVENT: TaskInit#Humidifier_Status=2,1
1335 : Info   : MCP  : Port=7 is offline (EVENT= -1)
1340 : Info   : MCP  : Port=3 is offline (EVENT= -1)
1344 : Info   : EVENT: TaskInit#FAN Status=3,1
1414 : Info   : EVENT: Clock#Time=Tue,03:40
1421 : Info   : BMx280 : Unable to detect chip ID (255, failed)
1422 : Info   : BMx280 : Unable to detect chip ID (255, failed)
1435 : Info   : MCP  : Port=7 is offline (EVENT= -1)
1439 : Info   : MCP  : Port=3 is offline (EVENT= -1)
1444 : Info   : EVENT: TaskInit#Fruiting 1 Climate 1 SHT35=4,1
1513 : Info   : MCP  : Port=7 is offline (EVENT= -1)
[... 820 lines cut ...]
v2843a5ac
~ld
¬U76 : Info   :

INIT : Booting version: pygit2_not_installed (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
77 : Info   : INIT : Free RAM:32912
78 : Info   : INIT : Exception #21 - Restart Reason: Exception
79 : Info   : FS   : Mounting...
104 : Info   : FS   : Mount successful, used 76053 bytes of 957314
122 : Info   : CRC  : SecuritySettings CRC   ...OK
125 : Info   : WiFi : Start network scan all channels
242 : Info   : INIT : Free RAM:29424
242 : Info   : INIT : I2C
243 : Info   : INIT : SPI not enabled
270 : Info   : MCP INIT=-1
274 : Info   : MCP INIT=-1
277 : Info   : MCP INIT=-1
282 : Info   : MHZ19: Init OK
517 : Info   : INFO : Plugins: 77 [Normal] [Testing] [TEST_A] (ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support)
519 : Info   : EVENT: System#Wake
557 : Info   : Webserver: start
559 : Info   : Time set to 1627357262.000
560 : Info   : Current Time Zone: STD time start: 2021-10-31 03:00:00 offset: 0 min
563 : Info   : Local time: 2021-07-27 03:41:02
564 : Info   : EVENT: System#Boot`

tonhuisman, 2021-07-27:
Seeing the `SYS` values (Free memory I assume) drop by that kind of figures, it looks like you have a loop in your rules, can you share your rules?

Humanoid-XXX, 2021-07-27:
The rules are below. on System#Boot do Gpio,16,0Gpio,16,1PWM,13,975mcpgpio,1,0 // Fruiting 1 Ventilation Out Onmcpgpio,2,0 // Humidifier UV Sterilizermcpgpio,4,0 // Lights Onendon From: Ton HuismanSent: Monday, July 26, 2021 11:46 PMTo: letscontrolit/ESPEasyCc: Humanoid-XXX; AuthorSubject: Re: [letscontrolit/ESPEasy] Software/System restart, Last Action before Reboot: 1 timer, id: 8 (#3716) Seeing the SYS values (Free memory I assume) drop by that kind of figures, it looks like you have a loop in your rules, can you share your rules?

Humanoid-XXX, 2021-07-27:
@TD-er @tonhuisman 

Ok after reflashing directly through usb... it is still restarting .... this time with a different reason: Hardware Watchdog

<html>
<body>

Unit Number: | 0
-- | --
Local Time: | 2021-07-27 07:21:05
Uptime: | 0 days 0 hours 3 minutes
Load: | 25.24% (LC=1518)
CPU Eco Mode: | false
Boot: | External Watchdog (6)
Reset Reason: | Hardware Watchdog
Last Action before Reboot: | 3 timer, id: 5
SW WD count: | 0
Memory
Free RAM: | 15656
Heap Max Free Block: | 12432
Heap Fragmentation: | 20%
Free Stack: | 3680
Network
IP Config: | Static
Connected: | 3 m 24 s
Number Reconnects: | 0
WiFi ❔
WiFi Connection: | 802.11n (RSSI -68 dBm)
Channel: | 11
Encryption Type: | WPA2/PSK
Last Disconnect Reason: | (1)
Configured SSID2: |  

WiFi Settings
Force WiFi B/G: | false
Restart WiFi Lost Conn: | false
Force WiFi No Sleep: | false
Periodical send Gratuitous ARP: | false
Connection Failure Threshold: | 0
Max WiFi TX Power: | 17.50
Current WiFi TX Power: | 14.00
WiFi Sensitivity Margin: | 3
Send With Max TX Power: | false
Extra WiFi scan loops: | 0
Periodical Scan WiFi: | false
Firmware
Build:⋄ | 20114 - Mega
System Libraries:⋄ | ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄ | pygit2_not_installed
Plugin Count:⋄ | 77 [Normal] [Testing] [TEST_A]
Build Origin: | GitHub Actions
Build Time:⋄ | Jun 15 2021 10:31:25
Binary Filename:⋄ | ESP_Easy_mega_20210615_test_A_ESP8266_4M1M
Build Platform:⋄ | Linux-5.4.0-54-generic-x86_64-with-glibc2.29
Git HEAD:⋄ | pygit2_not_installed
System Status
Syslog Log Level: | None
Serial Log Level: | Info
Web Log Level: | None
Network Services
Network Connected: | ✔
NTP Initialized: | ✔
ESP Board
ESP Chip ID: | 2447751 (0x255987)
ESP Chip Frequency: | 80 MHz
ESP Chip Model: | ESP8266
ESP Chip Cores: | 1
ESP Board Name: | PLATFORMIO_ESP12E
Storage
Flash Chip ID: | Vendor: 0x20 Device: 0x4016
Flash Chip Real Size: | 4096 kB
Flash IDE Size: | 4096 kB
Flash IDE Speed: | 40 MHz
Flash IDE Mode: | DOUT
Flash Writes: | 2 daily / 4 boot
Sketch Size: | 953 kB (2116 kB free)
Max. OTA Sketch Size: | 1019 kB (1044464 bytes)
OTA possible: | true
OTA 2-step Needed: | false
SPIFFS Size: | 934 kB (844 kB free)

</body>
</html>

tonhuisman, 2021-07-27:
Next methodical approach could be to disable all plugins, enable 1 and test for a while, and repeat enabling 1 and testing until you find the culprit. It seems one of the plugins is eating the memory or otherwise going haywire.

Humanoid-XXX, 2021-07-27:
It has been reliable when i used the non test build before i started using the SHT35.
However, i need to use the SHT35.
I have been tring to make a custom build with just the plugins that i use, PWM, MCP23017, BME280, SHT35, MHZ-19 but i cant figure out why platformio will not build without errors,

thank you for the help

`> Executing task in folder source: C:\Users\HumanoidX\.platformio\penv\Scripts\platformio.exe run --environment custom_ESP8266_4M1M <

Error: Traceback (most recent call last):
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\click\core.py", line 1256, in invoke
    Command.invoke(self, ctx)
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\platformio\__main__.py", line 63, in cli
    maintenance.on_platformio_start(ctx, force, caller)
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\platformio\maintenance.py", line 48, in on_platformio_start
    telemetry.on_command()
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 292, in on_command
    resend_backuped_reports()
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\platformio\telemetry.py", line 443, in resend_backuped_reports
    tm = app.get_state_item("telemetry", {})
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\platformio\app.py", line 191, in get_state_item
    with State() as state:
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\platformio\app.py", line 94, in __init__
    self.path = join(get_project_core_dir(), "appstate.json")
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\platformio\project\helpers.py", line 52, in get_project_core_dir
    ).get_optional_dir("core", exists=True)
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\platformio\project\config.py", line 389, in get_optional_dir
    result = self._get_core_dir(exists)
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\platformio\project\config.py", line 380, in _get_core_dir
    raise e
  File "c:\users\humanoidx\.platformio\penv\lib\site-packages\platformio\project\config.py", line 374, in _get_core_dir
    os.makedirs(core_dir)
  File "C:\Users\HumanoidX\.platformio\python3\lib\os.py", line 223, in makedirs
    mkdir(name, mode)
FileNotFoundError: [WinError 3] The system cannot find the path specified: ''`

Humanoid-XXX, 2021-07-27:
Maybe its important to note that they do often restart but also often will just freeze up and have to be manually reset.

TD-er, 2021-07-27:
It looks like you might have some issues with I2C as I see a lot of reports of sensors that cannot be found.
Can you test by running an I2C scan?
If the bus is in a locked state, you will see an error on all I2C addresses.
Not sure what all plugins may do if the sensor doesn't reply in an expected way.

Humanoid-XXX, 2021-07-27:
> It looks like you might have some issues with I2C as I see a lot of reports of sensors that cannot be found.
> Can you test by running an I2C scan?
> If the bus is in a locked state, you will see an [...]

That log was just when I had the ESP unplugged from the sensors to update via serial. when its plugged into the sensors i dont have access to the usb port.

So I disabled the SHT3X plugin on the latest build and the ESP works fine, no reboots.

I tried the ESP_Easy_mega_20200913_test_ESP8266_4M1M_VCC build and surprisingly it worked all night without issue,

So atleast I found a temporary solution for me..... but still on 2 different esp8266's the newest builds do not work with the SHT3X pluggin. Let me know if there is anything i can do to help.

TD-er, 2021-07-27:
Ah that's a great hint of where to look for recent changes...
or how you may call "last year of changes", maybe "recent" isn't the best term here.