letscontrolit/ESPEasy #3717: Add ULN2300 Device Stepper Motor

JeedHome44, 2021-07-25:
Hello,

I don't find ULN2003 in Device to control stepper motor with 4 GPIO.

We can control it with arduino. Why not with ESP Easy?

We need to know the 4 GPIO(12,13,14,15) , Speed(1-100) , Totaly step: 64x64 and clockwise/counterclockwise.

Can you add it in ESP Easy?

TD-er, 2021-07-26:
There are already other stepper motor drivers supported.

For example: [Adafruit Motorshield v2](https://espeasy.readthedocs.io/en/latest/Plugin/P048.html?highlight=stepper)

JeedHome44, 2021-07-26:
No I need 360 motor.
I need ULN2003

closed 2022-12-21 by JeedHome44

uzi18, 2024-07-17:
@TD-er Also see lack of stepper plugin. Yes you can emulate it in rules but it is not the right way.
Diving deeper in P048->Adafruit_StepperMotor it looks like it should be possible to add fallback to drive directly gpios from this lib.

https://github.com/letscontrolit/ESPEasy/blob/3d7f299edad9d17ff61b5c7680999f1e6dd84ee2/lib/Adafruit_Motor_Shield_V2/Adafruit_MotorShield.cpp#L465-L520