letscontrolit/ESPEasy #3719: bin files missing from the latest release

mrleongalaxyum, 2021-07-26:
bin folder inside zip is empty for some builds, for example ESPEasy_minimal_core_274_ESP8285_1M_OTA_FHEM_HA.zip

https://github.com/letscontrolit/ESPEasy/releases/tag/mega-20210503

TD-er, 2021-07-26:
The build script does check for the actual build size and if it doesn't fit it will not be included.

This happens especially on "minimal OTA" builds like the one you mentioned.

mrleongalaxyum, 2021-07-27:
So there is no way to OTA update to the new version since the limit is 600kB? Does this mean that upcoming releases also won't have minimal-OTA builds included?

TD-er, 2021-07-27:
Well I won't say it is final, as I have been able to get the build size down a few times before.
Also the builds tend to be slightly smaller if I build them in Windows (not sure why though) and I still plan on updating the 2-step OTA bin to support compressed bin files.

cybermcm, 2021-07-27:
Another similar question, the latest mega build is 20210503, on the new flashing page it is 20210615. Will there be a full build package?
The new web page flash doesn't work in my case (initializing, but then nothing). I like the possibility to flash via web interface, some of my sensors are not so easy to reach....

TD-er, 2021-07-27:
The binary files of that build you can find in the [https://td-er.nl/ESPEasy/static/](https://td-er.nl/ESPEasy/static/) folder.
I plan on making a new build soon, but there are still some issues with the WiFi I wanted to get fixed.
So that's why I haven't made a new build the last few weeks.

cybermcm, 2021-07-28:
@TD-er thx for your fast reply and your hard work on ESPEasy. I'd be great if you can sort out Wi-Fi issues, maybe related to https://www.letscontrolit.com/forum/viewtopic.php?t=5372?

TD-er, 2021-07-28:
Yep, among other reports and issues I found myself.

mrleongalaxyum, 2022-10-04:
Are there any new releases for ESP8285?

tonhuisman, 2022-10-04:
The ESP8266 releases recognize when running on an ESP8285 and act accordingly, so there are no longer separate releases for that chip type.

TD-er, 2022-10-04:
See also the readme included in the ZIP: https://github.com/letscontrolit/ESPEasy/tree/mega/dist

closed 2023-03-31 by mrleongalaxyum