letscontrolit/ESPEasy #3723: Unit Names not checked/forced to be compliant with RFC Hostnames – but should be!

Wookbert, 2021-07-29:
### Ensure that Unit Name entry is RFC compliant
The field `Config > Unit Name` (which actually is the ESPEasy unit’s hostname), currently allows _any_ character to be entered.

This leads to the problem that the network-published hostname has little to do with the actual given name. For instance:

[image]

... turns (here as displayed in a Fritzbox router) into 

[image]

This is problematic, when one intends to reach the ESPEasy unit via hostname instead of IP address, and when assigning a fixed IP address in the router. (The ESPEasy provided hostname can be overwritten in the router for such an assignment with a RFC compliant spelling, but then you end up with different hostnames in the ESPEasy and the router).

**_Solution:_** The firmware should either suppress non-compliant characters in the first place or – if that isn't technically possible –, at least check the entry and display a warning, as the Fritzbox does:

[image]

(Note that when you suppress non-compliant keyboard entries in the first place, one should add a tiny subline to the field saying _„Only letters (a-z, A-Z), numerals (0-9), and hyphens (-). Not numerals only. No hyphen at beginning or end.“_ Otherwise NOOBs or occasional users might be confused.

### Rename „Unit name“ to „Unit’s Hostname“
As it doesn’t become clear to the novice user, that the Unit Name is actually the published hostname. Plus: Two fields down, the wording is currently inconsistent, as it says _„Append Unit Number to hostname:“_ (instead of _„... to Unit Name“_).

Our suggestion therefore would be to do a little renaming for better clarity:
_[image]_

TD-er, 2021-07-29:
There is already a function to replace some characters, like the underscore, when generating the hostname.  This function should be extended with the suggestions in the Fritzbox screenshot.
But I guess it would make things a lot more clear if the actual hostname is also displayed on the page as a note.

Have to check whether the literal text "Unit name" is also used elsewhere, like in JSON, or else things may break when changing the text.

tonhuisman, 2021-07-29:
And then there is the issue that somewhere in this trajectory the unicode `ü` is flattened to a 2-byte sequence..., not sure if that's caused by ESPEasy or not accepted by FritzOS.
ESPEasy at least allows underscores (`_`) in the host- and task-names (but doesn't allow dashes `-` in task-names), but I think that is handled in that transformation TD-er mentions. The check on the task-names is not included in many builds, as it is inhibited by the `LIMIT_BUILD_SIZE` compile-time option, though it should be available in any normal, custom or max build.

TD-er, 2021-08-06:
Working on a fix for this.
Each Unicode char is replaced by a single hyphen and all other rules from the Fritzbox screenshot are also checked:
[image]

PR #3744 (2021-08-06): [Hostname] Make generated hostname RFC952 compliant (#3723)
Fixes: #3723
commit 593899d0 (2021-08-06): [Hostname] Make generated hostname RFC952 compliant (#3723)

Fixes: #3723
closed 2021-08-06 by TD-er