letscontrolit/ESPEasy #3728: Bug? JSON Wifi Encryption info in 20210615 release faulty

Wookbert, 2021-08-01:
I’m running ESP_Easy_mega_20210615_normal_ESP8266_4M1M (downloaded from https://td-er.nl/ESPEasy/) here, I noticed that there is an error in the JSON output, which isn't there in release 20210503:

[image]

Encryption Type says `–` but should say `""`:

[image]

For comparison – here’s what it looks in release 20210503:
[image]

Wookbert, 2021-08-01:
Actually not only the encryption is buggy, but also the info on the encryption type is missing. Another unit which strangely also runs 20210615 shows the correct info:
[image]

I don't see any difference in the Wifi setup between those two units.

Wookbert, 2021-08-01:
Next find: Rebooting the „faulty“ 20210615 unit fixed the issue. It now show `"WPA2/PSK"` instead of just `-`. Any idea what’s going on?

TD-er, 2021-08-01:
The "-" is fixed in the latest sources.
What was happening here is that when the unit rebooted and reconnected without the need for a new WiFi scan, it is unknown what the encryption type is.
But this did expose a bug in the check to see if something is a numerical (which is used in the generation of JSON, to see if a value needs to be wrapped in quotes)
The bug is that `-` is a valid begin of a number, but if there isn't any following digit, it still should be rejected.

TD-er, 2021-08-06:
This should now be fixed, so I close this issue

closed 2021-08-06 by TD-er