letscontrolit/ESPEasy #3731: Can not compile in Atom/Platformio

l1t7l3ph0o7, 2021-08-01:
I've tried to compile in Arduino but couldn't get that working so I shifted to Atom with PlatformIO.
I Followed along this tutorial: https://www.letscontrolit.com/wiki/index.php/Tutorial_building_and_uploading_with_platformio
I also followed along with the three tutorials listed at the top of that tutorial to set everything up.

problem is when I get to the Building the correct version part in the tutorial.
I hit f7 and select "PIO Build" then I get back this error:

`
Processing custom_ESP8266_4M1M (platform: espressif8266@2.6.3; board: esp12e; framework: arduino)
--------------------------------------------------------------------------------
Tool Manager: Installing git+https://github.com/esp8266/Arduino.git#2.7.4
Error: Traceback (most recent call last):
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\__main__.py", line 109, in main
    cli()  # pylint: disable=no-value-for-parameter
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\click\core.py", line 829, in __call__
    return self.main(*args, **kwargs)
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\click\core.py", line 782, in main
    rv = self.invoke(ctx)
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\commands\__init__.py", line 44, in invoke
    return super(PlatformioCLI, self).invoke(ctx)
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\click\core.py", line 1259, in invoke
    return _process_result(sub_ctx.command.invoke(sub_ctx))
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\click\core.py", line 1066, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\click\core.py", line 610, in invoke
    return callback(*args, **kwargs)
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\click\decorators.py", line 21, in new_func
    return f(get_current_context(), *args, **kwargs)
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 143, in cli
    is_test_running,
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\commands\run\command.py", line 175, in process_env
    result = {"env": name, "duration": time(), "succeeded": ep.process()}
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\commands\run\processor.py", line 79, in process
    result = p.run(build_vars, build_targets, self.silent, self.verbose, self.jobs)
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\platform\_run.py", line 60, in run
    self.autoinstall_runtime_packages()
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\platform\_packages.py", line 74, in autoinstall_runtime_packages
    self.pm.install(self.get_package_spec(name))
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 49, in install
    spec, silent=silent, skip_dependencies=skip_dependencies, force=force
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 97, in _install
    pkg = self.install_from_url(spec.url, spec, silent=silent)
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\package\manager\_install.py", line 141, in install_from_url
    vcs = VCSClientFactory.new(tmp_dir, url)
File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 55, in new
    src_dir, remote_url, tag, silent
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 137, in __init__
    self.configure()
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\package\vcsclient.py", line 146, in configure
    result = proc.exec_command([cls.command, "--exec-path"])
  File "C:\Users\Jharris\.platformio\penv\lib\site-packages\platformio\proc.py", line 117, in exec_command
    p = subprocess.Popen(*args, **kwargs)
  File "C:\Users\Jharris\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 775, in __init__
    restore_signals, start_new_session)
  File "C:\Users\Jharris\AppData\Local\Programs\Python\Python37\lib\subprocess.py", line 1178, in _execute_child
    startupinfo)
FileNotFoundError: [WinError 2] The system cannot find the file specified
 
============================================================
 
An unexpected error occurred. Further steps:
 
* Verify that you have the latest version of PlatformIO using
  `pip install -U platformio` command
 
* Try to find answer in FAQ Troubleshooting section
  https://docs.platformio.org/page/faq.html
 
* Report this problem to the developers
  https://github.com/platformio/platformio-core/issues
 
============================================================
`

TD-er, 2021-08-02:
Please have a look at [this video](https://www.youtube.com/watch?v=ArqwMcYSMsU&t=2s&pp=sAQA) to install PlatformIO along with VS Studio Code.
Also please don't use the wiki as it is quite outdated.
Please use [ReadTheDocs](https://espeasy.readthedocs.io/en/latest/)

Another pointers which may cause lots of debugging to find:
- Don't use spaces in the path where you keep your source files
- Make sure you have some command line Git tools installed (as is also mentioned in both links I showed earlier)

tonhuisman, 2021-08-02:
> Please have a look at [this video](https://www.youtube.com/watch?v=ArqwMcYSMsU&t=2s&pp=sAQA) to install PlatformIO along with VS Studio Code.

First thing PlatformIO IDE installed in (a freshly installed) Atom did is suggest to install VSCode with the PlatformIO plugin 🤣 🤣 🤣

TD-er, 2021-08-02:
And to be honest, once you have used it in VS code, you don't want to go back to Atom.
I did use Atom for over a year, maybe close to 2 years, but now I surely won't go back.
For example the intellisense so you can right-click on a function call to navigate to its implementation or declaration is such a time saver.

My typical desktop layout using VS-code:
[image]

(stretched over 2x 4k monitor)

tonhuisman, 2021-08-02:
I don't have that kind of screen real-estate available, but it looks very cool 😎 
So it's probably time we update the documentation, no longer suggesting the use of Atom, but only VSCode 😉 (as Arduino IDE is now feeling _really, very outdated_).

l1t7l3ph0o7, 2021-08-02:
OH Thank you so much, will get right on it.

l1t7l3ph0o7, 2021-08-02:
OK I've got platformIO setup in VSCode, everything looks to be working correctly (on the whole vs code/ platformIO side of things), except it still won't compile, I'm attempting to build with an added library from the playground (which seems to be generating all the errors in the build process). 

anyway, this is what I get back in the terminal when I attempt to build:
`
C:/Users/Jharris/Desktop/espeasy/ESPEasy-mega/src/_P152_MCP42010.ino: In function 'boolean Plugin_152(byte, EventStruct*, String&)':
C:/Users/Jharris/Desktop/espeasy/ESPEasy-mega/src/_P152_MCP42010.ino:36:37: error: 'SENSOR_TYPE_DUAL' was not declared in this scope
         Device[deviceCount].VType = SENSOR_TYPE_DUAL;
                                     ^
C:/Users/Jharris/Desktop/espeasy/ESPEasy-mega/src/_P152_MCP42010.ino:68:119: error: no matching function for call to 'addPinSelect(bool, const char [15], int&)'
         addPinSelect(false, "taskdevicepin1", Plugin_152_pin[0] = Settings.TaskDevicePluginConfig[event->TaskIndex][0]);
                                                                                                                       ^
C:/Users/Jharris/Desktop/espeasy/ESPEasy-mega/src/_P152_MCP42010.ino:68:119: note: candidates are:
In file included from src\_Plugin_Helper.h:44:0,
                 from C:/Users/Jharris/Desktop/espeasy/ESPEasy-mega/src/ESPEasy.ino:102:
src\src/WebServer/Markup.h:296:6: note: void addPinSelect(PinSelectPurpose, const __FlashStringHelper*, int)
 void addPinSelect(PinSelectPurpose           purpose,
      ^
src\src/WebServer/Markup.h:296:6: note:   no known conversion for argument 1 from 'bool' to 'PinSelectPurpose'
src\src/WebServer/Markup.h:299:6: note: void addPinSelect(PinSelectPurpose, const String&, int)
 void addPinSelect(PinSelectPurpose purpose,
      ^
src\src/WebServer/Markup.h:299:6: note:   no known conversion for argument 1 from 'bool' to 'PinSelectPurpose'
C:/Users/Jharris/Desktop/espeasy/ESPEasy-mega/src/_P152_MCP42010.ino:70:119: error: no matching function for call to 'addPinSelect(bool, const char [15], int&)'
         addPinSelect(false, "taskdevicepin2", Plugin_152_pin[1] = Settings.TaskDevicePluginConfig[event->TaskIndex][1]);
                                                                                                                       ^
C:/Users/Jharris/Desktop/espeasy/ESPEasy-mega/src/_P152_MCP42010.ino:70:119: note: candidates are:
In file included from src\_Plugin_Helper.h:44:0,
                 from C:/Users/Jharris/Desktop/espeasy/ESPEasy-mega/src/ESPEasy.ino:102:
src\src/WebServer/Markup.h:296:6: note: void addPinSelect(PinSelectPurpose, const __FlashStringHelper*, int)
 void addPinSelect(PinSelectPurpose           purpose,
      ^
src\src/WebServer/Markup.h:296:6: note:   no known conversion for argument 1 from 'bool' to 'PinSelectPurpose'
src\src/WebServer/Markup.h:299:6: note: void addPinSelect(PinSelectPurpose, const String&, int)
 void addPinSelect(PinSelectPurpose purpose,
      ^
src\src/WebServer/Markup.h:299:6: note:   no known conversion for argument 1 from 'bool' to 'PinSelectPurpose'
C:/Users/Jharris/Desktop/espeasy/ESPEasy-mega/src/_P152_MCP42010.ino:72:119: error: no matching function for call to 'addPinSelect(bool, const char [15], int&)'
         addPinSelect(false, "taskdevicepin3", Plugin_152_pin[2] = Settings.TaskDevicePluginConfig[event->TaskIndex][2]);
                                                                                                                       ^
C:/Users/Jharris/Desktop/espeasy/ESPEasy-mega/src/_P152_MCP42010.ino:72:119: note: candidates are:
In file included from src\_Plugin_Helper.h:44:0,
                 from C:/Users/Jharris/Desktop/espeasy/ESPEasy-mega/src/ESPEasy.ino:102:
src\src/WebServer/Markup.h:296:6: note: void addPinSelect(PinSelectPurpose, const __FlashStringHelper*, int)
 void addPinSelect(PinSelectPurpose           purpose,
      ^
src\src/WebServer/Markup.h:296:6: note:   no known conversion for argument 1 from 'bool' to 'PinSelectPurpose'
src\src/WebServer/Markup.h:299:6: note: void addPinSelect(PinSelectPurpose, const String&, int)
 void addPinSelect(PinSelectPurpose purpose,
      ^
src\src/WebServer/Markup.h:299:6: note:   no known conversion for argument 1 from 'bool' to 'PinSelectPurpose'
C:/Users/Jharris/Desktop/espeasy/ESPEasy-mega/src/_P152_MCP42010.ino:82:26: error: 'WebServer' was not declared in this scope
         String plugin2 = WebServer.arg("taskdevicepin1");
                          ^
*** [.pio\build\normal_ESP8266_4M1M\src\ESPEasy.ino.cpp.o] Error 1
====================================================== [FAILED] Took 137.29 seconds ======================================================

Environment          Status    Duration
-------------------  --------  ------------
normal_ESP8266_4M1M  FAILED    00:02:17.294
================================================== 1 failed, 0 succeeded in 00:02:17.294 ==================================================
The terminal process "C:\Users\Jharris\.platformio\penv\Scripts\platformio.exe 'run', '--environment', 'normal_ESP8266_4M1M'" terminated with exit code: 1.

Terminal will be reused by tasks, press any key to close it.
`

TD-er, 2021-08-02:
It does compile without this additional plugin?

Is this the source you want to compile? [https://github.com/letscontrolit/ESPEasyPluginPlayground/blob/master/_P152_MCP42010.ino](https://github.com/letscontrolit/ESPEasyPluginPlayground/blob/master/_P152_MCP42010.ino)

TD-er, 2021-08-02:
I just made a few changes to the file, so it will now compile:

```c++
//#######################################################################################################
//#################################### Plugin 152: MCP42010 #############################################
//################################## I use GPIO 12 / 14 / 15 ############################################
//#######################################################################################################
// written by antibill

// Usage:
// (1): Set value to potentiometer (http://xx.xx.xx.xx/control?cmd=digipot,0,255)
// (2): Set value to potentiometer (http://xx.xx.xx.xx/control?cmd=digipot,1,0)

#include <MCP42010.h>
// https://github.com/mensink/arduino-lib-MCP42010
static float Plugin_152_PotDest[2] = {0,0};

#define PLUGIN_152
#define PLUGIN_ID_152         152
#define PLUGIN_NAME_152       "MCP42010"
#define PLUGIN_VALUENAME1_152 "DigiPot0"
#define PLUGIN_VALUENAME2_152 "DigiPot1"

boolean Plugin_152(byte function, struct EventStruct *event, String& string)
{
  boolean success = false;

  switch (function)
  {
    case PLUGIN_DEVICE_ADD:
      {
        Device[++deviceCount].Number = PLUGIN_ID_152;
        Device[deviceCount].Type = DEVICE_TYPE_DUMMY;           // SPI pins for ESP8266 are CS=15, CLK=14, MOSI=13
        Device[deviceCount].Ports = 0;
        Device[deviceCount].VType = Sensor_VType::SENSOR_TYPE_DUAL;
        Device[deviceCount].PullUpOption = false;
        Device[deviceCount].InverseLogicOption = false;
        Device[deviceCount].FormulaOption = true;
        Device[deviceCount].ValueCount = 2;
        Device[deviceCount].SendDataOption = true;
        Device[deviceCount].TimerOption = true;
        Device[deviceCount].TimerOptional = true;
        Device[deviceCount].GlobalSyncOption = true;
        break;
      }

    case PLUGIN_GET_DEVICENAME:
      {
        string = F(PLUGIN_NAME_152);
        break;
      }

    case PLUGIN_GET_DEVICEVALUENAMES:
      {
        strcpy_P(ExtraTaskSettings.TaskDeviceValueNames[0], PSTR(PLUGIN_VALUENAME1_152));
        strcpy_P(ExtraTaskSettings.TaskDeviceValueNames[1], PSTR(PLUGIN_VALUENAME2_152));
        break;
      }

    case PLUGIN_GET_DEVICEGPIONAMES:
      {
        event->String1 = formatGpioName_bidirectional(F("CS"));
        event->String2 = formatGpioName_bidirectional(F("CLK"));
        event->String3 = formatGpioName_bidirectional(F("MOSI"));
        break;
      }

    case PLUGIN_WEBFORM_LOAD:
      {
        success = true;
        break;
      }

    case PLUGIN_WEBFORM_SAVE:
      {
        success = true;
        break;
      }

    case PLUGIN_INIT:
      {
        success = true;
        break;
      }

    case PLUGIN_WRITE:
      {

        String tmpString  = string;
        int argIndex = tmpString.indexOf(',');
        if (argIndex)
          tmpString = tmpString.substring(0, argIndex);

        if (tmpString.equalsIgnoreCase(F("digipot")))
        {
          int pot;
          int value;
          pot = event->Par1;
          value = event->Par2;

          MCP42010 digipot(Settings.TaskDevicePin1[event->TaskIndex], 
                           Settings.TaskDevicePin2[event->TaskIndex], 
                           Settings.TaskDevicePin3[event->TaskIndex]);
          Plugin_152_PotDest[pot] = value;
          digipot.setPot(pot,value);
          success = true;
        }

        break;
      }

    case PLUGIN_READ:
      {

        UserVar[event->BaseVarIndex + 0]= Plugin_152_PotDest[0] ;
        UserVar[event->BaseVarIndex + 1] = Plugin_152_PotDest[1] ;
        success = true;
        break;
      }

  }
  return success;
}
```

PLease let me know if works, then I will push it to the playground too

l1t7l3ph0o7, 2021-08-03:
It does Build Now Thank you very much.

TD-er, 2021-08-03:
Thanks, I have updated the same code on the Playground.