letscontrolit/ESPEasy #3739: Switch state 4 instead of 1 in log and send to controller

TungstenE2, 2021-08-05:
### Checklist
- x] The title of this issue is "to the point" and descriptive.
- [x ] This issue describes what is happening.
- [x ] This issue describes what components are affected (e.g. name of plugin/controller)
- [x ] This issue describes how to reproduce it.
- [x ] This issue describes when it was introduced (when known) and what version is now showing the problem.

### I have...
- [x ] searched the [issue tracker](https://github.com/letscontrolit/ESPEasy/issues) or the [forum](https://www.letscontrolit.com/forum/viewforum.php?f=1) for a similar issue. (include links when applicable)
- [ x] entered a system description using "Copy info to clipboard" on the sysinfo page. (when possible)
- [x ] entered the full filename of the used version (e.g. `ESP_Easy_mega-20181001_test_ESP8266_4096_VCC.bin` )
- [x ] given a list of active plugins or controllers when applicable.
- [x ] filled out all applicable fields below.

### Steps already tried...
- [x ] Tried a clean install (empty `.bin` files are included in the ZIP)
- [x ] Tested previous/other build (mention which one already tested)
- [ ] Tested on other node to make sure hardware isn't defective.
- [x ] Verified if the problem is limited to a single plugin/controller

### If you self compile, please state this and PLEASE try to ONLY REPORT ISSUES WITH OFFICIAL BUILDS! ###

### Summarize of the problem/feature request

I realized yesterday that the chicken door is working normal, but the wrong state (4) is send to the controller. Therefore my FHEM rules are not triggered as they are expecting 1 or 0. In the list of devices the value shows 1, which is correct, but in the log I see also value 4. But this is only the case for 1 out of 2 switches and happening radomly.

[image]

[image]

### Expected behavior

same value should be send to the controller which is shown in list of devices

### Actual behavior

YOUR TEXT GOES HERE

### Steps to reproduce

1. openeing closing the door triggers the reed switch
2. 
3. 

### System configuration

Restart did not solve the problem.

Tested with verion ESP_Easy_mega_20210114_normal_ESP8266_4M1M and  ESP_Easy_mega_20210802_normal_ESP8266_4M1M 

Hardware: D1 mini

Build:⋄
20114  - Mega
System Libraries:⋄
ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build:⋄
pygit2_not_installed
Plugin Count:⋄
47 [Normal]
Build Time:⋄
Aug  2 2021 17:05:34
Binary Filename:⋄
ESP_Easy_mega_20210802_normal_ESP8266_4M1M
Build Platform:⋄
Linux-5.4.0-54-generic-x86_64-with-glibc2.29
Git HEAD:⋄
pygit2_not_installed

ESP Easy version: 

ESP Easy settings/screenshots: 

### Rules or log data

```
![grafik](https://user-images.githubusercontent.com/39955685/128305967-c204edc1-e14b-449e-91b6-abcc844f3f6a.png)

```

tonhuisman, 2021-08-05:
Event value = 4 if the `Safe Button` state has been detected. The previous State will be restored after the event, at least that is what the code seems to do.
I think you can avoid that by disabling the `Use Safe Button (slower)` option, but that's probably the trick to make it work as reliable as you have it now 🤔

TungstenE2, 2021-08-05:
thx, but why is safe button state send to http controller while value in the device list is still previous state?

tonhuisman, 2021-08-05:
> thx, but why is safe button state send to http controller while value in the device list is still previous state?

Sorry, don't know who or why this is working this way.

giig1967g, 2021-08-05:
> thx, but why is safe button state send to http controller while value in the device list is still previous state?

After a "4" event (Safe Button) you will not see a "1" event because that would generate an unwanted event that could activate unwanted rules.
But the switch state is "1" (that you see in the devices list).
Just ignore the event=4.

TungstenE2, 2021-08-05:
so state 1 should be send to the controller, not state 4. Could you fix this?

giig1967g, 2021-08-05:
Well, not really. 
State=4 means that something went wrong and safe button has been activated. So you can monitor if anything is interferencing with your device or your wires, just checking for events=4 in your code or counting them...
That is why it is sent to http and mqtt.

I am not sure what your application is, but if you explain I can try to help you out.

tonhuisman, 2021-08-05:
The 4 is sent to controllers, and as it isn't currently handled by FHEM, maybe an option to not send it at all could be a considerable solution?
Another solution could be to have FHEM interpret the 4 as being a 1 (and unchanged), as it could mean a re-trigger, as apparently the state has changed to 1 just before the 4 is sent.

TungstenE2, 2021-08-05:
form user point of view I would expect the same value to be sent to the controller which is shown in device list.

giig1967g, 2021-08-05:
Does "longpress" or "doubleclick" work in your environment?
(they send event = 3, 10 and 11)

giig1967g, 2021-08-05:
Another thing: safe button fires event=4 when there is a false positive.
Have you tried finding the reason for this? Any interference?

TungstenE2, 2021-08-05:
@giig1967g please see this ticket for history and reason.

https://github.com/letscontrolit/ESPEasy/issues/3523#issuecomment-803290926

TungstenE2, 2021-08-06:
@TD-er what is your view on this issue?

Grovkillen, 2021-08-06:
I'd say it should show in the device page as well even though it's an
"error" state. If not it's not how every other plugin behaves. If it's not
supposed to be in the actual state we need to move it to an event
(task#valuename#error=...)

TD-er, 2021-08-06:
Another suggestion is to make something similar to what the Dallas plugin has.
It allows to assign a specific value/action on a fault situation.
Would that be useful here?

giig1967g, 2021-08-07:
Let me explain how it works:
- the STATE of a switch is the internal value of a PIN, that can have 2 or 3 values: 0: HIGH, 1: LOW, -1: OFFLINE (only for MCP and PCF)
- the EVENT is a trigger that the plugin sends to activate the rules

When there is a Safe Button event, the STATE of the switch doesn't change; but there is an EVENT generated to trace this fact.

As you can see from the log example, the EVENT is created but the state doesn't change:
```
2430665364: SW : SafeButton: false positive detected. GPIO= 4 State=1
2430665372: EVENT: d2#Switch=4.00

```
That is why on the device page and on the Pinstate page the value remains 1.

So I think that the solution could be to avoid sending the EVENT=4 to the controller.

@TD-er : what do you think, shall I filter the EVENT=4 messages in order to avoid them to reach the controller, but still generate events for the rules?

TungstenE2, 2021-08-07:
` I filter the EVENT=4 messages in order to avoid them to reach the controller, but still generate events for the rules?`

I would vote for this solution and is inline with my suggestion earlier.

giig1967g, 2021-08-08:
@TungstenE2 
please see #3755

uzi18, 2021-08-09:
@giig1967g maybe it should be as an option - someone may need it to act in rules ?

giig1967g, 2021-08-10:
You can still receive the EVENT in the rules if you activate the MONITOR command:
```
monitor,mcp,2
on mcp2devicename=4 do
....
endon
```

But I agree with you.  An option should be preferred

tonhuisman, 2023-09-02:
This seems to be resolved, so issue can be closed.

closed 2023-09-02 by TD-er