letscontrolit/ESPEasy #3741: Error 'Calculate: Unknown token' in rules, when eventvalue isn't given to the event

Micha-he, 2021-08-05:
If I query the value from an eventvalue in a rule and the eventvalue isn't explizit given, the error 'Calculate: Unknown token' is logged.

Rule example:
`on Schalten do
  if %eventvalue%=0 or %eventvalue%=1
    GPIO,5,%eventvalue%
  else
    if %eventvalue%>1
      // nichts tun
    else
      GPIOtoggle,5
   endif
  endif
endon`

Same with other eventvalues like %eventvalue1% and so on.

Discussed in the forum here: [](https://www.letscontrolit.com/forum/viewtopic.php?p=52331#p52331)

It would be great, if we could query if an %eventvalue% is unset in rules.  This would the rule above reduce to:

`on Schalten do
  if not %eventvalue%
    GPIOtoggle,5
  else
    GPIO,5,%eventvalue%
  endif
endon`

Micha-he, 2021-11-30:
I've a look in 'Rules_calculate.cpp', to find a solution for this error/feature. Quite complex code... and with my minimal skills in C++, there is nothing to be done.

Possibly an ESPEasy-crack has more success here. My thanks in advance.

TD-er, 2022-04-18:
Can you test PR #4004 ?

Micha-he, 2022-04-19:
Yes, gladly. Give me until tomorrow evening.

Micha-he, 2022-04-19:
I've found some time to test...

It didn't work better: Now a missing eventvalue is always set to 0. This isn't an improvement in my opinion.

TD-er, 2022-04-19:
> Now a missing eventvalue is set to 0 too. This isn't an improvement in my opinion.

Well, since I'm still working on it, what do you prefer?
An empty string?

TD-er, 2022-04-19:
Just an idea... instead of replacing it with some preset value when not present, maybe we should add a check to see if there is such an eventvalue?
Maybe something like `%haseventvalue1%`, which will be either 0 or 1?

TD-er, 2022-04-19:
> I prefer the first :-)

I understand, but that takes lot of code changes as we also need to add string support.

TD-er, 2022-04-20:
Can you try the latest code from #4004 ?
I now added ``%eventvalueX|Y%``  X = event value nr > 0, Y = default value when eventvalue does not exist. N.B. default value can be a string, thus ``%eventvalue3|[int#3]%`` should be possible as long as the default value not contains neither ``|`` nor ``%``.

N.B. this allows for some very tricky stuff like using the eventvalue as a command placeholder in rules.

Consider what this might do:

```
event,myEvent=%eventvalue100|factoryreset%

on myEvent* do
  %eventvalue1|%
endon
```

Micha-he, 2022-04-20:
??? It's too high for me :-(

Currently I call the event in three ways:
1. AsyncEvent,Schalten      (to toggle)
2. AsyncEvent,Schalten=1 (to switch it explicit to ON)
3. AsyncEvent,Schalten=0 (to switch it explicit to OFF)

The first one produce the error in the log, when I check the %eventvalue% to 0/1. The rule self has worked until yesterday!

What should I change in my rule (example above in post #1) or in the call to test the new firmware?
And is the new code changes a improvement to recognize an empty/missing eventvalue?
A workaround was always, to call the event with a value of e.g. 999 and check this to toggle.

In the past that rule has worked without an error. Later an error is produced in the log (at this point I started the discussion in the forum and I've been waiting to see which becomes the recommended solution) and now (with the firmware from yesterday) the toggle didn't work anymore.

Micha-he, 2022-04-20:
Should this rule work?

`on Schalten do
  if %eventvalue1|%=''
    GPIOtoggle,5
  else
    GPIO,5,%eventvalue%
  endif
endon`

or this?
`on Schalten do
  if %eventvalue|999%=999
    GPIOtoggle,5
  else
    GPIO,5,%eventvalue%
  endif
endon`

tonhuisman, 2022-04-20:
> `if %eventvalue|999%=999`

I that case it needs to be `%eventvalue1|999%`, not `%eventvalue|999%`, as in the new code `%eventvalue%` is replaced by `%eventvalue1%`, _just before_ processing the line...

Micha-he, 2022-04-20:
Yes, C&P error ;-)

tonhuisman, 2022-04-20:
More importantly, does it work with you recent adjustments❓

Micha-he, 2022-04-20:
I'll test it later, if I at home... With my inquiry I wanted to exclude attempts, which can not work.

Micha-he, 2022-04-20:
Both tests are not working properly.

First test with `if %eventvalue1|%=''`:
If-condition isn't true, GPIO is set ON with the else-condition (426506) and the log shows:
```
426505: Rules : Syntax error, missing '%' in '%eventvalue%'
426506: ACT : GPIO,5,GPIO,5,1
426509: GPIO : port#5: set to 1
426510: Command unknown: GPIO,5,GPIO,5,1
426526: EVENT: Relais#Status=1
```

Second test with `if %eventvalue1|999%=999`:
If-condition is true, GPIO is toggeled, but an error is logged:
```
510336: ACT : GPIOtoggle,5
510339: GPIO toggle: port#5: set to 0
510344: Rules : Syntax error, missing '%' in '%eventvalue%'
510464: EVENT: Relais#Status=0
```

But is this (`if %eventvalue1|999%=999`) a clear command and an improvement?

TD-er, 2022-04-20:
> Both tests are not working properly.
> 
> First test with `if %eventvalue1|%=''`:

This was explicitly told not to work.

> 510344: Rules : Syntax error, missing '%' in '%eventvalue%'

Not sure about that one, will test it myself to see what might be wrong here.

TD-er, 2022-04-20:
The error message is somewhat correct, like there is some strange mixup in argument parsing.
Probably something I may have broken this morning.
Will have a look at it now.

Just extended the log a bit more to see what's going on, given the rules you mentioned before:

```
16805 : Info   : HTTP: event,Schalten
16876 : Info   : EVENT: schalten
16886 : Info   : ACT  : GPIOtoggle,5
16889 : Error  : GPIO toggle: port#5: set to 0
16896 : Error  : Rules : Syntax error, missing '%' in '%eventvalue%': 'GPIO,5,GPIO,5,%eventvalue1' %-pos: -1
```

TD-er, 2022-04-21:
Just pushed a new commit to fix a bug I introduced yesterday which mangles the rules line when processing an eventvalue without | for the default value.

Micha-he, 2022-04-21:
Now it works without errros :thumbsup:

I now use this rule for the event, with -1 for the default-value:
```
on Schalten do
  if %eventvalue1|-1%=-1
    GPIOtoggle,5
  else
    GPIO,5,%eventvalue1%
  endif
endon
```

TD-er, 2022-04-21:
Great!
Thanks for testing.

closed 2022-04-22 by TD-er