letscontrolit/ESPEasy #3746: Move LTHO library from playground

TD-er, 2021-08-06:
The matching playground PR: https://github.com/letscontrolit/ESPEasyPluginPlayground/pull/174

svollebregt, 2021-10-07:
What should happen to continue this move? I will test it soon with the new 20211005 mega (not that I expect any issues).

tonhuisman, 2021-10-07:
You could port over the library and the ITHO plugin from the ESPEasy Playground to this repo by creating a PR for that.
The plugin should be renumbered to align with the next available number of plugins (next unused number is P118), and be enabled in a set like test_D (not sure if it will fit in all test_D configurations, we may need to add a test_E set) by adding a `#define USES_P118` in the right place in `define_plugin_sets.h`.

tonhuisman, 2021-10-09:
@svollebregt Will you be working on the transplant from EspEasyPlayground to this repo? If not I can take this one.

svollebregt, 2021-10-09:
@tonhuisman I can take care of this. Good practice in case I make future changes to the plugin.

svollebregt, 2021-12-24:
This issue can be closed now the plugin has been merged into the main repo.

closed 2021-12-24 by TD-er