letscontrolit/ESPEasy #3749: The month is incorrectly read from the external time source DS3231

Sminter4, 2021-08-07:
The month is incorrectly read from the external time source DS3231.
If the external time source is used and written with the command
`datetime,2021-08-07,10:58:00`
, the ESP first displays the correct time. After rebooting and del reading the external time source, the month minus 1 is given. So e.g. instead of August July.
Tested with a Wemos D1 mini and the EspEasy build normal from 2.8.21.

tonhuisman, 2021-08-07:
Smells like a bug in the RTClib library used.
When reading the month, it does like this:
`uint8_t m = bcd2bin(RTCWireBus->_I2C_READ() & 0x7F); // Skip century bit.`
when writing the month it does like this:
`RTCWireBus->_I2C_WRITE(bin2bcd(dt.month()) + 0x7F); // Need to write the century bit`
I'd expect that should be `| 0x80` to turn a bit on, not `+ 0x7F`, as that will lose that 1 value we are now missing when reading  back...

TD-er, 2021-08-07:
Hmm looks like something that I found as issue on the Adafruit library... but I am not sure if it was in this code or the other DS...

PR #3751 (2021-08-07): [RTClib] (Bugfix) Write month correctly to DS3231 (#3749)
After [a report](https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=8643#p54263) in the forum, then reported in the issue tracker.

Resolves #3749
closed 2021-08-07 by TD-er