letscontrolit/ESPEasy #3754: Duplicate MQTT messages after deepsleep

RicardoCst, 2021-08-08:
I keep getting duplicate MQTT messages after deepsleep happend, which messes with the Homeassistant logging.

INIT : Booting version: pygit2_not_installed (ESP82xx Core 929f0fb6, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2 PUYA support)
193 : Info : INIT : Free RAM:29272
194 : Info : INIT : Deep Sleep #4 - Restart Reason: Deep-Sleep Wake
195 : Info : FS : Mounting...
220 : Info : FS : Mount successful, used 75802 bytes of 957314
238 : Info : CRC : SecuritySettings CRC ...OK
240 : Info : Time set to 91.000 Time adjusted by 90761.00 msec. Wander: 25.21 msec/second Source: RTC at boot
242 : Info : Current Time Zone: STD time start: 1970-10-25 03:00:00 offset: 0 min
244 : Info : Local time: 1970-01-01 00:01:31
246 : Info : WiFi : Start network scan all channels
358 : Info : INIT : Free RAM:26368
359 : Info : INIT : I2C
360 : Info : INIT : SPI not enabled
606 : Info : INFO : Plugins: 77 [Normal] [Testing] [TEST_A] (ESP82xx Core 929f0fb6, NONOS SDK 2.2.2-dev(a58da79), LWIP: 2.1.2 PUYA support)
613 : Info : Webserver: start
**670 : Info : AM2320: Temperature: 26.5
671 : Info : AM2320: Humidity: 64**
1981 : Info : WD : Uptime 0 ConnectFailures 0 FreeMem 19296 WiFiStatus 0 ESPeasy internal wifi status: DISCONNECTED
2661 : Info : WiFi : Scan finished, found: 11
2664 : Info : WiFi : Best AP candidate: Ziggo5717463 AC:F8:CC:1B:9E:3C Ch:1 (-62dBm) WPA2/PSK
2666 : Info : WIFI : Connecting Ziggo5717463 AC:F8:CC:1B:9E:3C Ch:1 (-62dBm) WPA2/PSK attempt #0
3657 : Info : WIFI : Connected! AP: Ziggo5717463 (AC:F8:CC:1B:9E:3C) Ch: 1 Duration: 989 ms
3822 : Info : WIFI : DHCP IP: 192.168.178.45 (MicroBox) GW: 192.168.178.1 SN: 255.255.255.0 duration: 164 ms
3826 : Info : firstLoopConnectionsEstablished
3832 : Error : MQTT : Intentional reconnect
3873 : Info : MQTT : Connected to broker with client ID: MicroBox_0
3877 : Info : Subscribed to: MicroBox/#
3880 : Info : WD : Uptime 1 ConnectFailures 0 FreeMem 14696 WiFiStatus 3 ESPeasy internal wifi status: Conn. IP Init
**3897 : Info : AM2320: Temperature: 26.9
3898 : Info : AM2320: Humidity: 63**

TD-er, 2021-08-08:
Is this maybe because the message is put in a queue, which is then processed right after the MQTT client is connected to the broker?
It looks like your AM2320 task does sample at an interval of roughly 3 seconds.
Can you check what the set interval is for that task?
Is it a fix to set this interval to for example 60 seconds when it is running in deepsleep for most of the time?
And if this prevents the unit from sending a single value, you can also try to call `taskRun` from the rules to 'execute' that task, which will also flush the value to the connected controller.

RicardoCst, 2021-08-09:
The problem is that the interval was already 60 seconds, deep sleep awake time was 15 seconds and sleep time 300 seconds

tonhuisman, 2021-08-09:
The code looks a little bit weird (but not really _wrong_, I guess), having no `break` statement at the end of the `case PLUGIN_READ:` block.
The initial reads at 670/671 seem superfluous, as there shouldn't be a PLUGIN_READ at that time yet.

tonhuisman, 2021-08-10:
The `PLUGIN_READ` action is called (scheduled) right after `PLUGIN_INIT`, 'causing' the extra read&controller send. I think this is by design. Or maybe it could/should optionally be skipped after a wake_from_sleep situation. Not sure if that's the correct way to pursue.
Setting the Interval to 300 seconds (or more) should avoid the measurement to be sent twice during the wake-time.

RicardoCst, 2021-08-10:
I just checked and its still publishing twice

MicroBox/Environment/Temperature 23.6
MicroBox/Environment/Humidity 66
MicroBox/Environment/Temperature 23.6
MicroBox/Environment/Humidity 66
[image]

tonhuisman, 2021-08-10:
Have you tried to enable the De-duplicate option in the Controller settings?

[image]

RicardoCst, 2021-08-10:
The de-duplicate option only de-duplicates the messages that are the same, so for example the lines in bold wouldnt be filtered:
MicroBox/status/LWT Connected
**MicroBox/Environment/Temperature 23.6**
MicroBox/Environment/Humidity 65
**MicroBox/Environment/Temperature 23.5**
MicroBox/Environment/Humidity 65

which still causes issues since I still would have 2 readings 1 second apart from eachother.

tonhuisman, 2021-08-10:
I can't see why those different values, and the readings are 600 seconds apart, would be within the wake-time of ~15 seconds?
Are you also triggering a read via a `TaskRun,<task>` command from the rules? (But I would've then expected to see that in the logs...)

TD-er, 2021-08-10:
The deduplicate only checks what is still in the queue. Not the last message sent.

tonhuisman, 2021-08-10:
That screenshot of your initial report isn't relevant any longer, as you have changed settings since. Only 'fresh' logs are relevant, IMHO.

You don't seem to have Rules enabled, so sending commands from rules isn't relevant (nor possible) either.

I've now replayed most of your scenarios, but don't see the behavior you are reporting.
Can you provide new logs with the Interval setting at 300 seconds or larger?

RicardoCst, 2021-08-10:
Its even doing it without deep-sleep 

[image]

TD-er, 2021-08-10:
Is this specific to this plugin, or is it just about any plugin?

TD-er, 2021-08-10:
OK, found the cause.
Will make a fix :)

RicardoCst, 2021-08-11:
Is there any temporary workaround?

RicardoCst, 2021-08-11:
I thank you greatly! Ill keep this issue open until the pull request is merged

tonhuisman, 2021-08-11:
> Ill keep this issue open until the pull request is merged

That is the best way to go, as merging the connected PR will automatically close this issue 👍

TD-er, 2021-08-11:
Please let me know if the fix is working as intended.
Just to make sure there isn't another reason why the values are reported twice (or actually the readings are done twice)

RicardoCst, 2021-08-11:
Works great now with no bugs!

TD-er, 2021-08-11:
Great :)
I will then merge it.

PR #3757 (2021-08-11): [MQTT] Fix duplicate call task PLUGIN_READ at MQTT connect (#3754)
Fixes: #3754 

Problem was that upon MQTT connect to the broker, the tasks were re-scheduled to run.
This is not useful now the read values are queued.
It only needs to call the MQTT import tasks if present.
closed 2021-08-11 by TD-er