letscontrolit/ESPEasy #3763: TaskValueSet not working in ESP_Easy_mega_20210802_normal_WROOM02_2M256

mister-gefahrensucher, 2021-08-14:
I have just updated my two OBI plugs from 
ESP_Easy_mega_20210223_normal_WROOM02_2M256
to
ESP_Easy_mega_20210802_normal_WROOM02_2M256

this simple rule doesn't update my Task 3 anymore 

in 20210223 it's working long time in 20210802 not it's not working anymore

on button#state do
 if [button#state]=1
     gpio,4,1 //relay on
     gpio,13,1 //led_on
     TaskValueSet 3,1,1 // task relay state on
 else
     gpio,4,0 //relay off
     gpio,13,0 //led off
     TaskValueSet 3,1,0 // task relay state off
 endif
endon

so when i hit the button it fires 3 Actions but the state in Task 3 doesn't change
1883060: EVENT: button#state=1
1883083: ACT : gpio,4,1
1883085: GPIO : port#4: set to 1
1883090: ACT : gpio,13,1
1883092: GPIO : port#13: set to 1
1883095: ACT : TaskValueSet 3,1,1

1919657: EVENT: button#state=0
1919688: ACT : gpio,4,0
1919690: GPIO : port#4: set to 0
1919693: ACT : gpio,13,0
1919694: GPIO : port#13: set to 0
1919697: ACT : TaskValueSet 3,1,0

The relay and led is going on and off as expected,

[image]

i have no other rules.

ghtester, 2021-08-14:
This is not a bug but a feature... ;-)
Currently the TaskValueSet is allowed to Dummy device only.

TD-er, 2021-08-14:
After the 20210802 build I merged this commit: https://github.com/letscontrolit/ESPEasy/commit/713c3450839d03aa74c752c71ff6e3c0a78a8d10
So can you test a more recent build, based on the latest code.
I can also make you a build if you like.

It now has an option in tools->Advanced to allow the taskvalueset for all plugin types, not only for Dummy tasks.

TD-er, 2021-08-14:
I made you a text build: [ESP_Easy_mega_20210814_normal_WROOM02_2M256.bin](https://www.dropbox.com/s/78ilgcfx4umlmbt/ESP_Easy_mega_20210814_normal_WROOM02_2M256.bin?dl=0)

mister-gefahrensucher, 2021-08-15:
Thanks Build 20210814 working great now.

closed 2021-09-05 by TD-er