letscontrolit/ESPEasy #3764: BME680 - where can I select it as sensor

stoffelll, 2021-08-14:
Hi,

Maybe this is a stupid question.

I read in the changelog the P106/BME680 is moved from playground.
So I assume it is now a standard plugin and available in the build starting 2020-12-27.
Unfortunately I'm unable to find the sensor/BME680 in the build - tested it with various different builds including testing plugins etc.

So is the BME680 already "fully" integrated and available in ESPEasy and if yes, where do I find it?
Or which binary I've to use?

Thanks a lot!
sl

TD-er, 2021-08-14:
See [documentation](https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html#list-of-official-plugins) where it lists in which build what plugin is included.

BME680 is in build "testing B"

tonhuisman, 2021-08-14:
It is in the `test_B` builds.
There is an overview of all plugins, [here](https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html), listing the build they are in. Anything that is in `normal` is also in any of rhe `tsst` builds.

stoffelll, 2021-08-15:
thanks a lot!

closed 2021-08-15 by stoffelll