letscontrolit/ESPEasy #3776: Blynk 2.0 Compatibility

garudaonekh, 2021-09-03:
Hello,
Blynk 2.0 is a big change. How to make ESPEasy compatible with the new Blynk?

Thanks

TD-er, 2021-09-03:
Is there a new Blynk library for the Arduino platform?
We use [this library, versionn 6.1](https://github.com/blynkkk/blynk-library/tree/v0.6.1)
Although it looks like a big jump in version nr, the 1.0 and 1.0.1 are the next ones.
So you might start with upgrading that library and see if it still compiles :)

If that's enough, please let me know, so I can update it for other users too.

Any idea if the revisions are compatible?

garudaonekh, 2021-09-04:
Ok, i will try to upgrade the library and see if it works. But I am confused with C012 and C015 and BLYNK. Both c012 and c015 are Blynk. I enabled all and blynk controller appear.

I think Blynk and ESPEASY are best combinations. Blynk is rich in mobile and very user friendly for end users while ESPEASY are best the device level end. Correct me if I am wrong because I am from Tasmota background and just trying to understand ESPEasy for the last 2-3 days.

garudaonekh, 2021-09-04:
Ok, it works. Just drop the new library and it works like a charm.

TD-er, 2021-09-04:
I will re-open it as it still has to be done :)

TD-er, 2021-09-14:
Can you test this PR: https://github.com/letscontrolit/ESPEasy/pull/3783 ?

garudaonekh, 2021-09-16:
Thanks, it works. But TEMPLATE_ID field is missing in your controller setting. I don't know if in the future Blynk will require it or not.

By the way, to response to update action from Blynk app(ex. relay on/off), I will have to use Rules, is there anyway to update it directly without using Rule?

TD-er, 2021-09-16:
Do you have a link to how these actions are sent?
Some controllers, like OpenHAB and Domoticz MQTT allow for receiving messages and then hook into specific plugins like the Switch plugin.
If it is rather generic in Blynk, we can for sure add that.

What format does the TEMPLATE_ID have?

garudaonekh, 2021-09-17:
Here's the template id: format: #define BLYNK_TEMPLATE_ID "TMPL-5jj-xxx"

Blynk uses virtual pin: V0 up to V255. they can have 3 data types(INT,STRING, Double).  So for the case of switch, we use double with value of 1 and 0.

To get ESPEasy work I have to first set Device Value Name to V1,V2....
and then I set the following RULE in ESPEasy to response to App action and it works:

on blynkv2=1.00 do
   gpio,13,1
endon
on blynkv2=0.00 do
   gpio,13,0
endon

**The hard part here for everyone is setting the rule. needs to understand how rule in ESPEasy works.**

TD-er, 2021-09-17:
I would simplyfy the rules a bit since comparing with a double to exactly match is a bit tricky.
Thus better to have 1 rules block matching the event and in that block use `%eventavlue1%` ... `%eventavlue4%` compared to trying to match the eventvalue in the rules block `on ... do` part.

```
on blynkv2 do
  gpio,13,%eventvalue1%
endon
```

garudaonekh, 2021-09-17:
template length is 12 characters.
blynkv1 param name is based on the old C015 is used. 

`// This is called for all virtual pins, that don't have BLYNK_WRITE handler
BLYNK_WRITE_DEFAULT() {
  uint8_t  vPin     = request.pin;
  float pinValue = param.asFloat();

  if (loglevelActiveFor(LOG_LEVEL_INFO)) {
    String log = F(C015_LOG_PREFIX "server set v");
    log += vPin;
    log += F(" to ");
    log += pinValue;
    addLog(LOG_LEVEL_INFO, log);
  }

  if (Settings.UseRules) {
    String eventCommand = F("blynkv");
    eventCommand += vPin;
    eventCommand += F("=");
    eventCommand += pinValue;
    eventQueue.addMove(std::move(eventCommand));
  }
}`

tonhuisman, 2021-10-09:
Blynk-n00b question:
Is there a Template-ID for every separate device/sensor you want to send data to Blynk cloud, or is the same Template-ID used for multiple devices? As if it is per device/sensor, that would require to have a setting per device, and as we don't have setting reserved for that, probably the TaskName could be used as the Template-ID (except ESPEasy doesn't 'like' dashes in task names).

garudaonekh, 2021-10-10:
There can be multiple devices for a single template id.

garudaonekh, 2021-10-10:
In blynk, there are up to 255 datastream or params for a device thus it should more than enough for all tasks.

TD-er, 2021-10-10:
And how do you signal to Blynk which sensor device you're using?
In other words, how do you distinguish on the Blynk server a temperature from 2 different Dallas sensors for example?

TD-er, 2021-10-10:
OK clear.
Is it a good idea to add a checkbox to the Blynk controller to automatically rename the values to this format based on the task index and variable nr?
Or perhaps use the IDX value as a starting point for the value nr? (e.g. IDX = 100, then values of that task are 100, 101, 102 and 103)

commit 70d285b3 (2021-10-04): [Blynk] Update to library v 1.0.1

Fixes: #3776