letscontrolit/ESPEasy #3779: ESPEasy with GPRS/GSM 2G (Time Sync)

garudaonekh, 2021-09-07:
Hi,
My project is about remote controller for farm. So we need to use 2G.

I manage to get ESPEasy+Blynk Controller to work with 2G Modem on TTGO T-Call. So it can connect to my Blynk cloud using 2G.

But the problem I am facing now is the time. I have no idea how to set the time on ESPEasy when I get it from Blynk cloud time. I also use RTC.

Thanks;

TD-er, 2021-09-07:
See the [datetime command](https://espeasy.readthedocs.io/en/latest/Reference/Command.html?highlight=datetime#internal-commands)

garudaonekh, 2021-09-17:
`#include "src/Helpers/_CPlugin_Helper.h"
#ifdef USES_C015

# include "src/Globals/CPlugins.h"
# include "src/Commands/Common.h"
# include "src/ESPEasyCore/ESPEasy_backgroundtasks.h"

// #######################################################################################################
// ########################### Controller Plugin 015: Blynk  #############################################
// #######################################################################################################

// This plugin provides blynk native protocol. This makes possible receive callbacks from user
// like button press, slider move etc.
// This require much more ESP resources, than use of blynk http API.
// So, use C012 Blynk HTTP plugin when you don't need blynk calbacks.
//
// Only one blynk controller instance is supported.
//
// https://www.youtube.com/watch?v=5_V_DibOypE

// #ifdef PLUGIN_BUILD_TESTING

// Uncomment this to use ssl connection. This requires more device resources than unencrypted one.
// Also it requires valid server thumbprint string to be entered in plugin settings.
// #define CPLUGIN_015_SSL

# define CPLUGIN_015
# define CPLUGIN_ID_015         15
# define _BLYNK_USE_DEFAULT_FREE_RAM
# define BLYNK_TIMEOUT_MS 2000UL
# define BLYNK_HEARTBEAT      300
# define CPLUGIN_015_RECONNECT_INTERVAL 60000
#define BLYNK_PRINT Serial // Defines the object that is used for printing
#define BLYNK_DEBUG        // Optional, this enables more detailed prints

#define SIM800L_IP5306_VERSION_20200811
// Select your modem:
#define TINY_GSM_MODEM_SIM800

#include <TinyGsmClient.h>
#include "ttgo_utilities.h"
#include <BlynkSimpleTinyGSM.h>

// Select your modem:
#define TINY_GSM_MODEM_SIM800

// Set serial for debug console (to the Serial Monitor, default speed 115200)
#define SerialMon Serial

// Set serial for AT commands (to the module)
// Use Hardware Serial on Mega, Leonardo, Micro
#define SerialAT Serial1

// See all AT commands, if wanted
//#define DUMP_AT_COMMANDS

// Define the serial console for debug prints, if needed
#define TINY_GSM_DEBUG SerialMon

// Please select the corresponding model

// Your GPRS credentials, if any
const char apn[] = "smart";
const char gprsUser[] = "";
const char gprsPass[] = "";

#include <TinyGsmClient.h>
#include <BlynkSimpleTinyGSM.h>

#ifdef DUMP_AT_COMMANDS
[... 536 lines cut ...]
  }

  // addLog(LOG_LEVEL_INFO, F(C015_LOG_PREFIX "connected handler"));
}

// This is called when Smartphone App is opened
BLYNK_APP_CONNECTED() {
  if (Settings.UseRules) {
    eventQueue.add(F("blynk_app_connected"));
  }

  // addLog(LOG_LEVEL_INFO, F(C015_LOG_PREFIX "app connected handler"));
}

// This is called when Smartphone App is closed
BLYNK_APP_DISCONNECTED() {
  if (Settings.UseRules) {
    eventQueue.add(F("blynk_app_disconnected"));
  }

  // addLog(LOG_LEVEL_INFO, F(C015_LOG_PREFIX "app disconnected handler"));
}

#endif // ifdef USES_C015
`

tonhuisman, 2022-04-18:
This seems to be solved, so can be closed.

closed 2022-04-18 by TD-er