letscontrolit/ESPEasy #3781: AM2320 not shown in i2c-scanner (SDA=GPIO1, SCL=GPIO3)

Micha-he, 2021-09-12:
All tests are with image from 02.08. or the source from that date.

If I disabled the serial-port on a ESP and connect a AM2320 at the changed i2c-bus (SDA=GPIO1, SCL=GPIO3), the scanner didn't show the device, but it works flawlessly.

[image]
[image]

A test on a other ESP (firmware ESP_Easy_mega_20210802_normal_ESP8266_1M) with same serial-/i2c-settings, but with a MCP23017 as test-device, is displayed correctly in the i2c-scanner, even if the i2c-bus is configured on GPIO1 & GPIO3. 

Maybe a problem with the AM2320-address ?

Unfortunately, I cannot test the AM2320 on another ESP because I only have one and it is connected firmly.
The ESP in this Lingan SWA1 device, has only the RX/TX-pins available for i2c communication.

tonhuisman, 2021-09-12:
The I2C scanner will show all device-addresses where a response is received. Possibly it won't show the device-name or, if it is configured, the plugin name, depending on the build used (that part is left out f.e. from the `test` builds), but it _should_ be able to list the address.

tonhuisman, 2021-09-12:
According to the datasheet, the max. SCL clock frequency is 100 kHz, have you configured the I2C 'Slow device Clock Speed' at 100000 Hz (or lower)? As that setting is used by the I2C scanner. Your task should probably also have that option enabled.

Micha-he, 2021-09-12:
Yes, I've set the i2c-speed to slow in the device config. Otherwise the AM2320 doesn't provide any data!

tonhuisman, 2021-09-12:
It may be a peculiarity of that sensor, as the library used by ESPEasy has this code in the `read()` method:
```C#
  Wire.beginTransmission(AM2320_address);
  Wire.endTransmission();
  // запрос 4 байт (температуры и влажности)
  Wire.beginTransmission(AM2320_address);
  Wire.write(0x03);// запрос
```
So in fact it starts and stops an I2C transmission sequence, only to _then_ send the actual read command.

TD-er, 2021-09-26:
I think this may have been fixed by code I wrote yesterday.
There was an issue related to the frequency switching and initialization of I2C.

On the other hand, the code shown by  @tonhuisman may also be the true cause.
The first Wire.begin... and immediate Wire.end... is also referred to as waking up the I2C device.
So perhaps if you scan twice in short intervals, you may also see it?

Micha-he, 2021-09-26:
I'll do a test today evening...

Micha-he, 2021-09-26:
Yes, when I fast reload the i2c-scanner page, then the device(-address) is shown.

TD-er, 2021-09-26:
OK, I will see what I can do.
Not sure if we need to scan it twice on each scan, but maybe a flag can help here, or maybe check when the last scan was performed and if that was less than N seconds ago, we can scan each address twice.

tonhuisman, 2023-09-02:
This is most likely fixed, so issue can be closed.

Micha-he, 2023-09-04:
Nice, that you are closed this issue. But the problem still exists !

TD-er, 2023-09-05:
Thanks for getting back on this.

The main purpose for closing those 200+ issues was to only keep those still relevant issues.
Apparently this one is still relevant, thus let's keep it open.

TD-er, 2023-09-05:
Maybe the I2C scan should be done at 100 kHz?

tonhuisman, 2023-09-05:
I2C scan is already run at 100 kHz (Low I2C speed).

We might indeed need to wake I2C devices twice, but it is a particularity of this device that it won't respond like 99.9% of other I2C sensors. I'm not sure how other devices might like that, as it seems to be in the grey area of the specifications 🤔

Micha-he, 2023-09-05:
The following was noticed during the test with a current custom-firmware:

The AM2320-device isn't read anything and is disabled after some minutes. I'll do another test with disabled "Check I2C devices when enabled" option tody evening.

TD-er, 2023-09-05:
The I2C check does actually use a similar check, so that may explain why it is being disabled after a while.

So we need to exclude this device from the I2C device check procedure.

Micha-he, 2023-09-05:
Testresult from today: If I disable "Check I2C devices when enabled", the AM2320-device remains active !

tonhuisman, 2023-09-05:
@Micha-he I've started a fix for these issues in #4788 and a build is brewing in this [GH Actions run](https://github.com/letscontrolit/ESPEasy/actions/runs/6089193994).
Can you please test to see if this solves the issues listed above, and report your findings here?

Micha-he, 2023-09-05:
True, but I need a 2M256 build with the AM2320 plugin!

tonhuisman, 2023-09-05:
Any other required plugins in that build?

tonhuisman, 2023-09-05:
This is the Custom 2M256 build, with added P051, P037, P105 and P144 added: (I had that available, and just added P051 😉)
[ESP_Easy_mega_20230905_custom_ESP8266_2M256.zip](https://github.com/letscontrolit/ESPEasy/files/12528095/ESP_Easy_mega_20230905_custom_ESP8266_2M256.zip)

Micha-he, 2023-09-05:
And P001?
I'll flash it tomorrow and report here.

tonhuisman, 2023-09-05:
The default Custom set is included, as defined in [`pre_custom_esp82xx.py`](https://github.com/letscontrolit/ESPEasy/blob/mega/tools/pio/pre_custom_esp82xx.py), so P001 is included by default 😉

Micha-he, 2023-09-06:
The changes at the AM2320-plugin and the i2c-scanner works fabulously ! I didn't see any problems.

PR #4788 (2023-09-17): [P051] Fix device being automatically disabled and I2C scan issue
Resolves #3781 
Resolves #4750 

Bugfix:
- AM2320 got disabled after some time, so disabled the I2C Device check for this plugin
- I2C Scan is erratic for this device
- GPIO selector showed `- None - [SD-CARD]` if SD-Card was included in build but not configured

Improvements:
- Make I2C scan more robust/reliable by trying to wake-up the device first
- Update I2C Scan documentation
- Add example in Rules documentation for `PostToHTTP`/`PutToHTTP` commands

TODO:
- [x] Extend I2C documentation showing result with I2C Multiplexer
 [... cut]
closed 2023-09-17 by TD-er