letscontrolit/ESPEasy #3786: Read error PN532

PrutzerNL, 2021-09-25:
I have a PN532 connected on a NodeMCU (ESP8266) Everything was working fine. The NFC reader was working.
But as soon as I changed an unused pin GPIO boot state: mode GPIO-13 (D7) to output low the PN532 stopped the PN532 stopped working. In the log file I see read error’s.

When I change the unused pin to default it was working again.
But after some trying, the PN532 stopped working completely. Resetting the PN532, power everything off and on, nothing will help anymore. I also reset the whole EPS8266 to default, but it did not help.

Firmware: 20210802 (Energy) and 20210114 (Energy)
Is this a bug? I’m I doing something wrong?

TD-er, 2021-09-25:
What pins do you use for I2C? The default 4&5 ?

PrutzerNL, 2021-09-25:
I use:

SDA: GPIO-4 (D2)
SCL: GPIO-5 (D1)

tonhuisman, 2021-09-25:
Does the I2C bus have proper terminating resistors installed? (Though I think most of these readers have them installed by default, it's possible you have one that doesn't.)
You can try to add 10 k resistors between SDA and 3V3 and SCL and 3V3.

tonhuisman, 2021-09-25:
And another thing to check: Have you set 'I2C ClockStretchLimit:' (On Tools/Advanced page)?
The source code suggests it should be set to 2000 for this device (though that's probably the only place where that is mentioned, in a comment of the plugin source)

And OT:
This would have better gone in the Forum, as this very likely _not_ a software issue, as it has been working fine so far.

PrutzerNL, 2021-09-26:
I did set ClockStretchLimit to 2000. It did not help. I will try a 10K resistor.

tonhuisman, 2021-09-26:
Can also be a problem in the wiring of the board 🤔

PrutzerNL, 2021-09-26:
I did change SDA SCL a few time to test. That was not the case.
I use 3.3V to power the PN532. Should I try 5V?

tonhuisman, 2021-09-26:
> did change SDA SCL a few time to test

I was suggesting that maybe one of the wires is broken (even though it worked fine before), that sometimes happens.

PrutzerNL, 2021-09-26:
Change all the wires and measured all the wires. 
Changed to Tasmota. That is also now working. I think my PN532 is dead.

TD-er, 2021-09-26:
Sounds like it.
Still strange it may be destroyed by the actions you mentioned.
So I guess it might be just bad luck.

tonhuisman, 2023-09-01:
This can be closed.

closed 2023-09-01 by TD-er