letscontrolit/ESPEasy #3789: ESP32: GPIO-39 missing on _P002_ADC.ino

fredvduin, 2021-09-29:
After upgrading mega-20210503 to mega-20210802 GPIO-39 was missing on  _P002_ADC.ino.
Fix for me was to change  on file ESPEasyLimits.h   
#define MAX_GPIO 39   =>     #define MAX_GPIO 40

tonhuisman, 2021-09-29:
That is sort a crude fix, but the actual culprit [is this line](https://github.com/letscontrolit/ESPEasy/blob/mega/src/src/WebServer/Markup.cpp#L933)
If that single increment is dismissed, GPIO39 is shown again as intended, and the #define can keep the expected/intended value 😃

TD-er, 2021-09-29:
Yep, agreed the MAX_GPIO should not be higher than the max usable pin.
I guess we run into several other issues when increasing this nr.
Better fix it properly.

PR #3794 (2021-10-05): [PinSelect] Bugfix missing last GPIO in Touch and Hall Effect selector (ESP32)
Resolves #3789
closed 2021-10-05 by TD-er