letscontrolit/ESPEasy #3790: [P039] issues with MAX31855

megamarco833, 2021-09-30:
### Summarize of the problem/feature request
i'm with last release ESP_Easy_mega_20210802_normal_ESP8266_4M1M
 and if i use plugin [P039] with MAX31855 i receive some strange numbers as temperature: very variable and incorrect values.
example. i have the probe on my desk and i see values from -20 to -15 (that's not so cold :) )
the probe is working, it's a thermocouple type k and i have two, both with same issue.
if i take in my hands the probe i see values goes from -7 to -0.3°C
**i do not switch + and -**
i also have a max6675 and with same probes with this device i'm able to read correct values.

i try also to siwtch + and - (so wrong connection) and in this case i see positive values in the range of 8 to 12 °C (impossible, i have 27°C)
if i take the probe in my hand (with this wrong connection of + and -) i see that values goes to -20

### Steps to reproduce

1. ESP_Easy_mega_20210802_normal_ESP8266_4M1M
2. use MAX31855 and thermocouple type k

below some logs with the probe with correct connection of + and - and just place on my desk (temperature should be 27.5°C (measured with MAX6675 and same probe)
```
3859802: P039 : termoc : Temperature: -18.8
3859891: EVENT: termoc#Temperature=-18.8
3862723: P039 : termoc : Temperature: -23.0
3862738: EVENT: termoc#Temperature=-23.0
3865722: P039 : termoc : Temperature: -24.5
3865790: EVENT: termoc#Temperature=-24.5
3868668: P039 : termoc : Temperature: -25.5
3868694: EVENT: termoc#Temperature=-25.5
3871667: P039 : termoc : Temperature: -25.0
3871701: EVENT: termoc#Temperature=-25.0
3898735: EVENT: termoc#Temperature=-31.3
3901159: Dummy: value 1: 0.0
3901161: Dummy: value 2: 0.0
3901162: Dummy: value 3: 0.0
3901164: Dummy: value 4: 0.0
3901191: EVENT: setpoint#S1=0.0
3901290: EVENT: setpoint#S2=0.0
3901390: EVENT: setpoint#S3=0.0
3901490: EVENT: setpoint#S4=0.0
3901668: P039 : termoc : Temperature: -30.3
3901696: EVENT: termoc#Temperature=-30.3
3904668: P039 : termoc : Temperature: -30.3
3904692: EVENT: termoc#Temperature=-30.3
3907667: P039 : termoc : Temperature: -31.0
3907691: EVENT: termoc#Temperature=-31.0
3910723: P039 : termoc : Temperature: -30.8
3910736: EVENT: termoc#Temperature=-30.8
3913667: P039 : termoc : Temperature: -26.3
3913691: EVENT: termoc#Temperature=-26.3
3916669: P039 : termoc : Temperature: -31.3
3916692: EVENT: termoc#Temperature=-31.3
```

TD-er, 2021-09-30:
The problem with thermocouples is that they generate a voltage offset at the point where 2 different metals connect.
So in theory you can also have this effect at the point where you connect the wires of your thermo couple to the MAX31855.
Also the voltages generated by a thermo couple are relatively low at low temperatures.

According to [Wikipedia K-type thermocouple](https://en.wikipedia.org/wiki/Thermocouple#Type_K) this sensitivity is 41 µV/°C
So to me, these sensors are only usable to measure high temperatures and/or high ranges.
If -at room temperature- they are off by let's say 30 degree, then the voltage is only off by a little over 1 mV. (just assuming it is behaving linear)

1 mV can easily be achieved by the wires acting as an antenna, picking up 50Hz from the wall outlet.

I still need to dive into this subject myself, but at first glance I don't think you're using the right tool for measuring these temperatures.

What can be done:
- Add filtering in the code
- Look into settings related to 50/60 Hz filtering
- Maybe the sensor allows for some different approaches to measure the thermocouple effect?
- Maybe some other filtering can be added at the sensor input? (e.g. R/C filter?)

Still interesting to dive into this, but for now I don't think it is reasonable to see accurate results at the temperature you're testing with. (though a "stable" reading would be very welcome I guess)

TD-er, 2021-09-30:
Is the offset constant?

megamarco833, 2021-09-30:
> 
> 
> Is the offset constant?

if you are referring to sensor that is inside boiling water: yes it still measure between 65-68°C

TD-er, 2021-09-30:
More like, if you add like 32 to every value you measure, are the values then looking OK?
Or is it not a constant offset, but more like you need a factor (or a factor + offset) ?

megamarco833, 2021-09-30:
Got it, sorry
No it's not a constant value.
All values that have been showed are very strange...
Also negative as i report, but at room temperature, so per you explanation we can skeep the value at room temperature.
Anyway all values are strange, tested also in bbq at 180degree but i see values starting from 110 to 145 and bbq was very stable at 180

Again,with max 6675 and same probe i can measure correctly the room temperature, the bbq temperature and also the boiling water is report correctly at 100degree

TD-er, 2021-09-30:
Do you have a 2nd probe to make sure there is no bad connection?

megamarco833, 2021-09-30:
> 
> 
> Do you have a 2nd probe to make sure there is no bad connection?

yes, i wrote in first post, i have two probes, and both are working with correct values when connect to max6675 but fail on MAX31855.

TD-er, 2021-09-30:
Apparently you're not the only one.
[first hit on Google](https://forum.allaboutcircuits.com/threads/k-thermocouple-values-not-stable-max31855-information-on-relay.171630/) with links to other threads.