letscontrolit/ESPEasy #3791: Allow more eventvalues for events

TD-er, 2021-10-03:
As requested on [the forum](https://www.letscontrolit.com/forum/viewtopic.php?f=6&t=8774)

ToDo:
- [ ] Allow for N `%eventvalueN%` (not limited?)
- [ ] Replace unavailable event values with some default value (0 ???)

Micha-he, 2021-10-03:
#3741

TD-er, 2021-10-03:
Thanks, that was indeed the issue I had in mind for the 2nd point.

tonhuisman, 2021-10-03:
> Replace unavailable event values with some default value (0 ???)

Should probably be a setting (not sure what category it should go into though).
And maybe also add a command to set that default? Like: `defaulteventvalue,N`

TD-er, 2021-10-03:
We can also add a command like "letcheck" (or something with a better name) to not assume default values, but check for special values (also to check for NaN etc)

closed 2022-04-22 by TD-er