letscontrolit/ESPEasy #3793: TaskValueSet does not update the value

kini24, 2021-10-04:
I have an ESP8266 with PZEM-004t v3.0 connected. Everything worked fine for over a year, but I decided to update the firmware, hoping to improve stability and fix bugs.
Previously, everything worked like this. Domoticz sends a correction value to the ESP, which uses it to calculate Energy and display it on a display connected to the ESP. After updating the firmware, I noticed that the display shows data without correction. Data goes to Domoticz also without correction. Experiments have shown that the "TaskValueSet" function in rules does not work. I tried replacing it with TaskValueSetAndRun, SendTo and SendToHTTP - there were no changes.
I looked through similar threads, tried solutions that seemed correct to me, but they didn't work either.

Build: 20114  - Mega
System Libraries: ESP82xx Core 2843a5ac, NONOS SDK 2.2.2-dev(38a443e), LWIP: 2.1.2 PUYA support
Git Build: pygit2_not_installed
Plugin Count: 55 [Normal]
Build Time: Aug  2 2021 16:56:08
Binary Filename: ESP_Easy_mega_20210802_energy_ESP8266_4M1M
Build Platform: Linux-5.4.0-54-generic-x86_64-with-glibc2.29
Git HEAD: pygit2_not_installed

Devices and values:
[image]

Rule ("Let" and "[VAR#1]" me add during experiments):
```
on pzem#Power do
  Let 1,[pzem#Energy]+[Correction#Value]
  TaskValueSet 2,4,[VAR#1]
  Publish domoticz/in,'{"idx":105,"nvalue":0,"svalue":"[VAR#1];0;0;0;[pzem#Power];0","RSSI":%rssi%,"Battery":100}'
endon
```

Log:
```
1781892: EVENT: pzem#Power=261.20
1781911: ACT : Let 1,5671.15+17713.65
1781921: ACT : TaskValueSet 2,4,23384.8
1781943: ACT : Publish domoticz/in,'{'idx':105,'nvalue':0,'svalue':'23384.8;0;0;0;261.20;0','RSSI':-59,'Battery':100}'
1782017: EVENT: pzem#Energy=5671.15
1782079: EVENT: pzem2#PowerFactor=0.77
1782115: ACT : Publish domoticz/in,'{'idx':166,'nvalue':0,'svalue':'0.77','RSSI':-59,'Battery':100}'
1782180: EVENT: pzem2#Frequency=49.90
```

tonhuisman, 2021-10-04:
Well, this is in the "That's not a bug, that's a feature" category. 😄 
The `TaskValueSet` command is originally intended to _only_ update the value for a Dummy device, but it also updated the values of any other device. That has been changed, but because some people have used this behavior as a feature, a setting has been added on the Tools/Advanced page to still allow this, "Allow TaskValueSet on all plugins:" in the "Special and Experimental Settings" section.

kini24, 2021-10-04:
Unfortunately, I do not have such an option in the settings.
Ok, how then can the value of the above parameter be changed? Tried in the formula bar to write something like
`%value%+[Correction#Value]`
but it doesn't work and throws an error in the log

tonhuisman, 2021-10-04:
Aha, that setting may have been introduced after that release.
I can download the .bin file and post it here, if desired (or you may be able to download yourself from the Actions tab, here on Github, best pick the latest one titled 'Build').

kini24, 2021-10-04:
Fine! Thank you very much, everything works! :-)

tonhuisman, 2021-10-04:
Great. Can you then close this issue? (as I can't 😉)

closed 2021-10-04 by kini24