letscontrolit/ESPEasy #3810: Feature request: Choose behavior when calling the host

chromoxdor, 2021-10-14:
I wish there where a feature, where you can choose between an own created site or the dashboard and the main espeasy site when calling the mainpage.
e.g. if you call a device it shows the main-tab. but if you could somewhere enter a filename for a html file hosted on the device that will be loaded instead and the main-tab could be still accessible via ip-or-hostname/home

ironataerial, 2021-10-15:
If you upload a file name index.htm then the system will automatically direct you to that

chromoxdor, 2021-10-17:
I just found out, that if you use an index.hm you loose the ability to reboot via webinterface....

chromoxdor, 2021-10-17:
I am of course able to open the tools page and i am able to click the reboot button... but it doesn’t reboot but brings me back to the index.html.
and this index.html page is intended to be minimalistic. a reboot button there makes no sense...