letscontrolit/ESPEasy #3828: P062 keypad - MPR121 Touch

VitalikicN, 2021-10-31:
For correct operation, you need to make changes to the library.
The changes I made to the Adafruit_MPR121.cpp library file are:

Leave the thresholds the same - this should be, setThreshholds(12, 6);

change, writeRegister(MPR121_CONFIG1, 0x10); // default, 16uA charge current
to, writeRegister(MPR121_CONFIG1, 0x20); //increased to 32uA charge current

change, writeRegister(MPR121_CONFIG2, 0x20); // 0.5uS encoding, 1ms period
to, writeRegister(MPR121_CONFIG2, 0x3A); //0.5us encoding, 18 samples, 4ms period

Next change,
uint16_t Adafruit_MPR121::baselineData(uint8_t t) {
if (t > 12) return 0;
uint16_t bl = readRegister8(MPR121_BASELINE_0 + t);
return (bl << 2);
}

to,
uint16_t Adafruit_MPR121::baselineData(uint8_t t) {
if (t > 3) return 0;
uint16_t bl = readRegister8(MPR121_BASELINE_0 + t);
return (bl << 2);
}

After making a change, the sensor works through 6mm glass without false alarms
Information taken from here https://forums.adafruit.com/viewtopic.php?f=19&t=72025&start=15

TD-er, 2021-10-31:
Yep, should be configurable for sure.
N.B. I made your post a bit more readable by adding code block markings.

tonhuisman, 2021-12-29:
What's to be changed here, do we only need 1 setting, like `Allow 4-6mm glass in front of the sensor` or should it be something more elaborate with a couple of different sensitivity settings? (and where to get that settings data from?)

I can pick this one up, it won't be too complicated if it is only a simple toggle/checkbox, but would like confirmation of that.

tonhuisman, 2021-12-29:
I chose to have a combobox with room for more settings, this is what I got so far:

[image]

(_Including_ the implementation, as described above)

PR #3893 (2022-01-21): [P062] Add Panel sensitivity option
Resolves #3828 

- Add selection for Panel sensitivity, Normal as current, Extra sensitive to allow mounting the keypad behind 4-6 mm glass
- Reformatted the source using Uncrustify

TODO:
- [ ] Testing (help wanted)
- [x] Documentation (help and suggestions welcome)
closed 2022-01-21 by TD-er