letscontrolit/ESPEasy #3830: cannot compile in arduino ide

ChaitanyaYeole02, 2021-11-02:
I am new ESP32 and arduino ide. I have BL0937 and I have connected to my ESP32. But I am not able to compile the given example in this git hub repo. I am getting an error _Plugin_Helper.h not found.
Please help with this beginning steps

ChaitanyaYeole02, 2021-11-02:
`_P076_HLW8012.ino`

tonhuisman, 2021-11-02:
May I suggest to use PlatformIO, as that is what many developers for ESPEasy use, and is the endorsed development environment. A basic manual is available in [Read The Docs](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html), and I'm working on a more comprehensive version of that in this PR #3784 (check out the Files section).

ChaitanyaYeole02, 2021-11-02:
Yes, is there any way to run the same in Arduino IDE?

tonhuisman, 2021-11-02:
It is technically possible, but I don't use that.
Why do you want to use Arduino IDE?

ChaitanyaYeole02, 2021-11-02:
Because my remaining code is in Arduino IDE where I communicate with my server and send different sensor values to the server. 
I am kind of scared to shift to a different IDE which will take a lot of time. I may be wrong about this.

ChaitanyaYeole02, 2021-11-02:
Will my Arduino functions compile in PlatformIO?

tonhuisman, 2021-11-02:
> If you know the way for Arduino IDE please let me know

https://espeasy.readthedocs.io/en/latest/Participate/ArduinoIDE.html

I know it works (as others also insist on using Arduino IDE), but I never used it myself on ESPEasy.
For *me* there are 3 major (and dozens of minor) reasons for not using Arduino IDE:
- Library management is (mostly) manual (highly error-prone)
- Directory structure has to be changed (renamed) to be able to compile the project (**big** deal-breaker for me)
- The IDE is pretty awkward to use, looking very '1990', and not using modern keybindings (though this is all quite personal, I know)

TD-er, 2021-11-02:
Just putting the discussion of Arduino IDE vs. PlatformIO aside.
Just between the lines I read that you're not trying to compile ESPEasy, but rather using 1 file of ESPEasy in your own project.
The include you mention is on the 1st line of that .ino file, so even if that's going to be fixed, I guess you will be back here with dozens of other questions about compiling.

So first let's take a step back and try to make clear to what you intend to do here.
What is it you try to make and if you're not trying to build ESPEasy, why do you want to compile that single plugin?
That plugin code is very specific to be used in ESPEasy, as the non-INO files in the project are all there to accomodate the use of plugins. So without the rest of the code I don't think you can really use the plugin code in your project without lots of changes.

TL;DR
What is it you're trying to do here?

ChaitanyaYeole02, 2021-11-04:
Yeah, you are right. I have BL0937 and ESP32, I have done the connections and wanted to get the voltage and current values for the same. Then I saw that this file `_P076_HLW8012.ino` works for that sensor too. I was trying to compile the same file but it did not compile. I just needed help in getting the voltage and current values from the sensor by using this code. Will I be able to do that without using PlatformIO and shifting my whole setup over there. 
Thank you in advance and would be really helpful if you can help with the same.

TD-er, 2021-11-04:
OK, just to be sure we're talking about the same here.
You are NOT using ESPEasy source code, right? Only trying to use the mentioned .ino file in your project?

If so, then I think it is really really hard to help you for the simple reason that I can't possibly know what the source code of your project is doing.

tonhuisman, 2023-09-09:
Compiling the ESPEasy code using Arduino IDE is not endorsed, better use VSCode/PlatformIO as [documented](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html).
This issue can be closed.

closed 2023-09-09 by TD-er