letscontrolit/ESPEasy #3831: Add support for WT32-SC01

TD-er, 2021-11-02:
Nice board with color tour screen.
But information is hard to find, so this issue is also a bit of a central place to collect information.

* [WT32-SC01-Exp](https://github.com/seaniefs/WT32-SC01-Exp) demo showing rotating cube
* [WT32-SC01-Demo](https://github.com/sowbug/WT32-SC01-Demo) Another demo, more like a fork of the previous one.
* [WT32-SC01DataSheetV3.3-2-with-nuts.pdf](https://github.com/letscontrolit/ESPEasy/files/7463686/WT32-SC01DataSheetV3.3-2-with-nuts.pdf)
* [cnx-software short review](https://www.cnx-software.com/2020/11/13/wt32-sc01-thin-3-5-inch-esp32-color-touchscreen-display/) with some links.
* Uses [Adafruit_FT6206_Library](https://github.com/sowbug/Adafruit_FT6206_Library)
* [OpenHASP docs - WT32-SC01](https://haswitchplate.github.io/openHASP-docs/0.6/devices/wt32-sc01/) Also shows a 3D printed enclosure

tonhuisman, 2021-11-02:
It would really be nice to have support for that ST7796s chip in P116 (#3761).
So far I've only seen hacked ST7735/ST7789 drivers to support the ST7796' resolution of 320x480. Seems it's mostly initialization, and the color depth is the same, so there is quite a fair chance it's possible to patch that in 🤔 (I'll order a standalone display, if only to play with it, as that WT32-SC01 is a bit pricey for my taste 🚫)

TD-er, 2021-11-02:
This is -as far as I know- one of the least expensive units with ESP32 and touch screen.
I paid 32 euro for it, incl. shipping.
Just make sure not to order by this seller claiming to ship from France, as I am still waiting for that shipment while another one from China delivered within 10 days, incl VAT.

net1957, 2021-11-25:
I got one yesterday, but don't know how to start to hack it ...

tonhuisman, 2021-11-25:
You can upload a ESP32 `DISPLAY` build to the unit (assuming you got the 4MB version, for 16MB Flash edition you can use the `MAX` build) by downloading that from the [P116] [Pull Request build run](https://github.com/letscontrolit/ESPEasy/actions/runs/1486957955) using the [official Espressif flash tool](https://www.espressif.com/en/support/download/other-tools) (That is know to work reliably with all types of USB controllers/drivers).
You _will_ need that specific ESPEasy build to be able to use the ST7796 TFT controller on the device, as these ST77xx controllers are not yet supported in the currently released ESPEasy 😉 
Then connect to the WiFi Access Point presented by the device after the flash process is completed, and provide WiFi credentials so it will be visible on your WiFi network, so you can configure it.

tonhuisman, 2021-11-25:
NB: I'm already working on the FT6206 touchscreen controller, though that will take some more work (and a new PR [P124], to be created).

net1957, 2021-11-25:
Thanks!
I got it and this  espeasy version is now installed but with display should I select?
[image]

net1957, 2021-11-25:
Thanks,

I did flash it with this bin (HEAD_30f6e8d) an I'm connected.

As I'm new to display world, I have some difficulty to make it work...
With the schematics and the doc in the PR I can't figure out to correctly : 

- set I2c and SPI on the hardware page
- set the actuator on the plugin page

At this time only the Backlight percentage is working but nothing is on the display (it is white)
What is my mistake?

[image]
[image]
[image]
[WT32-SC01_V3.2-schematics.pdf](https://github.com/letscontrolit/ESPEasy/files/7605168/WT32-SC01_V3.2-schematics.pdf)

tonhuisman, 2021-11-25:
This is my Hardware setup:

[image]

[image]

And the configured device (Backlight percentage you can choose yourself):

[image]

net1957, 2021-11-25:
Yeah! Thanks for your help. It's displaying my statics lines defined in the Device

One more question.
Do you have a link that give me advices about all commands  for writing text,shapes,etc... ?

tonhuisman, 2021-11-26:
The last pdf version of the help for this plugin was published here: https://github.com/letscontrolit/ESPEasy/issues/836#issuecomment-905354032

That issue might be of interest to read, seeing me struggle with some bugs (mostly fixed by now, AFAIK) 😅

net1957, 2021-11-26:
Hello, It's working now. 
I checked howto to make a arc to display value (see: http://doc.8ms.xyz/docs/case-tutorial-en/case-tutorial-en-1cmict41o0sof) but this don't seems to be available in adafruit-gfx-graphics-library.

Any idea how to do that ?

net1957, 2021-11-26:
It's going well. But one more question about passing parameters on commands:
int#3 contain numeric value (Y position)
`st77xx,txtfull,395,[int#3],,,%eventvalue1%` work
`st77xx,txtfull,395,[int#3]+15,,,%eventvalue1%` don't work

is this a limitation of the rules syntax or of the plugin?

tonhuisman, 2021-11-26:
> It's going well. But one more question about passing parameters on commands: int#3 contain numeric value (Y position) `st77xx,txtfull,395,[int#3],,,%eventvalue1%` work
 `st77xx,txtfull,395,[int#3]+15,,,%eventvalue1%` don't work
> 
> is this a limitation of the rules syntax or of the plugin?

The argument processing is somewhat limited for most commands, but more extended using the `let` commands. It might help if you add a couple of round braces around the calculation, but I'm not sure that will help.
When using 'normal' rules, I'd advise to use a separate `let 103,[int#3]+15` and use `[int#103]` in the command itself.

> Any idea how to do that ?

I'm not a math nor graphics buff, so I wouldn't know how to draw that kind of shapes

TD-er, 2021-11-26:
Drawing a circle isn't really rocket science. But it is probably rather slow and at least unnecessary to implement it is rules.
So I think adding some functions to draw slightly more complex structures would be great.
Not sure if it should be all the basic stuff like circles, but maybe more like a gauge which can be given a few markup parameters and a value to display.

net1957, 2021-11-26:
Yes, with "let" it work, but we have to be careful in usage in asyncevents.

I like the idea of a gauge as ultimately it's the good representation that I need.
Cheers

TD-er, 2021-11-26:
If you use the `let` command right before calling where it is used, there is no issue with async events.

tonhuisman, 2021-11-28:
> I checked how to make a arc to display value (see: http://doc.8ms.xyz/docs/case-tutorial-en/case-tutorial-en-1cmict41o0sof) but this don't seems to be available in adafruit-gfx-graphics-library.

I just added a new subcommand `lm` (line, multiple segments), that makes it possible to come close to an arc, by drawing a line across multiple provided x/y coordinates, switching color on the fly, if desired, and optionally closing the shape.
Build is just started, [result will be here](https://github.com/letscontrolit/ESPEasy/actions/runs/1513047581) in a bit.

Documentation for this subcommand (source) is [here](https://github.com/letscontrolit/ESPEasy/pull/3761/files#diff-4f55377bc3cd9d940e43b6de0cb5e0e55cfc382e03b148f94c4116eadfdc79daR137-R147)

dreamer1976, 2022-11-18:
I have a WT32-sc01. and have tried many firmwares. but i get no image on my display only my backlite works. just like net1957.
I also tried the tonhuisman settings but still only backlite. Can someone help me to get the WT32-sc01 up and running, preferably with touch input. WT32-sc01 version V3.2

Ik heb een WT32-sc01. en heb heel veel firmwares geprobeert. maar ik krijg geen beeld op mijn dispay alleen mijn backlite werkt. net als bij  net1957.
ook de instellingen van tonhuisman heb ik geprobeert maar nog steeds alleen backlite. Kan iemand mij verder helpen om de WT32-sc01 aan de gang te krijgen het liefst met touch imput. WT32-SC01 V3.2

tonhuisman, 2022-11-18:
To have support for the display on WT32-SC01 you can use any current ESPEasy build that includes P095, so the `Display` or `Max` builds.
When testing the touchscreen support, you **must** use a build from #4016 as that's the only build that includes support for the FT62x6 touchscreen.
This is my current configuration:
**Hardware** tab
[image]

**Display** (Task 5 on my ESP)
[image]

And the **Content** section:

[image]

For the **touchscreen** (Task 7) I'm using this configuration for testing: (that's all I can cram in there 😝, Object 24..28 are unused/empty)
[image]

And in **Rules** I'm using this:
```
on window do
 if %eventvalue1|1%=1
  st77xx,defwin,20,50,390,219,1,1
  st77xx,win,1
  st77xx,clear
  st77xx,r,0,0,[st7796#width],[st7796#height],white
  st77xx,rot,0
  st77xx,txtfull,20,20,1,%sysbuild_filename%
  st77xx,l,0,0,[st7796#width],[st7796#height],white
  st77xx,rot,2
  st77xx,txtfull,20,20,1,%sysbuild_filename%
  st77xx,rot,3
  st77xx,txtfull,20,20,1,%sysbuild_filename%|%sysbuild_filename%
  st77xx,rot,1
  st77xx,txtfull,20,20,1,%sysbuild_filename%|%sysbuild_filename%
  asyncevent,window2seconds=1,2
  st77xx,win,0
 else
  asyncevent,clearwindow=1
 endif
endon

on window2 do
 if %eventvalue1|1%=1
  st77xx,defwin,411,50,69,109,2
  st77xx,win,2
  st77xx,clear
  st77xx,r,0,0,[st7796#width],[st7796#height],green
  st77xx,l,0,0,[st7796#width],[st7796#height],green
  st77xx,l,[st7796#width],0,0,[st7796#height],green
  asyncevent,window2seconds
  st77xx,win,0
 else
  asyncevent,clearwindow=2
 endif
endon
on window2seconds do
  if [st7796#iswin,%eventvalue1|2%]=1
    let,120,[st7796#win]
    st77xx,win,%eventvalue1|2%
    let,121,[st7796#txs]
    st77xx,txs,3
    let,122,[st7796#rot]
    st77xx,rot,%eventvalue2|0%
    let,20,([st7796#width]-[st7796#length,%systm_hm%])/2
    let,21,([st7796#height]-[st7796#textheight,%systm_hm%])/2
    st77xx,txtfull,[int#20],[int#21],3,red,black,%systm_hm%
    st77xx,txs,%v121%
    st77xx,rot,%v122%
    st77xx,win,%v120%
  endif
endon
on clearwindow do
  if [st7796#iswin,%eventvalue1|-1%]=1
    st77xx,win,%eventvalue1%
    st77xx,clear
    st77xx,delwin,%eventvalue1%
    st77xx,win,0
  endif
endon
On FT6206* do
  LogEntry,"*** Name: %eventname%, par: %eventpar%"
  If %eventvalue5|-1%=-1 // Respond to X/Y/Z/Group/<object> events
    TaskValueSet,TouchVars,Touch%eventpar%,%eventvalue1%
    AsyncEvent,drawScreen#%eventpar%=%eventvalue0%
  Else // Respond to Button-clicks
    AsyncEvent,drawButton=%eventvalue0%,`%eventpar%`
    If %eventvalue1% >= 0 and %eventvalue2% >= 0
      // Do interesting button-stuff
      LogEntry,"*** Button action %eventpar%"
    Endif
  Endif
Endon

on drawScreen#In do
  if %eventvalue1%=1
    touch,enable,filler
  else
    touch,disable,filler
  endif
endon
on drawScreen#Airco do
  AsyncEvent,window=%eventvalue1%
endon
on drawScreen#Room do
  AsyncEvent,window2=%eventvalue1%
endon
on drawScreen#w2 do
  touch,off,room
endon
on drawScreen#Swiped do
  LogEntry,"*** Swipe direction: [newFT6206#swipedir,%eventvalue1%], x: %eventvalue2%, y: %eventvalue3%"
  touch,swipe,%eventvalue1%
endon
on Clock#Time=All,**:** do
  st77xx,txtfull,400,6,2,white,blue,%systm_hm%
  asyncevent,window2seconds=1,2
  asyncevent,window2seconds=2
endon
```

dreamer1976, 2022-11-19:
Thank you for the explanation. the screen now works. Is there a (beta) bin for the touch. The version I have now only has the Touch - XPT2046 on a TFT display . I'm not that familiar with easyesp development.

tonhuisman, 2022-11-19:
You can download the binaries.zip file from an Actions run for that PR (search for P123), like [this most recent one](https://github.com/letscontrolit/ESPEasy/actions/runs/3499949032)
Then update with the same .bin file you have now installed (assuming Display on a 4MB WT32-SC01 or Max on a 16MB WT32-SC01), and you should have that plugin available.

dreamer1976, 2022-11-19:
Thank you very much.
It functions. Maybe I should wait a little longer.
but i can experiment with this now.
Eventually I want to use the display to control my air conditioning via MQTT, for example, if possible.

PR #4016 (2024-05-29): [P123] Add FT6206 Capacitive touch screen and [ESPEasy_TouchHandler] helper
Add support for FT6206 capacitive touch screen, as installed on the WT32-SC01
Resolves #3993 
Resolves #3831 

Uses an adapted version of the [Adafruit FT6206 Library](https://github.com/adafruit/Adafruit_FT6206_Library)

Functionality similar to P099 XPT2048 resistive touch screen

Introducing `ESPEasy_TouchHandler` helper module
- Define several button shapes and caption types, on/off state and enable/disable
- Define button groups to form a rudimentary menu system
- Define Slider/Gauge controls
- Handle clicking but [... cut]
closed 2024-05-29 by TD-er