letscontrolit/ESPEasy #3833: Add support for EM20918 - Long Distance proximity sensor

TD-er, 2021-11-03:
Used in WT32-SC01
Needed for: https://github.com/letscontrolit/ESPEasy/issues/3831

From the datasheet:

> EM20918 is a low-power I2C interface 2 in1 module includes Long-Distance Proximity Sensor (PS) and internal Current Drivers with IR VCSEL Diode.  It is designed for applications which requires long distance detection.
> EM20918 is application specific for custom-made products and no need of a controller to initialize and control. The gain and threshold values could be custom-made and detection result could automatically output. 
> Proximity Sensor is 850nm wavelength IR VCSEL Diode outputs IR light outwards and reflects to pixel arrays. In order to removing component variations and eliminate crosstalk caused by IR reflection, EM20918 set 2 registers for compensation. 8 steps programmable current driver is more flexible to remove component-to-component variation, and also increased dynamic range for PS detection

In other words, this plugin may need to be extended later with other possible settings, or maybe pre-programmed sets of settings to support various implementations in different products.

Features:

- Proximity Sensor with corner compensation, Long-Distance Proximity Sensor, 1 current drivers and IR VCSEL Diode integrated in a single Optical Module
- Temperature Compensation
- Higher Sensitivity of Proximity Sensing, Detecting Distance of grey card up to 50cm and white card up to 80cm
- Proximity Sensor (PS) with Ambient Light Rejection
- 8 Steps Programmable current driver for flexibility and removing component variations
- More than 2 pixels for proximity detection compensation
- Proximity detection in parallel for Long-Distance Proximity Detection
- Offset register adjustment for compensation of IR reflection caused system crosstalk
- Programmable Interrupt for PS
- Low Average Operation Power consumption
- Output Type: I2
- Operation Voltage 2.5V ~ 3.6V C Bus (ALS/PS) up to 400K Hz
- Logic Interface Voltage VBUS=1.8V or VBUS=VDD
- Minimum External Components
- Package: (4mmX2.4mmX1.4mm)
- Rosh package

tonhuisman, 2021-11-05:
Might be useful to integrate this in P099, as that already has logic for defining touch areas and 'button' like objects. Or put that logic in a separate module to reuse as much as possible without duplicating code.

TD-er, 2021-11-05:
Nope, it describes a completely different interface, so let's make it simple and make it a separate plugin.
If both the touch interface and this one should somehow interact with each other, we should make that work, but let's not make things complicated where the (simple)  design that makes this complicated is there for a reason :)

tonhuisman, 2021-11-08:
I mixed up this with the touch panel on the WT32-SC01, that's why I mistakenly suggested to include it in P099, sorry.

Is this sensor included in some HW versions of the WT32-SC01, or are you mounting it as an add-on next to the display? Because we already support a few ToF sensors, like the [VL53L0X](https://espeasy.readthedocs.io/en/latest/Plugin/P110.html#p110-page) (~10-200cm) and the [VL53L1X](https://espeasy.readthedocs.io/en/latest/Plugin/P113.html#p113-page) (~10-400cm), that could do nicely instead of this 'long distance' (50/80cm?) sensor.

TD-er, 2021-11-08:
I do have the datasheet, received via email from the manufacturer.
No idea if these differ a lot or not. It is installed in the WT32-SC01, or at least the schematics suggest it.

tonhuisman, 2021-11-08:
OK, but running an I2C scan on the device only shows the touch screen driver at 0x38, and no other devices, so that's why I asked 🤔

TD-er, 2021-11-08:
Ah. perhaps different versions then? Or they cloned a design with some parts left out ;)
Or maybe the chip can be disconnected, or put on another set of pins, so it can be run from the ULP processor to wake the unit.

tonhuisman, 2021-11-08:
In the pdf you linked to in the WT32-SC01 issue, the sensor _is_ in the schematics, indeed, but there is no mention of it further on where the separate components are described. There is some kind of see-through in the front-plate that could be used for a ToF/distance sensor, so it is quite likely that there are units available having this sensor 👍

uzi18, 2021-11-10:
Datasheet for this chip is available? 
Could sb attach file here.

TD-er, 2021-11-10:
I do have the datasheet, as I received it by mail from the vendor.
However I have not gotten a reply on my mail whether it may be published.

kdschlosser, 2023-02-13:
The data sheet is already published so it would not be a problem to pass it on.

The display does not have the proximity sensor installed. If you read the schematic is states in the schematic that it is not installed.

> EM20918 (SMT-NC)

In the parenthesis. SMT-Not Connected

kdschlosser, 2023-02-13:
I have this entire display mapped out, All the available pins and what is connected to where. I also have it running MicroPython with LVGL and a module is made to work inside of MicroPython. I have a driver written for the display and also a driver written for the touch interface.

So if there is anything you nee to know about this display I am able to provide it.

tonhuisman, 2023-02-13:
The display-controller is already supported by ESPEasy, haven't found one with the proximity sensor installed though (not even from the manufacturer...), not sure if will be available as a separate sensor, there still isn't much info to be found on-line about it.

kdschlosser, 2023-02-13:
The company that makes it Wireless-Tag sells these things to be rebranded by other companies. I am sure that the schematic was done for some customer at some point in time and the part for the proximity sensor was never removed.

It is the only ESP32 driven display made that doesn't have the majority of the IO pins used up on things like SD card readers and audio drivers. Or driving a 16 bit parallel display. This is the only one I have been able to find that has more than 3 or 4 pins available. I needed to have 10 pins for the project I am working on. It's a good little display. I am working on the interrupt portion of the touch driver and modifying LVGL so it will work with using an interrupt instead of polling the touch screen.

tonhuisman, 2023-02-13:
I have already a working (polling) plugin for the touchscreen in PR #4016 though that's not merged yet. (I need to work on the settings storage, can't fit enough in the 1kB currently reserved for custom settings)

As an alternative to this unit, maybe the [LilyGO Lily PI](http://www.lilygo.cn/prod_view.aspx?TypeId=50032&Id=1353), but I haven't started work on that touchscreen yet (it will be very similar to P123 in the PR I mentioned). Didn't check how many GPIO pins are available though.

kdschlosser, 2023-02-13:
4 GPIOs left on the LilyPi