letscontrolit/ESPEasy #3836: New Release

fly74, 2021-11-05:
Hi friends. Is it really necessary to download 600MB to get a few binaries.  😳

tonhuisman, 2021-11-05:
There is a _plan_ to provide a custom build service where you will be able to select (sets of) plugins, that will then be built on the fly or retrieve that previously created build from a 'cache' (especially for the standard builds).

And then there is of course the Web flash tool: https://td-er.nl/ESPEasy/

fly74, 2021-11-05:
Thanks Ton.

closed 2021-11-05 by fly74

TD-er, 2021-11-05:
It is already updated, which sadly is still a manual proces.
I am still updating the web flasher, but sadly I made a mistake here on my PC so I deleted the generated JSON files for ESP32 which I am now re-generating.

Also still a manual process at this moment.