letscontrolit/ESPEasy #3838: [HTTPS] Support for ESP32

TD-er, 2021-11-06:
Add support for HTTPS for the web pages of ESP32.

Proposed library to use: [ESP32 HTTPS Server](https://github.com/fhessel/esp32_https_server)

ironataerial, 2021-11-07:
SendToHttpS would be nice as well

TD-er, 2021-11-07:
> SendToHttpS would be nice as well

Yep, but that's going to be either a part of the MQTT TLS I'm working on or a separate pull request.
This is about acting as a webserver. SendToHttp/SendToHttpS are clients, not a server.

ironataerial, 2021-11-08:
Proper authentication with user name / password is also a missing option. Even with a single / admin user (at least for start.)

ironataerial, 2021-11-18:
Any chance GPIO output states to be stored on ESP32 to survive a power reboot ?

TD-er, 2021-11-18:
Future plan is to add support for I2C EEPROM/FRAM chips (pin- and software compatible) where states like these can be stored.
I2C EEPROMs can be written a lot more than typical flash, but still not an infinite amount of writes.
FRAM can be written an infinite amount of times (and the strange thing is reading isn't infinite, but still "enough" with 10^12 times)
I do have a few of them already here, and just ordered a few more with difference sizes to test if they behave differently.
But given the upcoming events in the following months, makes it hard to give a timeline for it.  (in February we get the keys of our rebuilt house, so we can move back.... finally)

ironataerial, 2021-11-18:
I was thinking of storing once every 30sec would be an acceptable time frame for actual smart home applications

Based on these calculations :

https://www.esp8266.com/viewtopic.php?f=32&t=6109

Would get nearly 30 years of actual usage which is far more than the life expectancy of the ESP itself

Congrats on getting back in the house. Has been quite some time now, no ?

TD-er, 2021-11-18:
> Congrats on getting back in the house. Has been quite some time now, no ?

Yep.
When we go back, we have been in the temporary house for 26 months.
And the whole process started when we were expecting our daughter. She will be 8 when we return.

BasvanH, 2024-12-10:
I'm trying to use `SendToHttp`, `PutToHttp` and `PostToHttp` to interact with an https API interface. Doenst seem to work, isnt SSL/TLS currently supported in these commands?

TD-er, 2024-12-10:
Nope not yet.
Only email and MQTT now support SSL.