letscontrolit/ESPEasy #3839: [Plugins] List of planned new plugins (work in progress, request a Plugin-ID here)

tonhuisman, 2021-11-07:
As it is sometimes quite hard to find the next available Plugin-ID for a new plugin, this list is supposed to be updated by either me or TD-er once a plugin is planned.
If you are working on a new plugin (for the `mega` branch, _not_ the ESPEasyPluginPlayground repository, that has its own numbers), you can request a Plugin-ID to be reserved, just add a comment, a short description/title for the plugin and we'll assign an available ID.
Once the plugin is merged it must be removed from this list, as the current list of plugins is available in this [Read The Docs page](https://espeasy.readthedocs.io/en/latest/Plugin/_Plugin.html)

New plugins, currently being worked on:
- **P116** Display - ST77xx TFT (ST7735/ST7789/ST7796) ( @tonhuisman) PR #3761 
- **P117** (todo)
- **P118** (todo)
- **P119** Gyro - ITG3205 ( @tonhuisman) PR #3823
- **P120** Acceleration - ADXL345 ( @tonhuisman) PR #3832 
- **P121** Position - HMC5883 ( svn2208?) PR #(todo)
- **P122** Environment - SHT21 ( beicnet?) PR #(todo)
- **P123** Touch - FT62x6 ( @tonhuisman) PR #(todo)

TD-er, 2021-11-07:
I added it to the pinned issues.

handfreezer, 2022-02-07:
Hello, 
  I would like to get a plugin number to pull request a plugin : it is intended to measure current through classic SCT013 with and ADS1015, using RMS method.
  It is already working based on Arduino pure project, then I forked ESPEasy on my github (I created a P127 plugin for local dev) and convert to a plugin. It is working well too. So I just need a plugin number I guess?

Best regards.

tonhuisman, 2022-02-07:
> it is intended to measure current through classic SCT013 with and ADS1015, using RMS method.

Isn't that already supported by [Plugin 25 ADS1115](https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P025_ADS1115.ino) (not fully compatible with ADS1015) or the [ADS1015 plugin from PluginPlayground](https://github.com/letscontrolit/ESPEasyPluginPlayground/blob/master/_P167_ADS1015.ino)? Or is it an extension/improvement of that plugin?
Ideally it should be as generic as possible, so it is also usable for other purposes.

P130 is the next available Plugin ID.

TD-er, 2022-02-07:
And add it to the "testing E" set.

handfreezer, 2022-02-07:
Hello,
   None of us. All the important thing is : "Measure of Current by **RMS** method" : the fact is to measure the current of a sinus form, and the RMS method require to sample a lot of time in only **ONE** wave . Say about 50Hz, this means you have to do a lot of adc conversion in only 20ms ! From what I understood from both plugin you suggested, it is not possible.
  Note I also added few parameters to do the job right like the availibility to enter the frequency, the number of wave to sample, ...
  I based my work on P025 (for structure code), but I had to rework about all the job:
[my local dev P127_data_struct.cpp](https://github.com/handfreezer/ESPEasy/blob/mega/src/src/PluginStructs/P127_data_struct.cpp)
[my local dev P127_data_struct.h](https://github.com/handfreezer/ESPEasy/blob/mega/src/src/PluginStructs/P127_data_struct.h)
[my local dev _P127_IRMS_ADS1015.ino](https://github.com/handfreezer/ESPEasy/blob/mega/src/_P127_IRMS_ADS1015.ino)

  Remark, I still have to do an improvement : I want to add a setup key to choose sample rate and a "just-in-time reader" in continuous mode sampling.

  I don't want to duplicate an already code existing if so. I just suggest my work .

handfreezer, 2022-02-07:
> And add it to the "testing E" set.

OK

TD-er, 2022-02-07:
OK, so it is indeed some different approach.
Have not yet looked at your code, but I will as soon as it appears as a PR. (I will add your nr to the post here)
Just one thought... is it blocking code for longer than a single period?

handfreezer, 2022-02-07:
> Just one thought... is it blocking code for longer than a single period?

Right : and this is exactly why I:
- added a warning in the setup page of the device,
- I didn't not notice any bad effect on 100ms, and as I said, only 40ms is enough,
- also, the final user can choose the comportment by changing in setup page the frequency of the wave AND the number of wave to use for sampling,
- and the measure of the current do not need a high frequency of reading, which means, the interval could be set to something like 30s or 60s

TD-er, 2022-02-07:
I can imagine that 20 msec isn't "stable" as you may just start on the "spike" of current on devices with poor power factor.
Then it becomes quite timing critical whether you may include this spike one time or twice. Also the net frequency isn't always constant.
I guess measuring for 30 msec would also be OK and then you need to look for the zero-crossings and measure in between. Just consider that it might need to "loop" on the sample buffer, or actively wait for the zero crossing and then start to accumulate the samples until you saw 2 more zero-crossings (make sure to include a timeout)

Upto roughly 100 msec is OK, but less is better as every blocking code for over 10 msec can lead to missing WiFi packets.
I do highlight the parts in the timing stats for > 100 msec as those are the ones to look at first when running into issues.

But these kinds of discussions are best for the actual PR I guess :)

handfreezer, 2022-02-08:
In fact, I wrote my own ino code at beginning (wifimanager + adafruit GFX + adafruit1X15 + MQTT + personnal code), and it was working for 4 months . I decided to switch to easyesp to avoid me managing the Wifi and the MQTT diffusion.
About the timing, from my understand of current RootMeanSquare formula : the most important is to get a half or a full sinus over and under zero, and the formula do the job. So, for a 50Hz voltage, you have to choose a factor of 10ms.

tonhuisman, 2022-02-09:
Have a look at the [guide for developers](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html#starter-guide-for-local-development-on-espeasy), new to Github and ESPEasy, part of our Read The Docs (RTD) documentation. On that same page, a few paragraphs up, there is also a guide on how to set up your development environment using VSCode and PlatformIO, as that's currently the preferred development environment.

TD-er, 2022-02-09:
> create a branch, and make a clean delivery commit in it?

That's the main thing you need to do.
No need to fork it again.
Then push the commits on this new branch and go to the Github page of the letscontrolit/ESPEasy repo (either on the pull request or the code page). There you will be presented with a yellow-ish bar asking you to create a pull request and follow those steps.
If you later push new commits to that same branch, those will be appended to that pull request as you essentially link that branch to the pull request.
All is explained in the link posted by Ton.

tonhuisman, 2022-03-06:
> I would like also to have a personnal profile build with only what I need. May I ask you how to do it without pulling it?

There are several ways to achieve that, the officially supported ways:
- PlatformIO (VSCode/Atom): Copy file `Custom-sample.h` to `Custom.h` and enable all features and settings to your liking. That file is excluded from git to be committed. Then from PlatformIO build any `custom` configuration, and it will use that .h file. As described in the [PlatformIO guide](https://espeasy.readthedocs.io/en/latest/Participate/PlatformIO.html)
- When using Arduino IDE, it's a bit more cumbersome, best is to follow the [Arduino IDE guide](https://espeasy.readthedocs.io/en/latest/Participate/ArduinoIDE.html)

Some alternatives: (when _not_ using a `Custom.h`, and assuming you use PlatformIO)
- Modify either (or both) `pre_custom_esp8266.py` or `pre_custom_esp32.py` (choose your used CPU-architecture) then build a `custom` configuration from PlatformIO. Be careful to not commit these files, as they are used for the provided custom builds, and TD-er usually determines what should go in there.
- Modify `define_plugin_sets.h` to include the plugin in the set you want it to be, like TESTING_E. Then build that configuration from PlatformIO. Only commit the change(s) from this file to include the plugin in the desired set, _not_ any possible other personal changes you've made there.

handfreezer, 2022-03-06:
OK, thanks for answer. So, adding a custom personnal env in platformio_esp32_envs.ini is not the right way, isn't it?

tonhuisman, 2022-03-06:
> adding a custom personnal env in platformio_esp32_envs.ini is not the right way, isn't it?

Adding such environment is only feasible if it adds an improvement to ESPEasy for all to benefit.
In your case it is _way_ easier to have a local `Custom.h` file. You can of course create several of these, and rename to `Custom.h` when desired. Just remember to not include them in a PR.

handfreezer, 2022-03-06:
@tonhuisman : Thanks for all your answer.
On my side, I have ended my first cleaning of code to do a first PR, BUT I would like to document the use of this plugin (with simple schema, and picture). May I ask you where this should be done to win a bit of time?

tonhuisman, 2022-03-07:
> compiling TEST_E : I failed sometime with a memory size error

Are you building that using VSCode/PlatformIO? And what operating system?

> I would like to document the use of this plugin

That is a great initiative 👍 

Documentation is part of the source code. The format used is ReStructured Text (.rst files), using the Sphinx tools. Initial setup is [documented here](https://espeasy.readthedocs.io/en/latest/Participate/Documentation.html). (Need to execute at least once). Once the PR is merged your doc work is available on the [ESPEasy readthedocs](https://espeasy.readthedocs.io/en/latest/) page.
What to do to add plugin documentation:
- Create a `P<pluginid>.rst` file (`P130.rst`) in folder `docs\source\Plugin`. I usually copy/paste/adapt an existing file. When copying an existing file, rename all placeholders to the new name. Write the documentation using the [rst format](https://www.sphinx-doc.org/en/master/usage/restructuredtext/basics.html). Preferably use .png format for screenshots. Photo's can be in .jpg format.
- Add the generic info to `docs\source\Plugin\_plugin_substitutions_p13x.repl` (file doesn't exist yet, create the file, and copy the P120* entries from `_plugin_substitutions_p12x.repl` and rename & adjust them.
- Add that _plugin_substitutions_p13x.repl file to `_plugin_substitutions.repl`.
- Add your plugin reference to `_plugin_categories.repl` in the mathcing category. Then it will automatically show up on the second part of the Plugin page.
- Add (copy) a line to `_Plugin.rst` to reflect your plugin P130, like P125 (the current last one in the list, first 'part' of the file).
- Build the documentation as per the instructions (`cd docs` and `.\make.bat html`) Most likely that will throw an error on the 'magick' tool, used for normalizing the image sizes, not being available, but AFAICS that will only work on Linux/MacOS, and it's no problem on Windows.
- Open the generated `index.html` file from `docs\build\html` in a webbrowser, and review your work.

handfreezer, 2022-03-13:
Waouh, ok, it is a lot of stuff, but I don't stop my work here, let me this week to have a look.
About my environment, I'm under W10+VSCode+PlatformIO.

handfreezer, 2022-04-03:
Hello, It was more complicated to start writing doc as I was thinking, but it is done, and commited on [here](https://github.com/handfreezer/ESPEasy/tree/P130_01) .
I feel my plugin is ready to create a PR. Maybe, do you want just to review the doc before?

tonhuisman, 2022-04-03:
> It was more complicated to start writing doc as I was thinking, but it is done

Understood, I've gone through the same learning curve, some time ago, but it _is_ looking good, though there are a few spelling thingies to be resolved.

> I feel my plugin is ready to create a PR

It sure does look like that 👍 so, just create the PR, and we'll comment on it (we're always nice 😄), but that is all part of the learning curve.

handfreezer, 2022-04-10:
Well, I have a problem:
[image]
I fell I'll have to solve this before to check my plugin is always working well.
I'll also include an option to convert Irms value to real value.

tonhuisman, 2022-04-10:
I'd not bother about that now and just create the pull request, as suggested, as that merge _can_ be done later without much issue.

handfreezer, 2022-04-18:
Pull request created, but I failed to remove some local commit on my repo from pull request like creating a custom build in general config file or adding an alert header in the readme.md file of the project to redirect people to the original project...

flashmark, 2022-11-18:
I am playing around with the MQ135 CO2 sensor (and MQ3 alcohol sensor) trying to combine them to a generic MQxxx sensor controlled by a table. Can you assign me a plugin ID for this attempt?

By the way, I am not sure about the quality of these sensors. Seems to deviate a lot from my MHZ19.

tonhuisman, 2022-11-19:
> Can you assign me a plugin ID for this attempt?

**`P145`** is assigned, please use `Gases - MQxxx (MQ135 CO2, MQ3 Alcohol)` for the PLUGIN_NAME, to keep naming in line with other sensors in this category.

Brommander, 2023-09-07:
I designed a few plugins for myself, which have been working well for me for about a year now. I would like to integrate this into ESP-Easy so that everyone can use it.
1. Read out the electricity meter from the energy supplier via the optical interface
2. Read out the inverter via Modbus (so far only Kostal have been tested) 
3. Read out a Varta Storage via Modbus
4. I2C-Analog Output card 4 Chanel 0-10V (https://www.horter.de/blog/i2c-analog-output-4-kanaele-10-bit/)
5. Controller that exchanges all device data with a Siemens PLC via S7 communication

Can I get a plugin ID for this?
Besides, I'm starting to create documentation for it.

TD-er, 2023-09-07:
I guess this is for the code in this PR: https://github.com/letscontrolit/ESPEasy/pull/4790

It looks like there is a lot of functional overlap with the P044 P1 gateway.
Is that correct?

For the Modbus plugins, are they almost copy/paste implementations of the AccuEnergy plugin?
Thing is that I'm working on making a more generic way to implement Modbus, so all Modbus plugins could use the same bus (if desired) and not be waiting till a reply is given.
Like the SenseAir S8 is quite slow in responding (200 msec typically) and the AccuEnergy is also a bit slow.
For the Eastron plugin, I already made a start to create a list of registers that need to be read and then handle them one at a time periodically polling whether a new result is in and start requesting the next one.
This way there is no blocking code.

The idea is to make this a generic framework for all Modbus plugins.

The recent changes to the Eastron plugin also deals with reducing the huge amounts of strings by generating them.

I can reserve a number of plugin IDs for your plugins, but please also have a good critical look at the code to see if some of my remarks here can be applied to your plugins/code.

- P155: SML smart meter
- P156: Kostal PV inverter
- P157: Varta Storage
- P158: I2C analog 4ch 0-10V
- P159: Siemens PLC S7

Brommander, 2023-09-14:
Sorry for the late feedback. It took some time to look at the mentioned plugins.

The P044 P1 gateway reads the serial data and makes them available to a web server without evaluation. My plugin evaluates the data and provides them as variables in the device. Furthermore the used interface is a little bit different.

The plugin AccuEnergy uses Modbus RTU. For the inverter and Varta memory Modbus TCP is used. I did not build the plugins generically but programmed them for my use cases. 
I welcome the idea of a generic framework for Modbus. It is ok for me then not to publish my plugins and use them only for me until the better version is ready.
I used the "old" Eastron plugin as a template to select the different registers. I will have a closer look at the last changes.

Thanks for the plugin ids. The communication with a Siemens PLC must be a controller. So that individual devices can be read / written via a selection.

tonhuisman, 2023-09-14:
> The P044 P1 gateway reads the serial data and makes them available to a web server without evaluation.

That's correct, but there is an open FR (and on my non-public todo-list 😉) to add that as a feature, for all available data, and also an FR to support the Swedish variation of the protocol.

TD-er, 2023-09-14:
> Thanks for the plugin ids. The communication with a Siemens PLC must be a controller. So that individual devices can be read / written via a selection.

Check!
I updated the start post, so you can use C020 for this.

flashmark, 2023-11-19:
I would like to play with the ens160 TVOC sensor (gases). This is yeat another I2C based gas sensor. Can I reserve a plugin number for this one? ScioSense provided an Arduino library on GitHub (https://github.com/sciosense/ENS160_driver). Sensor is still traveling from ALi to home...

TD-er, 2023-11-19:
There are lots of calls to `delay()` in that library, so please try to get rid of those as we really don't want to use delays in ESPEasy.

flashmark, 2023-11-19:
Thanks for the warning. Then I have to rewrite it to something more suitable for ESPEasy. Is there any convention on where to put such device specific code in ESPEasy? I see imported libraries in \lib. But if I have to rewrite stuff I assume it is better to put it all in the \src\PluginStructs\Pxxx_data_struct.cpp?  For now I will use P164, I can always renumber it.

tonhuisman, 2023-11-19:
> Is there any convention on where to put such device specific code in ESPEasy? I see imported libraries in \lib

Usually we add the library in the `lib` folder, and sometimes rename it, depending on the level of changes, or if it's backward compatible, just keep the original name. It's always nice to give the improvements back to the original project, though some of the libs haven't been actively maintained for a long time or even abandoned, so feedback can be slow at times.

tonhuisman, 2023-11-19:
The library has a quite peculiar way of controlling the sensor, pulling pins high and low, and also sends logging to the serial port when a variable is set.
And the coding style is, well, 'different' from what's more commonly seen in Arduino libraries.
I suggest to add `#ifdef ENS_DEBUG / #endif` around those debug statements (add a line `// #define ENS_DEBUG` in the .h file so it can optionally be enabled during development), as we want to keep the code as lean as possible.

mgx0, 2024-01-14:
As per P160 here's a playground plugin update: https://github.com/letscontrolit/ESPEasyPluginPlayground/pull/186

tonhuisman, 2024-01-14:
> As per P160 here's a playground plugin update: [letscontrolit/ESPEasyPluginPlayground#186](https://github.com/letscontrolit/ESPEasyPluginPlayground/pull/186)

I'll start migration soon-ish, still working on other issues and improvements 😉

andibaciu, 2024-02-24:
Hello,
I designed a plugins for IKEA Vindstyrka. I would like to integrate this into ESP-Easy so that everyone can use it.
With this plugin you can make next parameters from IKEA device:
- temperature
- humidity
- tVOC
- PM1.0
- PM2.5
- PM4.0
- PM10.0
- and i add a DewPoint parameter (use to calculate relation between temp and humidity)

Can I get a plugin ID for this?

tonhuisman, 2024-02-24:
I've assigned `P167` to the  IKEA Vindstyrka, and hope it can be combined to also support the Sensirion SEN5x devices.
Description for the device should probably be `Environment - Sensirion SEN5x (IKEA Vindstyrka)`, assuming the base sensor is the main feature.

chemmex, 2024-02-27:
@tonhuisman would it be hard to also add NOx from SEN55?

tonhuisman, 2024-02-27:
> @tonhuisman would it be hard to also add NOx from SEN55?

Have to look closely at the PR for P167 (there will be some feedback, as I already had a quick glance at the code 😉), but the NOx measurement is interesting enough to be included in SEN55 mode 👍

andibaciu, 2024-02-28:
> @tonhuisman would it be hard to also add NOx from SEN55?

After PR for P167 finish, I intend to make an update and integrate standalone SEN54 and SEN55 sensors.

flashmark, 2024-03-02:
I am playing with a plugin to control the pump of my floor heating. I want local control with remote override and use a RGB led for status. This made it a bit too complex for rules. However, it is not really a standard plugin, although reusable for thermostats alike control. Question is how to handle such plugins? Is there a reserved plugin range to prevent overlap with the growing amount of plugins? I started with P999, but it looks like the plugin ID is a 8 bit integer, now using P254 locally. Looks likewe are rapidly filling the available space.

tonhuisman, 2024-03-02:
We already have the [ESPEasyPluginPlayground](https://github.com/letscontrolit/ESPEasyPluginPlayground) repository for a long time, but reality is that, because of many improvements to the ESPEasy Core code, and not many updates of the code on the playground, a lot of the plugins there are no longer compatible with the current ESPEasy code. And another issue is that a lot of code is written with quite basic Arduino-knowledge & skills, resulting in, well, less-optimal code, to name it nicely.
I've migrated a few of the plugins there to this repo, and it's usually a lot of work adjusting to both the current code, and fixing inefficient code.

The PluginID is indeed limited to an 8 bit unsigned int (`uint8_t`), so 254 is a safe limit to apply.

By the gist of your explanation, it seems quite reasonable that other ESPEasy users might be interested in your plugin, so the simple solution is to assign the next available plugin ID (`P168`), I just need a matching description to add it to the list 😉 
The only thermostat-like plugins currently available are the Level Control (P021) and ThermoOLED (P109), so there is room for extension there.

flashmark, 2024-03-02:
Thanks for feedback. P109 is too complex for my use case. But I missed P021. This looks close to my prototype, except for the LED feedback. I have to study P021 closely to estimate impact on adding my stuff or just reuse that one.

Current design:
- Several control modes:  (OFF, ON, STANDBY, LOCAL, REMOTE)
- Basic thermostat with hysteresis (now swiching on when temperature is high, but can be made configurable) in LOCAL and REMOTE
- Maintenance run (prevent pump is off for too long) in  STANDBY, LOCAL, REMOTE
- Completely off in OFF
- Always on in ON
- Receive "on"/"off" from an external controller
Still to implement the remote control part.

Note I am pretty well aware of real-time software, but more involved with UML and Powerpoint than having C++ experience... So review is required. It would be nice to have a "plugin builders manual" providing references and design guides and rules. I can help with the writing of such documentation as long as I get the right feedback from the architects.

One of the questions for me is how to access IO pins. Direct Arduino function calls ere available, but I also see some ESPEasy abstraction levels. I need to do reverse engineering of existing plugins which could be potentially bad examples. A cookbook would tell me how to do this best...

flashmark, 2024-03-10:
Still playing with my pump controller. Probably the name fits best as "regulator - heating pump" or "regulator - pump".
Main difference with P021 is additional control to keep a pump running for a minimum time and to switch it on for s short maintenance run. An optional 3-color LED can be used to see if pump is switched on, switched off or running in maintenance mode. If this sounds as an interesting plugin to share I will add the documentation. I am thinking of ways to make it more versatile. But the main purpose is to keep my heating pump running when needed.

TD-er, 2024-03-10:
Not sure if it should be 1 plugin and if it is only a minor extension to an existing plugin, why not adding such a feature to the existing plugin via some parameter or as a command?
Or maybe P021 is not sending an event so maybe that's all what is needed to make it all work?

tonhuisman, 2024-03-11:
> * Several control modes:  (OFF, ON, STANDBY, LOCAL, REMOTE)

The OFF and ON states are clear (as described), but not sure what kind of purpose or meaning Standby, Local or Remote would have?

If the state can be changed by a command, then you have the option of remote-control, as there are several ways of sending commands to a unit, HTTP, UDP (P2P), MQTT, Time based.
One factor to take care of would be long-time timers for the maintenance mode, like 2 weeks or smt. in that range, as you would want that to survive a reboot/power-cycle. And currently, P021 is one of the few plugins that can actually store a value (probably should be a separate task instance), but storing that too often will cause damage to the flash 🤔 

Controlling a multi-color led, either 3-color via a few GPIO pins, or a Neopixel that has a full RGB color scale, could be handled from rules, using the correct events. Adding some new events shouldn't be too hard. Currently, an event is generated when the output state changes (`sendData(event)`).

Controlling GPIO state can be done in the standard 'Arduino' way via `digitalWrite()`, or using some more low-level, but much faster, functions. Several examples can be found, even in P021.

flashmark, 2024-03-13:
Thanks, good food for thoughts and helping me to understand the design patters for ESPEasy. 

My initial idea was to configure the "mode" from the web interface and use an on-off command only for remote control. But that can be changed.
Background: Floor heating pump with standard thermostat control. Hot water coming from heater will start the pump. When heater is switched off (low supply temperature) pump will run for a predefined time to spread remaining heat. 
However, the pump should run at least once a day for 5 minutes to prevent bearings will become dry and pump fails. Additional modes are OFF: Never switch it on; OF: keep it always running; STANDBY: Keep it off, but still run 5 minutes a day; REMOTE: on demand force pump on by a remote control algorithm. 
LED can indeed be separated as long as the internal state is made available. This makes an extension of P021 much easier. I will first finish hardware and current private plugin. Once I get everything running I will study migration to P021.

scottw-finao, 2024-05-16:
Is there any chance of a plugin for the Seeedstudio (grove) i2c capacitive water level sensor?

[Seeed Wiki page](https://wiki.seeedstudio.com/Grove-Water-Level-Sensor/)
[Github code example](https://github.com/SeeedDocument/Grove-Water-Level-Sensor)
[Catalog page](https://www.seeedstudio.com/Grove-Water-Level-Sensor-10CM-p-4443.html)

tonhuisman, 2024-05-16:
> Is there any chance of a plugin for the Seeedstudio (grove) i2c capacitive water level sensor?

I've reserved Plugin ID **`P170`** for the Seeed Studio liquid level sensor, though I'm not yet sure if it should be in the (new) category 'Level', but I couldn't think of a better category to put it in, yet.
I've put it on my name, as from the context [you wrote in the forum](https://www.letscontrolit.com/forum/viewtopic.php?t=10180), I'm assuming you won't be developing this plugin 😅

flashmark, 2024-07-06:
I have been playing with my floor heating pump controller plugin. I see some issues in merging it into P021. This will make the new version not backwards compatible with the original P021. The new plugin is simple, keeping all data in the standard structures. Main difference is the state machine I added to implement the additional, pump related, requirements like maintenance runs, operation modes and time extensions. I still tried to keep it generic. In case it is seen as reusable I can refactor it with a new plugin ID. Let me know if the plugin is seen as useful. If so, provide me with a plugin ID and collection.

tonhuisman, 2024-07-06:
You could add a Yes/No checkbox to enable/disable your new features. (Add a new settings section below Auto-save interval.) For inspiration of how that's usually done, check P037 MQTT Import, where several features and settings are enabled/shown or disabled/hidden using a checkbox (that can submit the page)

And, most likely and said before, there _will_ be someone interested in functionality like this. Especially if you explain in the documentation how it is intended to be used 👍

flashmark, 2024-07-07:
One of the new features is also to provide 2 "values". The output set to the GPIO pin (pump on/off) and the state of the internal state machine. If this can also be managed in a backwards compatible way I will try to merge the two different concepts.

Minor issue is the way how to handle hysteresis. I prefer to use a "switch on" setpoint and a hysteresis value added/subtracted to/from the setpoint to switch off. P021 uses a symmetrical window around the setpoint. But that can be managed too. The concept to modify the setpoint using a command can be merged with the command to force the output to "on" which I need to run my pump whenever the CV is enabled.  

I have to investigate if there are enough PCONFIG parameters available to store all the static data. Both plugins use some of them to store timers and other data. Moving to a P021_data_struct make the code a bit more complex.

I am still wondering how this "auto save" should work in P021. Quite some code to guard saving parameters too often. Something my original plugin does not need, but can be added too. 

So main question for now is can I add a new value to P021 without breaking existing use cases and can I keep the code lean and mean.

tonhuisman, 2024-07-07:
Having a variable number of values, depending on settings, is handled via the DeviceGetValueCount, like done in P026:
```
    case PLUGIN_GET_DEVICEVALUECOUNT:
    {
      event->Par1 = P026_NR_OUTPUT_VALUES;
      success     = true;
      break;
    }
```
That setting can also be a conditional returning either 1 or 2, P026 allows setting 1..4 output values 😃 

Besides the `PCONFIG(0..7)` settings values there are also `PCONFIG_LONG(0..3)` and `PCONFIG_FLOAT(0..3)` available to store values. And there's an alias (union) `PCONFIG_ULONG()` parallel on `PCONFIG_LONG()` to be able to use all 32 bits, as has been done in some plugins like P116 and P150 to store multiple settings (1, 2, 3 4 or 8 bit) in such `ulong`.

tonhuisman, 2024-07-07:
> I am still wondering how this "auto save" should work in P021. Quite some code to guard saving parameters too often.

That second sentence is exactly the reason it is implemented. Some people update the value from rules every few seconds, causing an implicit save, that will eventually wear out the flash memory, rendering the ESP useless. To avoid that, the AutoSave feature keeps track of the currently stored value, and only saves to flash after the timeout is reached if the value is effectively changed.
This plugin is one of the few ways of storing a user value that will survive a cold boot/power cycle of the ESP.

> Minor issue is the way how to handle hysteresis. I prefer to use a "switch on" setpoint and a hysteresis value added/subtracted to/from the setpoint to switch off. P021 uses a symmetrical window around the setpoint.

A more fine-grained way of handling the hysteresis is certainly an improvement over the current handling, as long as it can be made backward compatible (read: current settings value should still result in the same behavior after an upgrade. No requirement of saving the task should to 'fix' that.)

flashmark, 2025-08-24:
I am playing with a generic Modbus RTU plugin for an RS485 link. It is based on Modbus energy sensors. But all device specific behavior is removed. Just an interface to read 4 registers from the device. I will use it on a Modbus rainsensor from Ali. If interested and provide me with a plugin number then I will provide a pull request with code and documentation (still in development).

tonhuisman, 2025-08-24:
We already have plans for a generic `Modbus RTU` interface, as it is in fact a generic interface, quite similar to I2C, SPI and even CAN-bus (there's a PR open for that already), and there are already a couple of plugins that use Modbus RTU for a limited set of devices (P078 as a very good example, P102 as a simple example, just the first few that come up in memory).
I'm not sure if we should add more plugins with a _specific_ Modbus target, but instead go for a more generic approach, much in the line of my PR for a generic I2C interface in P180, that can equally well be applied to SPI, Serial, and in the future Modbus RTU, for example.

tonhuisman, 2025-08-24:
> a Modbus rainsensor from Ali.

Can you provide a link to the exact sensor you are using?

I'll reserve **`P183`** for this plugin. Be prepared it might get a full makeover in the future... 😆

flashmark, 2025-08-24:
Note that the rain sensor is very simple: it is wet or it isn't. The Plugin will provide (up to) 4 registers mapped to the plugin values. I am still not sure how to visualize this in terms of value names and units. The idea is that any simple Modbus RTU device with some registers to expose and no context required can be connected.

[16 more replies]