letscontrolit/ESPEasy #3840: PMS5003T support?

sacy80, 2021-11-07:
[image]

any possible support for PMS5003T with temp and humidity?

TD-er, 2021-11-07:
It is already partially implemented in the code.
If you define `#define PLUGIN_053_ENABLE_S_AND_T` in the code, it might already work, but I have not had one of those PMS5003T or PMW5003S models, so I cannot test it.

Have you tried using the PMS5003ST setting?
I know the S and T version don't have the Formaldehyde sensor, so not 100% sure if it will give the correct values on an S or T version.

sacy80, 2021-11-07:
i try older build with PMS5003ST setting. but got error on frame error.  now trying to compile new build :)

TD-er, 2021-11-07:
That plugin has had quite a mayor overhaul in code in the past month, so you can also just try the latest build I made a few days ago.

tonhuisman, 2021-11-07:
It should have the 'extra' sensors on any 4MB build, I excluded them only on the 1M builds (except the Custom builds).

sacy80, 2021-11-07:
[image]
Look like not able to get Temp and Humi

tonhuisman, 2021-11-07:
If those values are somewhat reasonable for temp and hum, then maybe I have to map them differently, but the documentation & code samples aren't that clear. I don't own that device (and neither any of the other PMS models), so I have to depend on other testers feedback 😅 
You have enabled `PLUGIN_053_ENABLE_S_AND_T` in file `src\src\PluginStructs\P053_data_struct.h` (line 18), and selected that Sensor model, right?

sacy80, 2021-11-07:
[image]

Yup.. 

i try run AC for awhile, look like cnt5.0= Temp ,  cnt10=humi .. the number drop :)

tonhuisman, 2021-11-07:
That #define should be in the `P053_data_struct.h`, where it is at line 18 but commented. Uncomment that and it will enable stuff in the functions that fetch the measurements. Please try that, and retest.

sacy80, 2021-11-08:
i think there diff in data frame for PMS5003ST and PMS5003T. 

it ok. i try recompile again with #define PLUGIN_053_ENABLE_S_AND_T in "P053_data_struct.h" and remove define PLUGIN_053_ENABLE_S_AND_T in "_P053_PMSx003.ino" :)

sacy80, 2021-11-08:
tested.. still the same.  no temp and humi .. look like it diff data byte.  the data sheet that seller sent to me.  temp are data 10 , humi are data 11.. from the P053_data_struct.h  the assign temp are data 13 and humi are data 14.  🤔

tonhuisman, 2021-11-08:
I've patched the offsets by moving the data around if the PMS5003T is selected. This should give you the Temperature and Humidity readings. Unsupported outputs should best be ignored 😉.

A local Custom build with P053 included:
[ESP_Easy_mega_20211108_custom_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/7499908/ESP_Easy_mega_20211108_custom_ESP8266_4M1M.zip)

sacy80, 2021-11-10:
look working :)  thanks alot !!  ☺
[image]

tonhuisman, 2021-11-11:
Please re-open this issue, work is still in progress, and this will be closed automatically once the PR is merged 😃

tonhuisman, 2021-11-12:
I updated the code to _not_ generate events for `#cnt5.0` and `#cnt10` for the PMS5003T sensor, as they are not available in the data.
@sacy80 can you please update with this version and post a similar log like above?

[ESP_Easy_mega_20211112_custom_ESP8266_4M1M.zip](https://github.com/letscontrolit/ESPEasy/files/7529643/ESP_Easy_mega_20211112_custom_ESP8266_4M1M.zip)

PR #3842 (2021-11-18): [P053] Fix support for PMS5003T
Fix reading the Temperature and Humidity from the PMS3005T sensor model.

Resolves: #3840 

To use this sensor requires having a Custom build, and enable compile-time define `PLUGIN_053_ENABLE_S_AND_T` in `Custom.h` or a `pre_custom*.py` script.

May need adjusting of event names for this sensor model.
closed 2021-11-18 by TD-er