letscontrolit/ESPEasy #3841: Electricity cutoff logs?

cookins, 2021-11-08:
Is there any plugin that love electricity cutoffs? Like when esp8266 gone off and when gone on, as far as there time from internet can be checked? It's could be really good plugin. Thanks.

tonhuisman, 2021-11-08:
Some possible solutions: (No plugin needed, AFAICS)
- Use a battery powered/buffered ESP (and WiFi router/internet connection) and a sensor that measures the mains availability
- Write log to a syslog server (with battery backed power supply) that registers the every minute `Clock#Time` event log (Info level), so you know with 1 minute accuracy when power was removed and restored

For an exact time-log you should enable NTP time and Rules.

cookins, 2021-11-08:
> Some possible solutions: (No plugin needed, AFAICS)
> - Use a battery powered/buffered ESP (and WiFi router/internet connection) and a sensor that measures the mains availability
> - Write log to a [...]

There is much easier and more handy solution. In hardware, in values, you can choose "uptime". [image]
Also, I meant, to upload it to iot server, as I got (but my free server can keep only 2 days logs).
[image]

TD-er, 2021-11-08:
You can add a single check to make sure it wasn't a crash, but just a power cycle.
If you store a value in a dummy task, (e.g. the last known uptime or simply a "1"), then this value will be restored at boot when the unit remained powered.
But as soon as the power was lost, it will not restore this dummy task value and thus it is a 0.

So something like this in the rules:
```
on System#BootMode do
  if [dummy#one] = 0 // Assume there is a task named "dummy" with the first value named "one"
    // Do whatever is needed to mark a power cycle
    taskvalueset,1,1 // Assuming the dummy task named "dummy" is at task 1
  endif
endon
```

tonhuisman, 2021-11-08:
>`  taskvalueset,1,1 //...`

That should probably be `taskvalueset,dummy,one,1` 😉

TD-er, 2021-11-08:
Yep, that's also possible (in more recent builds ;) )

tonhuisman, 2023-09-01:
No longer valid, it seems, so can be closed.

closed 2023-09-01 by TD-er