letscontrolit/ESPEasy #3843: RFID Wiegand issue with newer Builds after 02.08.2021

sanyafifa, 2021-11-11:
Hi! 
In recent releases, I am not getting any tags from the device.
My working firmware is:
**ESP_Easy_mega_20210802_normal_ESP32_4M316k.bin**

TD-er, 2021-11-11:
This was the commit to this file after your last-working-build: https://github.com/letscontrolit/ESPEasy/commit/5ac08ecfef72cbd1efd3a31c9b5edb1466a1d2e5#diff-90d3cf82dd22241e8be16d82f211f7536f6bdc5217491cc2bab1854f08f2272d
Made by @tonhuisman

TD-er, 2021-11-11:
One of the more obvious changes I can see in that commit is that the pins may have changed.
Can you test with D0 and D1 swappped? (can be done in the settings :) )

sanyafifa, 2021-11-11:
> Can you test with D0 and D1 swappped? (can be done in the settings :) )

in my setup do i need to change GPIO25 from GPIO26? ))

TD-er, 2021-11-11:
Yep, just in the settings you showed as screenshot.

sanyafifa, 2021-11-11:
OK! I will report the result

sanyafifa, 2021-11-11:
@TD-er ,
does not work,
log:
[image]

TD-er, 2021-11-11:
OK, so then @tonhuisman may want to have a look at it, as he was the last one to commit on that plugin. (assuming it is caused by a change in the plugin and not something else)

tonhuisman, 2021-11-11:
I haven't found any obvious error in my changes yet, but one thing that comes to mind is that the interrupt handler functions don't have the `ICACHE_RAM_ATTR` attribute applied, that is applied in some other interrupt driven plugins, but not in all... Not sure if that makes a difference. Maybe combined with the changes in the Arduino libraries, and the fact this is running on ESP32 (at least the unit is named ESP32_SKUD...) this is something that could be a breaking factor.
Again, I don't own this type of hardware, so not really possible to test this myself 🔕

TD-er, 2021-11-11:
ESP32 doesn't name it `ICACHE_RAM_ATTR`.
For ESP32 it is `IRAM_ATTR` and ESP8266 does have an define for that... or is that in my (still pending) core 3.0.0 PR?

tonhuisman, 2021-11-12:
In P008 that `ICACHE_RAM_ATTR` attribute was a bit hidden, as it is applied at the function declaration (only), not at the implementation. So that's probably not the cause of the issue here.
The counter variable used doesn't have the `volatile` attribute added. That _could_ be more important when running on ESP32, but again, IDK...

Made a local build with that change, @sanyafifa can you please test?
[ESP_Easy_mega_20211112_normal_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7528002/ESP_Easy_mega_20211112_normal_ESP32_4M316k.zip)

TD-er, 2021-11-12:
When not using "volatile" where it should, the ESP will more likely crash instead of no longer working as described here.

tonhuisman, 2021-11-12:
Yes, that's what I expected, but I can't find anything obvious here, so I'm in nit-picking mode now 🤔

TD-er, 2021-11-12:
OK, then I'll leave you to it, as nit-picking should not be disturbed :)

sanyafifa, 2021-11-12:
> Made a local build with that change, @sanyafifa can you please test?
> [ESP_Easy_mega_20211112_normal_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7528002/ESP_Easy_mega_20211112_normal_ESP32_4M316k.zip)

@tonhuisman 
ok i will test in the next 24 hours.

sanyafifa, 2021-11-13:
@tonhuisman 
Does not work :(
Swapped D0 and D1

[image]

tonhuisman, 2021-11-14:
I made a build on current `mega` branch, but with the Wiegand RFID plugin reverted to 20210802 source (only reformatted using Uncrustify to be able to compare with current source).
If this solves your issue then it is me to blame that it broke (and will try to fix, of course 😄), if this _doesn't_ solve your issue, then I'm out of clues, I guess.

So: `mega` build with [P008] from 20210802 release:
[ESP_Easy_mega_20211114_normal_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7533448/ESP_Easy_mega_20211114_normal_ESP32_4M316k.zip)

@sanyafifa Please report your findings.

sanyafifa, 2021-11-14:
@tonhuisman 

> So: `mega` build with [P008] from 20210802 release: [ESP_Easy_mega_20211114_normal_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7533448/ESP_Easy_mega_20211114_normal_ESP32_4M316k.zip)
> 

Works! Thanks for your time on my problem!
[image]

tonhuisman, 2021-11-14:
Well, it still isn't solved. The changes were done to solve another issue.
Please re-open again, so this will stay in sight.

tonhuisman, 2021-11-14:
Did some fine-tuning after reverting to the newest plugin source (and comparing the differences, of course).
Added some values to the log, so if you can please test this build, and again show a similar log as above, that would be very helpful, so TIA!

[ESP_Easy_mega_20211114_normal_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7534487/ESP_Easy_mega_20211114_normal_ESP32_4M316k.zip)

ypoosn, 2021-11-19:
Hi 
maybe I can drop in, because I have the same problem. My RFID-reader is a TF1 with Wiegand interface.
I also tested with tasmota and I can confirm that the HW setup is working correctly.

These are the results with your build ESP_Easy_mega_20211114_normal_ESP32_4M316k.bin given above:

**Test A1** 
- Wiegand **26bits** for reader and ESPEasy
- hex output
- card WEG34 0003215576 or WEG27 04904312

log output ESPEasy:
`RFID : New Tag: 13561639, 0xceef27, mask: 0xffffff Bits: 26`

log output tasmota (for reference)
```
17:54:29.106 MQT: tele/NodeMCU-2/SENSOR = {"Time":"2021-11-19T17:54:29","Wiegand":{"UID":3215576,"Size":26}}
17:55:03.440 MQT: tele/NodeMCU-2/SENSOR = {"Time":"2021-11-19T17:55:03","Wiegand":{"UID":3110D8h,"Size":1Ah}}
```
---

**Test A2** 
- Wiegand **26bits** for reader and ESPEasy
- hex output
- keycode input: 123456#

log output ESPEasy:
`23135831: RFID : New Tag: 16653759, 0xfe1dbf, mask: 0xffffff Bits: 26`

log output tasmota (for reference)
```
17:57:22.006 MQT: tele/NodeMCU-2/SENSOR = {"Time":"2021-11-19T17:57:21","Wiegand":{"UID":1E240h,"Size":1Ah}}
17:57:56.020 MQT: tele/NodeMCU-2/SENSOR = {"Time":"2021-11-19T17:57:56","Wiegand":{"UID":123456,"Size":26}}
```
---

**Test B1**
- Wiegand **34bits** for reader and ESPEasy
- hex output
- card WEG34 0003215576 or WEG27 04904312

log output ESPEasy:
`22554229: RFID : Old Tag: 0, 0x0, mask: 0x0 Bits: 34`

log output tasmota (for reference)
```
18:03:47.819 MQT: tele/NodeMCU-2/SENSOR = {"Time":"2021-11-19T18:03:47","Wiegand":{"UID":4298182872,"Size":34}}
18:04:34.485 MQT: tele/NodeMCU-2/SENSOR = {"Time":"2021-11-19T18:04:34","Wiegand":{"UID":1003110D8h,"Size":22h}}
```

---

**Test B2**
- Wiegand **34bits** for reader and ESPEasy
- hex output
- keycode input: 123456#

log output ESPEasy:
`23035230: RFID : Old Tag: 0, 0x0, mask: 0x0 Bits: 34`

log output tasmota (for reference)
```
18:06:49.551 MQT: tele/NodeMCU-2/SENSOR = {"Time":"2021-11-19T18:06:49","Wiegand":{"UID":10001E240h,"Size":22h}}
18:07:57.980 MQT: tele/NodeMCU-2/SENSOR = {"Time":"2021-11-19T18:07:57","Wiegand":{"UID":4295090752,"Size":34}}
```

hope this will help to fix the bug.

best,

Ypo

tonhuisman, 2021-11-19:
@ypoosn thank you for your valuable feedback! 👍 

A question: the tags scanned using ESPEasy and Tasmota, are they supposed to be the same, or are they from different cards? Because if it is the same card then there seems to be some kind of a decoding issue 🤔

ypoosn, 2021-11-19:
> @ypoosn thank you for your valuable feedback! 👍
> 
> A question: the tags scanned using ESPEasy and Tasmota, are they supposed to be the same, or are they from different cards? Because if it is the same card then there seems to be some kind of a decoding issue 🤔

The card I used is the same. This is the card number in two different encodings printed on the card 
WEG26: 04904312
WEG34: 0003215576

The keycode I used was 123456# 

let me know if you need further testing or input.

best

Ypo

tonhuisman, 2021-11-19:
@ypoosn The 0-result for > 32 bit configuration is fixed now. Casting is required when applying that to a > 32 bit sized variable.

Can you also swap the GPIO pins in ESPEasy configuration, to help investigate if the bits are handled in the incorrect order? So after swapping these GPIO's I'd expect the correct result for the 26 bit card.
If that fixes it, I'll re-wire the GPIO handling (was suggested before but not confirmed).

ypoosn, 2021-11-19:
> Can you also swap the GPIO pins in ESPEasy configuration, to help investigate if the bits are handled in the incorrect order? So after swapping these GPIO's I'd expect the correct result for the 26 bit card. If that fixes it, I'll re-wire the GPIO handling (was suggested before but not confirmed).

Yes, after swapping GPIO pins the log shows the correct number
`1571235: RFID : New Tag: 3215576, 0x3110d8, mask: 0xffffff Bits: 26`

Keypad codes are also shown correctly.

Great!

Best, Ypo

tonhuisman, 2021-11-19:
Thanks, I'll modify that too.

tonhuisman, 2021-11-20:
Code has been updated to have a 'Backward compatibility mode`, enabled after upgrade, disabled for newly added tasks. Builds can be downloaded from the Github Actions page for the PR (linked above), or requested here if you can't download it from there.

ypoosn, 2021-11-20:
Thumbs up! Build works for me as expected.

tonhuisman, 2021-11-20:
Great, thank you for testing 👍

sanyafifa, 2021-11-20:
> and again show a similar log as above, that would be very helpful, so TIA!
> 
> [ESP_Easy_mega_20211114_normal_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7534487/ESP_Easy_mega_20211114_normal_ESP32_4M316k.zip)

hi! @tonhuisman 
unfortunately does not work :(

[image]

Came back to this version:

> So: `mega` build with [P008] from 20210802 release:
> [ESP_Easy_mega_20211114_normal_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7533448/ESP_Easy_mega_20211114_normal_ESP32_4M316k.zip)

tonhuisman, 2021-11-20:
@sanyafifa 

> > and again show a similar log as above, that would be very helpful, so TIA!
> > [ESP_Easy_mega_20211114_normal_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7534487/ESP_Easy_mega_20211114_normal_ESP32_4M316k.zip)
> 
> hi! @tonhuisman unfortunately does not work :(

Well, that version you tried is from _before_ I actually fixed the issue, this one should have the fix though:
[ESPEasy_normal_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7575020/ESPEasy_normal_ESP32_4M316k.zip)
(Downloaded from Github Actions page)

sanyafifa, 2021-11-20:
> Well, that version you tried is from _before_ I actually fixed the issue, this one should have the fix though:
> [ESPEasy_normal_ESP32_4M316k.zip](https://github.com/letscontrolit/ESPEasy/files/7575020/ESPEasy_normal_ESP32_4M316k.zip)
> (Downloaded from Github Actions page)

with this version everything is ok! thanks!

[image]

tonhuisman, 2021-11-20:
Thanks for reporting 👍

PR #3853 (2022-01-09): [P008] Wiegand bugfix for 32 bit or larger frames and swapped GPIOs
- Fixes a casting issue for the bit-shifting introduced in a previous improvement
- Fixes swapped GPIO pin setup with added backward compatibility setting

**NB:** The GPIO pins seem to have been 'swapped' since the initial version of this plugin, so the compatibility setting is required for getting the same results as with older versions of this plugin.
New instances of this device will un-check this by default, and existing tasks will have it enabled after an ESPEasy update (and still work after downgrading to an older ESPEa [... cut]
closed 2022-01-09 by TD-er