letscontrolit/ESPEasy #3847: Rotary Encoder issue when send TaskValueSet

rizzo86, 2021-11-14:
Hello

When send Command TaskValueSet,4,1,36 this put in task Encoder Counter to 36 but when i roll encoder they reset value 60 and start incrise from 0... 1,2,3.... I need start incrising from 36...37,38,39... Is this OK or I do something wrong? Also I have set in encoder setings Limit Min 30 and Limit max 60.

Thanks
Best regards

**Hardware:**

ESP12E

**ESP Easy version:** 

 ESP_Easy_mega_20211105_normal_ESP8266_4M1M_VCC Nov 5 2021

tonhuisman, 2021-11-14:
You probably haven't yet found the new option `Allow TaskValueSet on all plugins:` on the Tools/Advanced settings page. That setting is disabled by default.

rizzo86, 2021-11-14:
I forget write. I enable this...

tonhuisman, 2021-11-14:
I've not used this plugin myself, but it looks like it handles the command `encwrite,<count>` to set a different value to the counter.

When using multiple of these encoders you can prefix the command with `[taskname].encwrite,<count>` to address a specific task by name, where the square braces around `taskname` are optional.

rizzo86, 2021-11-14:
Its work... Thanks. Anyway how find this command? I search on wiki page and did't find this?

tonhuisman, 2021-11-14:
Documentation is somewhat of a weak point, especially on older plugins, where sometimes there wasn't even documentation on the Wiki, and the original author is probably no longer working on ESPEasy (didn't investigate for this plugin though 😊), so there is nothing to move to the current Read The Docs documentation (that is generated from content in the source-code repository, here on Github). Reverse engineering commands is the easy part, how it is supposed to be used, or how it was intended, is much harder.

The command I [found in the source-code](https://github.com/letscontrolit/ESPEasy/blob/mega/src/_P059_Encoder.ino#L179-L199) of this plugin, but then you have to at least partially understand how ESPEasy internals work, and it helps if you can read C++ source code 😉

rizzo86, 2021-11-14:
Aha from source i see now.

Thanks. I resolved problem we can close this now.

closed 2021-11-14 by rizzo86