letscontrolit/ESPEasy #3857: increasing the trigger

Ricco59163, 2021-11-21:
hi all, 

I use the SR04M and i want to increasing the trigger  to 20ms , where is this option ?

thx all

tonhuisman, 2021-11-21:
That will need a change in the source code of plugin P013, but I'm unsure if a) the sensor can handle that measuring frequency, and b) the ESP can handle the increased load this will cause, as each trigger will start a parsing of the rules files (assuming you have Rules enabled), and 50x per second is (way) too often.

Ricco59163, 2021-11-21:
what is the defaut trigger ?

TD-er, 2021-11-21:
Not sure if you both are talking about the same here.
Without looking at the code, it seems like the request here is to increment the sending duration of the ultrasonic transceiver to 20 msec.
So it isn't like what we call in ESPEasy the "interval", but the duration of sending an ultrasonic pulse.

One side-effect I can think of here is that increasing the trigger pulse width, will also affect the measurable range.
But what effect does it have?

TD-er, 2021-11-21:
Hmm looking at the source code, it seems like the trigger width is more like microseconds, which makes more sense.
```c++
	*_triggerOutput &= ~_triggerBit;   // Set the trigger pin low, should already be low, but this will make sure it is.
	delayMicroseconds(4);              // Wait for pin to go low.
	*_triggerOutput |= _triggerBit;    // Set trigger pin high, this tells the sensor to send out a ping.
	delayMicroseconds(10);             // Wait long enough for the sensor to realize the trigger pin is high. Sensor specs say to wait 10uS.
	*_triggerOutput &= ~_triggerBit;   // Set trigger pin back to low.
```
Maybe this 10 usec in the code is what you meant?
Do you have a link to where this 20 msec is adviced?

Ricco59163, 2021-11-22:
https://tutorials.probots.co.in/communicating-with-a-waterproof-ultrasonic-sensor-aj-sr04m-jsn-sr04t/

TD-er, 2021-11-22:
Check!
I think it is an error in the text to mention 20 msec, as the chart also mentions used and the default is also noted as 10 msec, while it clearly is 10 usec.

Anyway, this can be made into a trigger width value. (in usec)
Not sure if we need to make it a combobox, or a numerical input field.

Ricco59163, 2021-11-22:
numerical input field plz

tonhuisman, 2022-12-29:
@Ricco59163 I have created a PR (see link above) to add this feature. Can you test the related Github Actions build to validate it working as intended, with your setup?

tonhuisman, 2023-01-15:
@Ricco59163 Ping? You can use the `Binaries.zip` download from this Github [Actions build](https://github.com/letscontrolit/ESPEasy/actions/runs/3925314544) (when it's finished).

PR #4442 (2023-03-29): [P013] Add setting for trigger-width and Combined mode
Resolves #3857 
Replaces #3157

Features:
- Add a setting to configure the trigger-width for starting a measurement. Range 10..20 usec. Defaults to 10 (the current fixed width).
- ~Add a setting to optionally disable repeating the last value event when using Mode = State.~ Removed/inhibited the added setting, but allow an Interval of 0, effectively suppressing any events.
- Add Combined mode, as started in #3157, also adding the missing bits.

NB: The added features are _not_ available in the `1M` and `Minimal` builds for .bin size reasons [... cut]
closed 2023-03-29 by TD-er