letscontrolit/ESPEasy #3858: Shelly 1 v3 - Rules System#Boot, disable Switch input

stwa2020, 2021-11-21:
Shelly 1 v3 
Erased with 'blank_2MB.bin'
Flashed with 'ESP_Easy_mega_20211105_hard_Shelly_1_2M256.bin'
Connected Shelly to my WiFi Network
Changed Hardware -> GPIO SDA / SCL to NONE
Configured GPIO 4 as Switch and connected a Lamp
Configured GPIO 5 as Switch and connected the Switch.
https://forum.creationx.de/forum/index.php?thread/1144-shelly-bei-wechsel-kreuz-stromsto%C3%9F-schaltung/
Enabled Rules & NTP

_Rule1
on System#Boot do
gpio,4,0
gpio,5,0
endon

on Button#Switch=1 do
 gpio,4,1  
endon

on Button#Switch=0 do
 gpio,4,0  
endon_

Switch button works
Switch button _doesn't work_ after a power off/on, disconnect from FTDI or 240V and connect again.

When I remove the following section from the rules the Switch start to work after a reboot is done.
_on System#Boot do
 gpio,4,0
 gpio,5,0
endon_

I've  now flashed again and use 'ESP_Easy_mega_20211105_custom_alt_wifi_ESP8266_1M.bin' and the GPIOs are on '0' after several reboots have been done but I can't define the status in the 'system#boot'

If also tried,

To check if the settings are used 
 _gpio,4,1
 gpio,5,1_
and it set them to 1 but Switch doesn't work, Input is not recognized.

This setting has no affect and the Switch works
 _TaskValueSet 2,1,0  
 TaskValueSet 1,1,0  
or
 TaskValueSet 2,1,1  
 TaskValueSet 1,1,1_  

When the Input Switch doesn't work I can use the command gpio,5,0 or gpio,5,1 and this works but not the Input using the Physical Switch.

Any Ideas, should I test additional points ?

Cheers,

Stefan

tonhuisman, 2021-11-21:
You shouldn't set `gpio,5,x`, as that will set the gpio for _output_, so it won't do proper, or at least erratic, _input_ handling.

tonhuisman, 2023-09-02:
No response, can probably be closed.

closed 2023-09-02 by TD-er