letscontrolit/ESPEasy #3859: Cannot get P076 to work - am I doing something wrong?

blazej222, 2021-11-22:
I have a very uncommon device DGM-P162 equipped with BL0937 sensor. I've previously flashed it with `normal_ESP8266_1M` - completely normal build. Shortly after I noticed my device contains energy meter, I wanted to try it out. Unfortunately, all `energy` builds are too big to fit in 1M flash, so I compiled my own, custom version containing basically only `P076` plugin and OTA Updates by defining `PLUGIN_BUILD_MINIMAL_OTA` in my Custom.h (I really want the OTA functionality since this device's construction makes it very hard to connect serial ports, and when they're connected it is impossible to connect device to high voltage because circuit lacks isolation. I'd have to do soldering every time I wanted to flash firmware).

From Tasmota users, I know this device has following pin configuration:
gpio4 - CF
gpio5 - CF1
gpio12 - SEL

After creating sensor device in Devices tab, it indeed downloaded calibration values, and comparing to what other people saw on their devices running Tasmota, those seem to be correct. 
However, I'm unable to get a reading on Voltage, Current, Power and PowerFactor values. 
No matter how I set CF1 and CF interrupt edge, all values stay set to `0.00`. 
Only when I switch my relay off to disable my 80W bulb, Power and PowerFactor momentarily change to some junk values.

Here's my Custom.h file. Do I need to include anything else to make it work? Did I compile something wrong?

<details>
<summary>Config.h</summary>

```CPP
#ifndef ESPEASY_CUSTOM_H
#define ESPEASY_CUSTOM_H

#include <Arduino.h>

/*
    To modify the stock configuration without changing the EspEasy.ino file :
    1) rename this file to "Custom.h" (It is ignored by Git)
    2) define your own settings below
    3) define USE_CUSTOM_H as a build flags. ie : export PLATFORMIO_BUILD_FLAGS="'-DUSE_CUSTOM_H'"
 */

/*
 #######################################################################################################
   Your Own Default Settings
 #######################################################################################################
    You can basically ovveride ALL macro defined in ESPEasy.ino.
    Don't forget to first #undef each existing #define that you add below.
    But since this Custom.h is included before other defines are made, you don't have to undef a lot of defines.
    Here are some examples:
 */

#ifdef BUILD_GIT
# undef BUILD_GIT
#endif // ifdef BUILD_GIT

#define BUILD_GIT           "My Build: "  __DATE__ " "  __TIME__

#define DEFAULT_NAME        "MyEspEasyDevice"                        // Enter your device friendly name
#define UNIT                0                                        // Unit Number
#define DEFAULT_DELAY       60                                       // Sleep Delay in seconds

// --- Wifi AP Mode (when your Wifi Network is not reachable) ----------------------------------------
#define DEFAULT_AP_IP       192, 168, 4, 1                           // Enter IP address (comma separated) for AP (config) mode
#define DEFAULT_AP_SUBNET   255, 255, 255, 0                         // Enter IP address (comma separated) for AP (config) mode
#define DEFAULT_AP_KEY      "configesp"                              // Enter network WPA key for AP (config) mode

// --- Wifi Client Mode -----------------------------------------------------------------------------
#define DEFAULT_SSID                         "xxx"                   // Enter your network SSID
#define DEFAULT_KEY                          "xxx"      // Enter your network WPA key
#define DEFAULT_SSID2                        ""                      // Enter your fallback network SSID
#define DEFAULT_KEY2                         ""                      // Enter your fallback network WPA key
#define DEFAULT_WIFI_INCLUDE_HIDDEN_SSID     false                   // Allow to connect to hidden SSID APs
#define DEFAULT_USE_STATIC_IP                false                   // (true|false) enabled or disabled static IP
#define DEFAULT_IP                           "192.168.0.50"          // Enter your IP address
#define DEFAULT_DNS                          "192.168.0.1"           // Enter your DNS
#define DEFAULT_GW                           "192.168.0.1"           // Enter your Gateway
#define DEFAULT_SUBNET                       "255.255.255.0"         // Enter your Subnet
#define DEFAULT_IPRANGE_LOW                  "0.0.0.0"               // Allowed IP range to access webserver
#define DEFAULT_IPRANGE_HIGH                 "255.255.255.255"       // Allowed IP range to access webserver
#define DEFAULT_IP_BLOCK_LEVEL               1                       // 0: ALL_ALLOWED  1: LOCAL_SUBNET_ALLOWED  2:
// ONLY_IP_RANGE_ALLOWED
[... 343 lines cut ...]
// #define USES_C008   // Generic HTTP
// #define USES_C009   // FHEM HTTP
// #define USES_C010   // Generic UDP
// #define USES_C011   // Generic HTTP Advanced
// #define USES_C012   // Blynk HTTP
// #define USES_C013   // ESPEasy P2P network
// #define USES_C014   // homie 3 & 4dev MQTT
// #define USES_C015   // Blynk
// #define USES_C016   // Cache controller
// #define USES_C017   // Zabbix
// #define USES_C018   // TTN/RN2483

/*
 #######################################################################################################
 ###########     Notifiers
 #######################################################################################################
 */

// #define USES_N001   // Email
// #define USES_N002   // Buzzer

#endif // ESPEASY_CUSTOM_H

```
</details>

tonhuisman, 2021-11-22:
Have you tried any of the predefined devices in the device configuration, aside from Custom?

TD-er, 2021-11-22:
The DGM-P162 is not mentioned in the predefined pin settings selector.
So it is a bit of guess work to see which combination you need for the Custom Pin Settings.
Here the settings for the Sonoff POW r1 to get an idea of the values you may need to change:

[image]

In total there are 32 combinations.
Maybe you can see what others may have used on Tasmota?
If you found settings that work, please post them here, so I can add them to the predefined selector.

tonhuisman, 2021-11-22:
I guess it is this one: https://templates.blakadder.com/smartdgm_PP-W162.html ?

blazej222, 2021-11-22:
> I guess it is this one: https://templates.blakadder.com/smartdgm_PP-W162.html ?

Yup, that's my device, some letters on cover are kinda faded, hence I said it was P-162. My bad.

 > The number of hits on Google is extremely low.

I think it's sold only in Poland as a cheap Smart Home solution. I might be wrong tho.

> Have you tried any of the predefined devices in the device configuration, aside from Custom?

I've tried every device that has the same pins defined. So effectively I've tried those:

```CPP
case P076_KMC           : SEL_Pin = 12; CF_Pin =  4; CF1_Pin =  5; Cur_read = HIGH; CF_Trigger =  CHANGE; CF1_Trigger = CHANGE; break;
case P076_Aplic         : SEL_Pin = 12; CF_Pin =  4; CF1_Pin =  5; Cur_read =  LOW; CF_Trigger =  CHANGE; CF1_Trigger = CHANGE; break;
case P076_SK03          : SEL_Pin = 12; CF_Pin =  4; CF1_Pin =  5; Cur_read =  LOW; CF_Trigger =  CHANGE; CF1_Trigger = CHANGE; break;
case P076_Teckin        : SEL_Pin = 12; CF_Pin =  4; CF1_Pin =  5; Cur_read =  LOW; CF_Trigger = FALLING; CF1_Trigger = CHANGE; break;
case P076_Gosund        : SEL_Pin = 12; CF_Pin =  4; CF1_Pin =  5; Cur_read =  LOW; CF_Trigger = FALLING; CF1_Trigger = CHANGE; break;

```

None of it worked.

Could this be caused by define `PLUGIN_BUILD_MINIMAL_OTA` disabling some functionality the plugin needs to work? Or maybe this plugin needs another plugin to work, like `P003` ?

I also had to comment out `undef` here so it would compile in 1M version.
https://github.com/letscontrolit/ESPEasy/blob/549f76e23fcb872de74b53958be5d3460fc4e804/src/src/CustomBuild/define_plugin_sets.h#L1614-L1617

I will try all combinations once again, but maybe I'm doing something wrong at compile time?

tonhuisman, 2021-11-22:
Well, as TD-er suggested, there are about 32 possible combinations of Custom settings for CF, CF1 and SEL, so there is some stuff to play with, if you haven't done that yet.

Doesn't look like any feature is crippled by the plugin being included in an 1MB build like `MINIMAL_OTA`, other than the entire plugin being disabled, but you resolved that already.
Because of the size of things, including the version of Arduino framework used, I currently don't expect that this will fit in an OTA build, but I haven't tried that. It will require the 2-step OTA method to start with.

blazej222, 2021-11-22:
I've gone through all 32 combinations with bulb connected and relay powered on. After every change I refreshed Devices page. Still, nothing. Shows `0.00` in all fields 😞

TD-er, 2021-11-22:
Are you sure none of these GPIO pins are used elsewhere?
For example GPIO4 and 5 are used for I2C by default.

blazej222, 2021-11-25:
Ok, I finally got it to work properly. Turned out my unit was broken. I took the device apart, checked connections, and indeed, `GPIO4` was CF, `GPIO5` was CF1, but SEL was not connected to anything. 

I soldered a few cables to create working serial connection and connected it to my PC and 230V at the same time (using isolation transformer ofc).
 Then, I reprogrammed it with a simple sketch I had written, which basically kept printing results of `pulseIn()` on GPIO4 and 5 to serial. I was getting 0's all the time. Measured everything by myself with oscilloscope, and there was nothing on CF1 and CF going out, no matter what load was connected.

So, I grabbed a new unit (bought like 20 of those for half free). Still, couldn't get any good results with EasyESP, so I wrote my own code. 
First version measured CF and CF1 pulse with `pulseIn()`. 
Since this function returns pulse width, and according to manual [here](https://www.belling.com.cn/media/file_object/bel_product/BL0937/datasheet/BL0937_V1.02_en.pdf) width is constant 38us, I quickly came up with following formula to convert it to frequency : 
`1 000 000us/(pulse_width+38us)` where `pulse_width` is distance between `LOW` and `HIGH` state. I started getting somewhat-consistent results.

Connected some load - 20W, 500W and 1KW. Approximated that I have to multiply the formula used above by ~1,78 to get correct readings on Power (the number depends on load connected and it isn't a linear function, but with ~1,78 it's always within 5% of real value). 

Next step was converting my code to use interrupts, and from datasheet we know that both interrupts need to be triggered on `FALLING`.

Then, I looked into `HLW8012.cpp` file and looked up differences between my code and code of the plugin. After doing some math I knew correct multipliers for my device, flashed EasyESP and configured everything properly. 

And it *kinda* worked. Kinda, because Voltage value is off by a lot. 
It happens because plugin sets `SEL` pin to HIGH, treats CF1 input as Voltage, then changes `SEL` to LOW and treats CF1 input as current. In my case, **`SEL` PIN IS NOT CONNECTED ANYWHERE(IS ALWAYS LOW)**. 

So, @TD-er, proper configuration for those units (DGM-PP162, but I think it applies to all BL0937-based chips) is `FALLING` for both CF1 and CF interrupt edges. SEL Pin is `LOW`, because it isn't connected to anything and is always `LOW` (CF1 always returns frequency for current).

My suggestion is to change the code of the plugin to include a possibility when SEL pin is always either HIGH or LOW and cannot be changed. In that case, we could simply calculate Voltage as `(Power/Current)`.

What do you think about this solution? If you agree with my suggestion, I can take care of changing the code and commiting my changes to this repo (this can take some time though, I'm very busy with other stuff recently).

TD-er, 2021-11-25:
Just curious, how does it measure power if Voltage isn't known.
Or am I missing something here and is the chip just assuming a constant voltage but you may have to perform the integration yourself so you can also compute a cos-phi?

blazej222, 2021-11-25:
> Just curious, how does it measure power if Voltage isn't known. Or am I missing something here and is the chip just assuming a constant voltage but you may have to perform the integration yourself so you can also compute a cos-phi?

According to [documentation](https://www.belling.com.cn/media/file_object/bel_product/BL0937/datasheet/BL0937_V1.02_en.pdf)

```
The current and voltage signals are sampled with high precision and converted from analog
signals to digital signals. The instantaneous power signal p(t) is obtained after the digital signals
are calculated by digital multiplier. In order to extract the real power component (i.e., the dc
component), the instantaneous power signal is low-pass filtered, and then accumulated (e.g., by a
counter) to generate real energy information. This information is sent to the digital-to-frequency
converter module, and converted to a pulse signal of the certain frequency which is proportional to
the real power.
Similarly, the voltage and current RMS are calculated and then sent to the
digital-to-frequency converter module, converted to a pulse signal of the certain frequency which
is proportional to the voltage and current RMS.
```

So I think BL0937 "knows" Voltage and Current internally, but cannot simultaneously output both of them. In my case, there's no way to force BL0937 to output voltage information on CF1, because I can't change `SEL` pin status (well, without hardware modification).

uzi18, 2021-11-26:
Such option could be good for debug when other users got problems, just mention this in ESPEasy docs. 

@blazej222 do you have only one module or all of yours are broken with same conditions? 
Later It is better to fix module and use full features of plugin

blazej222, 2021-11-26:
> @blazej222 do you have only one module or all of yours are broken with same conditions?

First unit I tried turned out to be completely broken (chip that measures power was dead for some reason).

When I disassembled it, `SEL` pin was not connected to anything, and it looked like it was designed that way. For whatever reason I assumed it was the case for all other modules aswell.
When I got my hands on the other, working module I didn't disassemble it but flashed it using tuya-convert, so I didn't check connections. It also made sense in my head - they don't need to measure voltage, because in such a simple device they would just assume it's constantly 230V, so they don't connect `SEL` to anything.

But now I tested the new module, and indeed, `SEL` is connected to GPIO12. Must've been a unit that was really poorly manufactured.

But I still think the option to work in single mode (voltage or current) might be useful. For debugging, for people who want to measure either one more frequently or simply don't need one of those values, or to simply make units where `SEL` pin malfunctions/is not connected still useful.

Sorry for the trouble!

closed 2021-12-30 by blazej222